emsesp/EMS-ESP32 #1148: Power entity

Olgierd-1, 2023-03-26:
Is it possible to add a boiler entity of actual power in kW?
Now I can see the modulation of the burner (in %).
I know that it is very simple to calculate but it would be usefull to have information about actual power of the heat source.

I have e32 gateway with latest firmware 3.5.1
It is connected to Bosch Condens 9000i boiler.

Regards
Olgierd

proddy, 2023-03-26:
It's possible, but goes against the design on EMS-ESP which is to read and process the raw data from the EMS bus and pass it on to more sophisticated automation systems (Home Assistant, IOBroker, etc) to do the calculations.  Currently we don't have any custom entities other than a special class to calculate if the hot-water/dhw is running (for the shower alert). There was some discussion earlier to make custom telegrams (https://github.com/emsesp/EMS-ESP32/issues/922).

Olgierd-1, 2023-03-26:
Thank you for your reply.
Can you check if the value of the current boiler output is available in the data on the EMS bus.
This information depends on two properties:
1. the current burner modulation
2. the maximum boiler output, which is a property of the specific boiler model. The boiler output is stored in the boiler settings by the boiler manufacturer.
Therefore, I presume that the value of the current boiler power should be available for reading from the EMS bus.

proddy, 2023-03-26:
I haven't been able to find the max KW. Even on my Dutch Nefit TrendLine HRC 30 which is rated for 30Kw the front-panel only shows the % burner power and not the KW.

Vikingn, 2023-03-26:
Same here   on boiler system pressure ,exhaust temperature , input 1 options , input 2 options , input 4 options 
aux heater max limit  is always 0 I think.
But I have a Bosch 6000AV with just electricity.  But when it find this elements should it not also give values ?

Olgierd-1, 2023-03-26:
The front panel of my boiler shows the % burner modulation and value of the maximum boiler output (kW) only.
Yes, you are right I can not find the the current boiler output. I finded this entity in the Bosch thermostat integration in the Home Assistant. That is why I thought it was possible to add such an entity to the EMS-ESP.

proddy, 2023-03-26:
Ah good, that means the kW is somewhere hidden in one of the telegrams most likely. We need to go hunting again!

proddy, 2023-04-07:
The kw can be calculated, possibly, using the new Telegram entity customization. But it's still experimental so wait a while.

Olgierd-1, 2023-04-14:
Good news. Thank you.

MichaelDvP, 2023-09-02:
Please all who are interested in power, check in terminal `read 8 4` and post the result and the nominal power of the heating here, 
Maybe offset 4 shows the nominal power, but we need some more checks.

tp1de, 2023-09-02:
Looks good - maybe other users should test as well:
[image]

proddy, 2023-09-02:
On my combi-CV, which is rated 30KW but running at 22,5KW (manually set burner max power to 75%) it shows:

```
ems-esp:$ read 8 4
003+15:34:31.387 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x04), data: 05 0E 06 40 1E 17 73 00 00 69 6B 68 00 00 5C 04 00 00 00 23 81```

tp1de, 2023-09-02:
The telegram is only send on repowering heating system. Otherwise it needs to be polled

proddy, 2023-09-02:
> The telegram is only send on repowering heating system. Otherwise it needs to be polled

can't remember if we manually poll the telegrams in 'custom entities' @MichaelDvP ? If not, probably should.

tp1de, 2023-09-02:
Yes custom entities are polled every minute

proddy, 2023-09-02:
[image]

tp1de, 2023-09-02:
@proddy I just recognized that only the first custom entity is send to mqtt. Same for you?

proddy, 2023-09-02:
> @proddy I just recognized that only the first custom entity is send to mqtt. Same for you?

I get all of the entries

`ems-esp/custom_data = {"boiler_flowtemp":5.00,"nominalpower":30.00,"minmodulation":23.00,"maxmodulation":115.00}`

tp1de, 2023-09-02:
custom_data I get as well . ... just the homeassistant discovery entries are missing ... or?

proddy, 2023-09-02:
yes you're right. I'm seeing the same. That needs to be fixed.

MichaelDvP, 2023-09-03:
> @MichaelDvP agree?

Yes, and reset the ha_registered flag when adding new entities. Both also for the scheduler.  Should i make the change together with power entities?

proddy, 2023-09-03:
Yes please. I'm offline today

MichaelDvP, 2023-09-03:
@tp1de Please check my testbuild at https://github.com/MichaelDvP/EMS-ESP32/releases
Energy-calculation is stored on burner off, every hour, telnet reset command, webpage reset, web update, web upload settings.
My dhw is heated by solar, burner wll not start for some time, can not test.

tp1de, 2023-09-03:
I installed this version. No parameters... you assume that 0x04 shows nominal power? 
The entities are not in mqtt ... or?
I will observe the values

MichaelDvP, 2023-09-03:
> No parameters... you assume that 0x04 shows nominal power?

0x04 is read, If there is a value it set, otherwise read from NVS. I set it is saved in NVS, but a valid 0x04 value has priority.

>The entities are not in mqtt and api ... or?

Should be in boiler_data: nompower, nrgHeat, nrgWw

tp1de, 2023-09-03:
OK I found the entities within api and in ioBroker now.

Within HA  I see only boiler_nompower. The nrg-Entities are seen within mqtt explorer but are not in ha.
max-value is set to -3 (65533 in api)  and step is 1 and should be less (0.1).

{"uniq_id":"boiler_nrgww","obj_id":"boiler_nrgww","cmd_t":"ems-esp/boiler/nrgww","mode":"box","step":1,"min":0,"max":-3,"stat_t":"ems-esp/boiler_data","name":"Energie Warmwasser","unit_of_meas":"kWh","val_tpl":"{{value_json.nrgww if value_json.nrgww is defined else 0}}","dev":{"ids":["ems-esp-boiler"]},"avty":[{"t":"ems-esp/status","val_tpl":"{{'online' if value == 'online' else 'offline'}}"},{"t":"ems-esp/boiler_data","val_tpl":"{{'online' if value_json.nrgww is defined else 'offline'}}"}],"avty_mode":"all"}

tp1de, 2023-09-03:
@MichaelDvP It's the  max:-3 which hinders ha to discover the entity.

tp1de, 2023-09-03:
@MichaelDvP I switched ww preparation on for a couple of minutes. The recorded energy consumption was:

ems-esp gateway: 3 kWh  (no decimal places)
Buderus App:        3.72 kWh 
ioBroker:               3.79 kWh  (15 seconds samples)

MichaelDvP, 2023-09-04:
Please test new dev0c build. The max should now work. 
The decimals are counted internal, should the output be rounded?

MichaelDvP, 2023-09-04:
Hm, proddys HA generation sets state_class/device_class only for non writable sensors, but this is writable and number. Don't know what's right, I've never seen a understandable concept in the HA rules, for me it looks like they dice on every new release. 

Decimals for the int32 is actually not supported, we don't have this kind of values yet. Also the counter will get big and we don't want a rollover, adding longer ints is much more work. The heatpumps also publish the energy counters only as full kWh.

tp1de, 2023-09-04:
You might be right with the numbers ... all other energy entities in HA are sensors ... we need to check if the energy dashboard accepts number entities as well. Otherwise we need to define new energy sensor entities within HA.
The energy statistics (short- and long-term statistics) written into the database are still not fully understood by myself.
I had problems updating my solar inverter destroying statistics. I have written a node-red script to delete the wrong statistic records. Maybe @proddy can help what to do?

Are the energy entities already rounded (without decimal places)?

tp1de, 2023-09-04:
I checked with number entities .... statistics does not (seem) to work in HA. 
Shall we consider to migrate to sensor entities? ... and provide a separate api command to re-set value when necessary?

MichaelDvP, 2023-09-04:
We can set in the `publish_ha_sensor_config` for the has_cmd:
```
        case DeviceValueType::ULONG:
            snprintf(topic, sizeof(topic), "sensor/%s", config_topic);
            set_ha_classes = true;
            break;
```
The energies are the only ULONG with has_cmd.
I'l update my dev0d to test.

tp1de, 2023-09-04:
@MichaelDvP looks much better now. The energy entities can now be selected for the ha energy dashboard. 
I will report as soon as I get data. 

I understood that internally you use decimal places to calculate the aggregates. 
Why do you need to round to full int when sending the value by mqtt / api?

MichaelDvP, 2023-09-05:
> Why do you need to round to full int when sending the value by mqtt / api?

We have only a few data-types that are used on the bus. the longest is 24 bit and does not use decimals. I have to add new data-type to registration/mqtt/web/api/storage/etc. to handle the decimal. The 4M esp32 is 99.7% full and this will add a lot of new code. Is it worth? The counter will show large numbers after a while, who cares about decimals on values > 100000?

tp1de, 2023-09-05:
The memory limitations of the ESP32 has to be discussed within the near future anyhow. (when to migrate to S3).
Actually I just wanted to check your calculation (accuracy) while comparing the figures. That's a bit difficult and needs longer periods.

Please check your rounding. I think it's not working properly.
Statistics are prepared hourly. Energy consumption for ww preparation was this morning (7:55 to 8:10):

0.88   2.98  --> Sum 3.86 kWh (own calculation within ioBroker  using15 seconds samples)) 
vs
1.12   2.85  --> Sum 3.97 kWh (Buderus App)
vs. 
1.02   2.85  --> Sum 3.87 kWh (Recordings from API without samples interpolation - 59 samples instead of 60 in 2nd hour)
vs.
0.00   3.00  --> Sum 3.00 kWh  (values from energy counters)

All the rest is working fine and HA energy dashboard is showing the values as expected (just rounding missing)

MichaelDvP, 2023-09-05:
Have changed to two digits for api/mqtt, on the webpage it does not work, it shows still full kWh.
Please check.

tp1de, 2023-09-05:
I will check .... start value of nrgww  is now 0.16   was 16 before

MichaelDvP, 2023-09-05:
> as meter m3->kWh: 4.16 kWh 4,5% higher then energy counters consumption

Can you do a watch to telegram 0x18 (or 0xE4 on ems+) when the boiler starts. I use the boiler states heating+flame and dhw+flame. Maybe flame-signal is set later then gas is opend and we lose the startup loss.

> I just ask myself if the nominal power out of telegram 0x04 will work for all users. 

I does not, my boiler reports 0 in telegram 0x04. I read the telegram and if there is zero, the value from nvs is used (nompower is writeable).

tp1de, 2023-09-05:
I am not sure what's the exact definition for nominal power. I would assume heating power. For ww prep with high temperatures the efficiency is below 100%. I believe the difference is caused by integral differences (see higher difference for Buderus values with 60 second samples) and the efficiency. On the other hand I had quite some changes on the gas type - kwh per m3. How can the boiler take this into account? (+2% caloric increase since 8 months).
Buderus indicates differences between 7 and 10% as normal deviations.
The signals are exact and ww loading pump starts even 60 seconds earlier.

tp1de, 2023-09-17:
@proddy @MichaelDvP The power entities are not in 3.6.2-dev.0 yet. When is it planned to have them included?

proddy, 2023-09-26:
in 3.6.2 dev now right?

closed 2023-09-26 by proddy