emsesp/EMS-ESP32 #1152: worcester bosch cannot change heating temperature

higopi18, 2023-04-02:
Hardware S32 V2 ; Software v3.5.1

Either on home assistant or EMS-ESP GUI ; Unable to change Heating Temperature .

I am able to edit the value in home assistant ( or modify on ems-esp gui and press send) but temperature does not change.

I am able to make changes to "Burner Max Power" successfully ; so wondering if "heating Temperature" is somehow locked ....

Thanks
Gopi

higopi18, 2023-04-03:
> start with https://emsesp.github.io/docs/Troubleshooting/#changing-a-value-on-an-ems-device-doesnt-work and see what is being sent and received.

Looks like the command to set heating to 70 goes out ; not sure why it stays unchanged 

2023-04-03 08:48:24.921 D 397: [command] Calling command 'boiler/selflowtemp' (selected flow temperature) with value 70
2023-04-03 08:48:24.921 D 398: [emsesp] Write command successful
2023-04-03 08:48:24.927 D 399: [telegram] Sending write Tx [#152], telegram: 8B 08 1A 00 46
2023-04-03 08:48:24.973 D 400: [emsesp] Last Tx write successful
2023-04-03 08:48:24.973 D 401: [telegram] Sending post validate read, type ID 0x18 to dest 0x08
2023-04-03 08:48:27.219 D 402: [telegram] Sending read Tx [#153], telegram: 8B 88 18 00 1B
2023-04-03 08:48:27.279 D 403: [emsesp] Last Tx read successful

higopi18, 2023-04-03:
Looks like the command to set heating to 70 goes out ; not sure why it stays unchanged

2023-04-03 08:51:45.580 D 482: [command] Calling command 'boiler/heatingtemp' (heating temperature) with value 70
2023-04-03 08:51:45.580 D 483: [emsesp] Write command successful
2023-04-03 08:51:45.623 D 484: [telegram] Sending write Tx [https://github.com/emsesp/EMS-ESP32/issues/163], telegram: 8B 08 16 01 46
2023-04-03 08:51:45.656 D 485: [emsesp] Last Tx write successful
2023-04-03 08:51:45.656 D 486: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
2023-04-03 08:51:45.936 D 487: [telegram] Sending read Tx [https://github.com/emsesp/EMS-ESP32/issues/164], telegram: 8B 88 16 00 1B
2023-04-03 08:51:45.986 D 488: [emsesp] Last Tx read successfu

proddy, 2023-07-03:
I think this is a known problem with some Worcester boilers. There's a short mention in the doc [here](https://emsesp.github.io/docs/Troubleshooting/#changing-a-value-works-at-first-but-is-then-reset-to-its-original-value).

@MichaelDvP anything to add to this?

MichaelDvP, 2023-07-03:
Could be a lot of things, overwrite by thermostat or controller, times treset by boiler, etc. Without system info, settings and full log can't say anything.

basak, 2023-09-02:
On my Greenstar 30 CDi, "heating temperature" changes when I move the physical temperature dial on the boiler. Combined with your result, this suggests to me that the boiler won't accept this being changed externally.

MichaelDvP, 2023-09-29:
> On my Greenstar 30 CDi, "heating temperature" changes when I move the physical temperature dial on the boiler. 

For most boilers (bosch changes it's logic sometimes) you can only set the heating temperature via bus to a value lower than the selected value of the physical dial. And after ~2 minutes the boiler resets to the value of the dial. So the heating temperature command has to be repeated every minute to stay active. That's why we have added the scheduler in ems-esp. (And before the force heating off).

closed 2023-11-19 by proddy