emsesp/EMS-ESP32 #1155: custom telegram handler

proddy, 2023-04-07:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1079

<div type='discussions-op-text'>

<sup>Originally posted by **proddy** January 21, 2023</sup>
from @tp1de 

> Very long time ago I made a proposal to introduce "customizable" entities by telegram-type and offset. Maybe still an idea to support special installations and requirements.

I don't recall that, but its really good idea. We could add to the Customization page the ability to add a new Telegram to the EMS device in which you choose a name, the type (short, int), the divider (div10, div2 etc) and position. This would only be for read telegrams though.

_Originally posted by @proddy in https://github.com/emsesp/EMS-ESP32/issues/908#issuecomment-1399201035_
            </div>

proddy, 2023-04-07:
Michael created this in https://github.com/emsesp/EMS-ESP32/pull/1151

proddy, 2023-04-07:
@MichaelDvP the webUI needs a little tidying up and also some mock data in server.js. Shall I do this?

MichaelDvP, 2023-04-07:
Yes, please do this.

proddy, 2023-04-07:
> Yes, please do this.

I've been working on the web code, making minor changes and then decided to look at replacing all the form validation code with a new library, and at the same time refactoring the component rendering since it's really quite inefficient (many renders). So it's going to take some time. I'm checking it into my local repo

proddy, 2023-04-23:
think we can close this. The updated web code is in PR https://github.com/emsesp/EMS-ESP32/pull/1162 

It's a great feature, thanks.

closed 2023-04-23 by proddy