emsesp/EMS-ESP32 #1158: Bosch Greenstar HIU

Rich456tf, 2023-04-13:
### PROBLEM DESCRIPTION

When connecting EMS-ESP to a Bosch Greenstar HIU many data points are missing.
I have attached the log file.
When logging I made some adjustments to some of the HIU parameters in order to help find the issues.

Timings are approximate but should be within a few seconds...
At 30:17 I enabled the keep-warm feature which should set the keep-warm temperature to 41°C 
At 31:41 I disabled the keep warm feature
The keep-warm feature can be adjusted with a Bosch Sense II controller but it would be great to be able to adjust this via EMS-ESP so the Sense II is not required

At 33:50 I enabled a cold water flowrate of approximately 12 l/min
At 36:01 I stopped the cold water flowrate

At 40:50 I adjusted the heat network flow temperature
At 41:18 I stopped adjusting the heat network flow temperature 

Also worth noting that I have tried to adjust the heating selected flow temperature via EMS-ESP but it does not seem to adjust. I can turn the heating on or off but I can't seem to adjust the setpoint. This would be a great feature to have if it's possible!

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
[emsesp_info (1).txt](https://github.com/emsesp/EMS-ESP32/files/11220985/emsesp_info.1.txt)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/11220986/log.txt)

Attached as log file

```

### TO REPRODUCE

I have attached potentiometers to adjust temperature readings for testing purposes

### EXPECTED BEHAVIOUR

Included in problem description

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

There are quite a few other data points that are not read by EMS-ESP but it would be great to get these important ones sorted then I can work on adjusting other sensors and finding other areas that aren't operating properly.

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2023-04-13:
For a first test: update to dev9 build and then unzip and upload this file:
[emsesp_entities_HIU.zip](https://github.com/emsesp/EMS-ESP32/files/11225121/emsesp_entities_HIU.zip)
You should get in dashoard a custom device with 3 new entities: keepwarmtemp, netflowtemp and cwflowrate. Check it his is right.

Rich456tf, 2023-04-14:
Ok, I've uploaded the file and yes I get 3 custom devices and all seems to work.

keepwarmtemp works correctly and allows me to adjust the temperature to whatever I want which is great. This function sets a temperature to pre-warm the HIU so it will generate hot water quickly when a tap is opened.

netflowtemp displays the correct temperature. However this should be a read only figure. No adjustment option is needed.

cwflowrate displays they correct flowrate of 12.1 however the unit displayed is kWh but it should be l/min

Using my Sense II I've found another function. I set the 'set return temp' to 60°C and I believe the lines below are this...
000+00:44:17.617 T 4869: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x0772), data: 3C (offset 2)
000+00:44:18.355 T 4870: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0772), data: 3C (offset 2)
This is is called 'set return temperature' and it allows the maximum heating return temperature to be adjusted. This should be an adjustable figure.

I've been trying to adjust the selected flow temperature to 75°C but it will not change. I get the line below, is there any more information I can provide which will help find how to do this?
Normally this temperature is set using a rotary dip switch on the HIU.
000+00:38:16.017 T 2415: [emsesp] Me(0x0B) -W-> boiler(0x08), UBASetPoints(0x1A), data: 4B

MichaelDvP, 2023-04-15:
> netflowtemp displays the correct temperature. However this should be a read only figure. No adjustment option is needed.

Actually all custom entities are shown as writeable in dashboard, maybe we should add a setting to make it readonly.
> cwflowrate displays they correct flowrate of 12.1 however the unit displayed is kWh but it should be l/min

Go to Settings->custom entities, click cwflowrate and change the uom to  l/min
For the `setreturntemp`, click ADD and input this values: (there is a bug in input of device/type_id, clear the predefined input and type in the value again).
[image]
`
The selected flow temp is normally set by thermostat, your sense-II sends it to the HIU with this 0x1A telegram. It should work also from ems-esp if the setting on the rotary is higher than the sended value, but is overwritten by the thermostat, that sends the calcuated value frequently.

Rich456tf, 2023-04-17:
Thanks Michael, that's very helpful indeed.
I have added that in as a custom entity. Also I found the hot water control valve position and have added that one too.
[emsesp_entities.zip](https://github.com/emsesp/EMS-ESP32/files/11251218/emsesp_entities.zip)

I'm still struggling to control heating flow temperature though. 
Ideally I'd like to be able to control flow temperature without a Sense II connected at all.
Sending via 0X1A doesn't appear to do anything.

I have logged what happens when I use my Sense II to adjust the flow temperature from 26 up to 70 Degrees and attached the log below.
Trace 201 is around where I gave the command but I can't seem to find the relevant area. Any help would be much appreciated!
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/11251310/log.1.txt)

MichaelDvP, 2023-04-17:
Looks like telegram 0x02E0 is the command telegram. In T223 the thermostat publishes the new value in his monitor, sends the command in T224 to the boiler and boiler publishes in his monitor in T225. 
Maybe it's a similar structure as the 0x1A command telegram, then try 0x02E0 offset 1 as flowtemp, offset 2 as heating valve and offset 3 as dhw valve command. The 1A command has to be send every minute, otherwise the boiler resets the values.  Could be the same with this 0x2E0. (T190 could be a periodic sending from thermostat with 26°C flowtemp).

BTW: @proddy I've not expected that the custom entities are so helpfull :-). 4 new entities without a single testversion-compile!

Rich456tf, 2023-04-19:
Ok so I've tried setting 0x02E0 offset 1 as a custom entity and when I have my Sense II connected up I can adjust the flow temperature correctly via 0x02E0 offset 1. But as you mentioned this needs to be sent every minute otherwise the thermostat resets the value.

However without the Sense II connected up I can't seem to get it to adjust the flow temperature via  0x02E0 offset 1.

Attached are 2 logs, one with Sense II connected and the other without. In both logs the starting flow temperature setpoint is 70°C and I try to reduce it down to 55°C using 0x02E0 offset 1.
In the log with Sense II T1112 is the 0x02E0 offset 1 command. 
In the log without Sense II T440 is the 0x02E0 offset 1 command.

Can you see any reason why I can only reduce flow temperature when the Sense II is connected up?

[log without Sense II.txt](https://github.com/emsesp/EMS-ESP32/files/11272348/log.without.Sense.II.txt)
[log with Sense II.txt](https://github.com/emsesp/EMS-ESP32/files/11272349/log.with.Sense.II.txt)

MichaelDvP, 2023-04-19:
The 2E0 is readable, that makes it a bit easier. Without sense II the boiler starts with the last set value:
`000+00:01:00.961 TRACE 96: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x02E0), data: 37 64 (offset 1)`
but resets after some time both, flowtemp and valve opening to zero:
`000+00:04:00.938 TRACE 256: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x02E0), data: 00 00 (offset 1)`
when writing the flowtemp the boiler sets the flowtemp to 37, but the valve keeps closed:
`000+00:07:01.635 TRACE 461: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x02E0), data: 37 00 (offset 1)`
Try to also set the valve with offset 2 and value 100. 
Or try to set both values in USHORT as flowtemp * 256 + 100.

BTW: to trigger the value periodic, we have the scheduler. 
- create the custom entitiy `flowtemp` as 0x02E0, offset 1
- create a schedule, type timer, time 00:01 with command `custom/flowtemp` , value '55', name ' setflowtemp' 
- send new value to api/mqtt command : `scheduler/setflowtemp` and the scheduler repeats the setting every minute.

proddy, 2023-07-03:
@Rich456tf any followup or can we close this?

closed 2023-08-31 by proddy