emsesp/EMS-ESP32 #1161: Problem mit Buderus AM200 in Kodierung 10

Frank1451, 2023-04-21:
### PROBLEM DESCRIPTION

The Buderus AM200 was switched from code 1 to code 10, after that no more parameters were indicated, even after restarting the EMS-ESP.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.0-dev.9",
    "platform": "ESP32",
    "uptime": "001+04:26:41.397",
    "uptime (seconds)": 102401,
    "free mem": 128,
    "max alloc": 77,
    "free app": 6147,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -77,
    "IPv4 address": "192.168.178.118/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 32426,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 153284,
    "bus reads (tx)": 25630,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 1,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 19
  },
  "Devices": [
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "18.06",
      "entities": 47,
      "handlers received": "0x06 0x02BA 0x02B0 0x029C 0x02BB 0x02B1 0x029D 0x02BC 0x02B2 0x029E 0x02BD 0x029F 0x02BE 0x02A0 0x02BF 0x02A1 0x02C0 0x02A2 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x0472 0x02A7 0x0473 0x02A8 0x0474 0x02A9 0x02B3 0x0475 0x02AA 0x02B4 0x0476 0x02AB 0x02B5 0x0477 0x02AC 0x02B6 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xE6 0x0236 0xF7 0x0358 0xF6 0x033B 0x02F6 0x0291 0x0292 0x0293 0x0294 0x02E0 0x02EA 0x054C 0x0313 0x035A 0x0380 0x0295 0x0296 0x0297 0x0298 0x0426 0x0427 0x0428 0x0269 0x026B 0x026A 0x026D 0x026C 0x0422 0x043F 0x0440 0x0424 0x0441 0x0421 0x0442 0x02C3 0x0423 0x0449 0x0425 0x0683 0x068D 0x02FF 0x03"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x28",
      "product id": 160,
      "version": "24.05",
      "entities": 8,
      "handlers received": "0x0331",
      "handlers fetched": "0x0313",
      "handlers ignored": "0xF6 0xD4 0x0255 0x03C7 0xF7 0xF9 0xBF"
    },
    {
      "type": "solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "21.04",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E",
      "handlers pending": "0x035D 0x0391 0x035F 0x035C 0x0361",
      "handlers ignored": "0xF7 0x035B 0x035E 0x0360 0x0368 0x038A 0x038D 0xF6 0x0359 0x03DA 0xE8 0x0345 0x038B 0x038C 0xF9 0xBF"
    },
    {
      "type": "heatsource",
      "name": "AM200",
      "device id": "0x08",
      "product id": 228,
      "version": "03.03",
      "entities": 0,
      "handlers ignored": "0x04 0xE6 0xD1 0xEA 0xE4 0xE9 0xE3 0xF7 0xF9 0xF6 0x02CC 0x02D6 0xEF 0x054D 0x054E 0x0550 0x054C 0xBF"
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2023-04-21:
The AM200 uses the boiler device-id and sends out some boiler telegrams 0xE3... but also some AHS telegrams 0x054D...
Plese make a full log of a few minutes to see if we can register as normal boiler or need some special additions.
What configuraton are us using? What sensors/valves are connected?

Frank1451, 2023-04-22:
I'm not sure if I created the telegram correctly, is that done via the WebGUI with SystemLog?

 

The AM200 is connected to the MM100 (for hot water) and the SM100 (for solar). An RC310 is connected as a display. The AM200 controls a pellet stove, with modulation, no other heat sources. Code 10 on the AM200 stands for autonomous operation.

 

000+17:55:20.855 TRACE 366: [telegram] Rx: 30 00 FF 00 02 68 0C 00 1E
000+17:55:21.068 TRACE 367: [telegram] Rx: 30 00 FF 00 02 6A 03 03 03 00 03 03 03 03 03 00 04 03 EB
000+17:55:25.512 TRACE 368: [telegram] Rx: 08 00 E3 00 00 00 00 00 00 00 00 00 00 00 00 01 94 00 00 50 00 00 00 00 33
000+17:55:25.730 TRACE 369: [telegram] Rx: 08 00 E4 00 00 00 00 00 00 00 00 01 94 64 00 00 00 00 00 80 00 01 8C 00 00 00 00 01 94 01 94 4A
000+17:55:26.005 TRACE 370: [telegram] Rx: 08 00 E4 1B 00 00 00 00 80 00 00 00 00 7D
000+17:55:26.208 TRACE 371: [telegram] Rx: 08 00 E9 00 0F 02 48 02 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F E7
000+17:55:26.384 TRACE 372: [telegram] Rx: 08 00 FF 00 01 D6 00 F4
000+17:55:26.603 TRACE 373: [telegram] Rx: 08 00 FF 00 04 4D 01 94 01 8C 80 00 02 56 80 00 02 48 80 00 02 1F 00 00 00 80 0A 6C
000+17:55:26.795 TRACE 374: [telegram] Rx: 08 00 FF 00 04 4E 00 00 00 00 32 00 1C
000+17:55:27.011 TRACE 375: [telegram] Rx: 08 00 FF 00 04 50 00 06 FF 00 00 FF 0D 00 01 00 00 00 00 00 03 01 00 03 00 00 17 D4 02 62 1E A4
000+17:55:27.242 TRACE 376: [telegram] Rx: 08 00 FF 19 04 50 00 FF FF FF 00 7C
000+17:55:27.242 TRACE 376: [telegram] Rx: 08 00 FF 19 04 50 00 FF FF FF 00 7C
000+17:55:33.385 TRACE 377: [telegram] Rx: 10 00 FF 00 01 67 00 00 AB
000+17:55:33.635 TRACE 378: [telegram] Rx: 10 00 FF 00 01 A5 00 FE 21 24 00 00 24 00 02 ED 03 03 01 02 ED 00 D3 00 00 11 01 03 09 EE 00 7B
000+17:55:34.090 TRACE 379: [telegram] Rx: 10 00 FF 19 01 A5 06 07 FF E2 00 00 FF 00 26 00 3C 01 FF 01 02 D7
000+17:55:34.289 TRACE 380: [telegram] Rx: 10 08 FF 00 01 E0 01 00 00 00 01 D4
000+17:55:34.328 TRACE 381: [telegram] Rx: 10 08 FF 00 01 EA 00 C3
000+17:55:34.367 TRACE 382: [telegram] Rx: 10 00 FF 00 02 1D 00 00 08 07 06
000+17:55:34.733 TRACE 383: [telegram] Rx: 10 28 FF 00 02 3B 00 00 00 40
000+17:55:35.529 TRACE 384: [telegram] Rx: 08 00 E3 00 00 00 00 00 00 00 00 00 00 00 00 01 94 00 00 50 00 00 00 00 33
000+17:55:35.748 TRACE 385: [telegram] Rx: 08 00 E4 00 00 00 00 00 00 00 00 01 94 64 00 00 00 00 00 80 00 01 8C 00 00 00 00 01 94 01 94 4A
000+17:55:35.973 TRACE 386: [telegram] Rx: 08 00 E4 1B 00 00 00 00 80 00 00 00 00 7D
000+17:55:36.176 TRACE 387: [telegram] Rx: 08 00 E9 00 0F 02 48 02 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F E7
000+17:55:38.326 TRACE 388: [telegram] Rx: 10 00 06 00 17 04 09 16 1F 1C 05 01 10 FF 00 76
000+17:55:44.493 TRACE 389: [telegram] Rx: 28 08 D4 00 00 00 00 16
000+17:55:44.525 TRACE 390: [telegram] Rx: 28 00 FF 00 01 55 00 23
000+17:55:44.746 TRACE 391: [telegram] Rx: 28 00 FF 00 02 31 01 D2 00 0F 00 0F 3C 46 00 03 03 00 3C 35
000+17:55:45.518 TRACE 392: [telegram] Rx: 08 00 E3 00 00 00 00 00 00 00 00 00 00 00 00 01 94 00 00 50 00 00 00 00 33
000+17:55:45.736 TRACE 393: [telegram] Rx: 08 00 E4 00 00 00 00 00 00 00 00 01 94 64 00 00 00 00 00 80 00 01 8C 00 00 00 00 01 94 01 94 4A
000+17:55:46.010 TRACE 394: [telegram] Rx: 08 00 E4 1B 00 00 00 00 80 00 00 00 00 7D
000+17:55:46.214 TRACE 395: [telegram] Rx: 08 00 E9 00 0F 02 48 02 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F E7
000+17:55:47.690 TRACE 396: [telegram] Rx: 08 00 D1 00 00 80 47
000+17:55:49.034 TRACE 397: [telegram] Rx: 30 00 FF 00 02 66 01 62 00 12 73
000+17:55:55.586 TRACE 398: [telegram] Rx: 08 00 E3 00 00 00 00 00 00 00 00 00 00 00 00 01 94 00 00 50 00 00 00 00 33
000+17:55:55.804 TRACE 399: [telegram] Rx: 08 00 E4 00 00 00 00 00 00 00 00 01 94 64 00 00 00 00 00 80 00 01 8C 00 00 00 00 01 94 01 94 4A
000+17:55:55.998 TRACE 400: [telegram] Rx: 08 00 E4 1B 00 00 00 00 80 00 00 00 00 7D
000+17:55:56.202 TRACE 401: [telegram] Rx: 08 00 E9 00 0F 02 48 02 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F E7
000+17:56:01.562 TRACE 402: [telegram] Rx: 0B 90 FF 00 19 01 A5 FD
000+17:56:01.646 TRACE 403: [telegram] Rx: 10 0B FF 00 01 A5 00 FE 21 24 00 00 24 00 02 ED 03 03 01 02 ED 00 D3 00 00 11 01 03 09 EE 00 85
000+17:56:01.901 TRACE 404: [telegram] Rx: 0B 90 FF 19 19 01 A5 35
000+17:56:01.965 TRACE 405: [telegram] Rx: 10 0B FF 19 01 A5 06 07 FF E2 00 00 FF 00 26 00 3C 01 FF 01 02 07
000+17:56:02.172 TRACE 406: [telegram] Rx: 0B 90 FF 00 19 01 B9 E1
000+17:56:02.245 TRACE 407: [telegram] Rx: 10 0B FF 00 01 B9 FF 2E 2A 26 1E 01 00 FF FF F6 2A 01 E1 20 01 0F 05 00 00 02 22 14
000+17:56:02.476 TRACE 408: [telegram] Rx: 08 00 07 00 09 01 00 00 01 01 00 00 00 00 00 00 00 00 00 89
000+17:56:02.672 TRACE 409: [telegram] Rx: 0B 90 FF 00 19 01 AF F7
000+17:56:02.723 TRACE 410: [telegram] Rx: 10 0B FF 00 01 AF 03 28 00 00 23 2D 0F 01 19 1E 00 05
000+17:56:02.982 TRACE 411: [telegram] Rx: 0B 90 FF 00 19 01 9B C3
000+17:56:03.031 TRACE 412: [telegram] Rx: 10 0B FF 00 01 9B 01 01 00 00 00 02 05 30 26 D4
000+17:56:03.232 TRACE 413: [telegram] Rx: 0B 90 FF 00 19 01 CC 94
000+17:56:03.260 TRACE 414: [telegram] Rx: 10 0B FF 03 01 CC 01 F6
000+17:56:03.483 TRACE 415: [telegram] Rx: 0B 90 FF 00 19 01 F5 AD
000+17:56:03.557 TRACE 416: [telegram] Rx: 10 0B FF 00 01 F5 02 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01 AA
000+17:56:03.823 TRACE 417: [telegram] Rx: 0B 90 FF 00 19 01 3A 62
[... cut ...]
000+18:03:06.405 TRACE 1027: [telegram] Rx: 08 00 E4 1B 00 00 00 00 80 00 00 00 00 7D
000+18:03:06.609 TRACE 1028: [telegram] Rx: 08 00 E9 00 0F 02 47 02 47 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F 18
000+18:03:10.431 TRACE 1029: [telegram] Rx: 10 00 FF 0D 01 A5 02 E5 00 DB 0D
000+18:03:13.575 TRACE 1030: [telegram] Rx: 30 28 FF 00 02 45 64 01 01 1A
000+18:03:13.671 TRACE 1031: [telegram] Rx: 30 00 FF 00 02 62 01 8F 01 07 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00

Frank1451, 2023-04-22:
Part 2:

 

000+18:09:49.675 WARNING 1517: [telegram] Incomplete Rx: B0 00 30 00
000+18:09:49.858 TRACE 1518: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xD1), data: 00 84
000+18:09:50.204 TRACE 1519: [emsesp] thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 17 04 09 16 2D 28 05 01 10 FF 00
000+18:09:56.204 TRACE 1520: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE3), data: 00 00 00 00 00 00 00 00 00 00 00 01 89 00 00 50 00 00 00 00
000+18:09:56.421 TRACE 1521: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE4), data: 00 00 00 00 00 00 00 01 89 64 00 00 00 00 00 80 00 01 80 00 00 00 00 01 89 01 89
000+18:09:56.676 TRACE 1522: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE4), data: 00 00 00 00 80 00 00 00 00 (offset 27)
000+18:09:56.880 TRACE 1523: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE9), data: 0F 02 46 02 46 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F
000+18:10:01.429 TRACE 1524: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19
000+18:10:01.516 TRACE 1525: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 FE 21 24 00 00 24 00 02 DF 03 03 01 02 DF 00 E1 00 00 11 01 03 09 EE 00
000+18:10:01.779 TRACE 1526: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 0x19 (offset 25)
000+18:10:01.843 TRACE 1527: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 06 07 FF E2 00 00 FF 00 26 00 3C 01 FF 01 02 (offset 25)
000+18:10:02.049 TRACE 1528: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Set(0x02B9), length: 0x19
000+18:10:02.122 TRACE 1529: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: FF 2E 2A 26 1E 01 00 FF FF F6 2A 01 E1 20 01 0F 05 00 00 02 22
000+18:10:02.360 TRACE 1530: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Summer(0x02AF), length: 0x19
000+18:10:02.409 TRACE 1531: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Summer(0x02AF), data: 03 28 00 00 23 2D 0F 01 19 1E 00
000+18:10:02.659 TRACE 1532: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Curves(0x029B), length: 0x19
000+18:10:02.708 TRACE 1533: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Curves(0x029B), data: 01 01 00 00 00 02 05 30 26
000+18:10:02.909 TRACE 1534: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Set2(0x02CC), length: 0x19
000+18:10:02.937 TRACE 1535: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set2(0x02CC), data: 01 (offset 3)
000+18:10:03.160 TRACE 1536: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode(0x02F5), length: 0x19
000+18:10:03.234 TRACE 1537: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 02 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
000+18:10:03.500 TRACE 1538: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300OutdoorTemp(0x023A), length: 0x19
000+18:10:03.532 TRACE 1539: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 00 75
000+18:10:03.720 TRACE 1540: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 0x19
000+18:10:03.792 TRACE 1541: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 32 00 01 00 00 00 00 00 FF 01 F6 06 FF FF 00 00 00 00 00 FF
000+18:10:04.021 TRACE 1542: [emsesp] Me(0x0B) -R-> mixer(0x28), MMPLUSConfigMessage_WWC(0x0313), length: 0x19
000+18:10:04.076 TRACE 1543: [emsesp] mixer(0x28) -W-> Me(0x0B), MMPLUSConfigMessage_WWC(0x0313), data: 04 FF 3C 3C 32 2D 0A F6 12 46 3C 0A
000+18:10:04.341 TRACE 1544: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19
000+18:10:04.422 TRACE 1545: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF 00 FF 02 00
000+18:10:04.621 TRACE 1546: [emsesp] Me(0x0B) -R-> solar(0x30), SM100SystemConfig(0x0358), length: 0x19 (offset 24)
000+18:10:04.646 TRACE 1547: [emsesp] solar(0x30) -W-> Me(0x0B), SM100SystemConfig(0x0358), data: <empty> (offset 24)
000+18:10:04.882 TRACE 1548: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CircuitConfig(0x035A), length: 0x19
000+18:10:04.941 TRACE 1549: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CircuitConfig(0x035A), data: 78 05 00 5A 14 01 01 32 64 00 00 00 5A 0C
000+18:10:05.201 TRACE 1550: [emsesp] Me(0x0B) -R-> solar(0x30), SM100CollectorConfig(0x0380), length: 0x19
000+18:10:05.247 TRACE 1551: [emsesp] solar(0x30) -W-> Me(0x0B), SM100CollectorConfig(0x0380), data: 5A 64 00 00 24 02 00 00 01
000+18:10:05.451 TRACE 1552: [emsesp] Me(0x0B) -R-> solar(0x30), SM100Energy(0x038E), length: 0x19
000+18:10:05.503 TRACE 1553: [emsesp] solar(0x30) -W-> Me(0x0B), SM100Energy(0x038E), data: 00 00 03 FC 00 00 00 66 00 00 69 70
000+18:10:05.855 TRACE 1554: [emsesp] heatsource(0x08) -B-> All(0x00), UBADevices(0x07), data: 09 01 00 00 01 01 00 00 00 00 00 00 00 00 00
000+18:10:06.221 TRACE 1555: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE3), data: 00 00 00 00 00 00 00 00 00 00 00 01 89 00 00 50 00 00 00 00
000+18:10:06.439 TRACE 1556: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE4), data: 00 00 00 00 00 00 00 01 89 64 00 00 00 00 00 80 00 01 80 00 00 00 00 01 89 01 89
000+18:10:06.674 TRACE 1557: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE4), data: 00 00 00 00 80 00 00 00 00 (offset 27)
000+18:10:06.877 TRACE 1558: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0xE9), data: 0F 02 46 02 46 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0F
000+18:10:08.148 TRACE 1559: [emsesp] solar(0x30) -W-> mixer(0x28), ?(0x0345), data: 64 01 01
[... cut ...]
000+18:15:27.593 TRACE 2009: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0x054D), data: 01 83 01 7A 80 00 02 4B 80 00 02 45 80 00 02 1E 00 00 00 87 0A
000+18:15:27.785 TRACE 2010: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0x054E), data: 00 00 00 00 32 00
000+18:15:28.000 TRACE 2011: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0x0550), data: 00 06 FF 00 00 FF 0D 00 01 00 00 00 00 00 03 01 00 03 00 00 17 D4 02 62 1E
000+18:15:28.192 TRACE 2012: [emsesp] heatsource(0x08) -B-> All(0x00), ?(0x0550), data: 00 FF FF FF 00 (offset 25)
000+18:15:34.164 TRACE 2013: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Flo

Frank1451, 2023-04-22:
AM200 Funktionsmodul für alternative Wärmeerzeuger

MM100 (Adr.: 9)Warmwassermodul (bei Anschluss eines Warm-

wasserspeichers)

 

SM100  Solarmodul (bei Anschluss einer Solaranlage)

 

T1          Außentemperaturfühler

TA1       Vorlauftemperaturfühler alternativer Wärmeerzeuger

TB1       Pufferspeicher-Temperaturfühler oben

TB3       Pufferspeicher-Temperaturfühler unten

TB4       Vorlauftemperaturfühler 3-Wege-Mischventil VB1

TC1       Warmwasser-Temperaturfühler (am Warmwasserspeicher)

TC1       Temperaturfühler Heizkreis

TR2       Rücklauftemperaturfühler Heizkreis

TS1        Temperaturfühler Kollektorfeld

TS2        Warmwasserspeicher-Temperaturfühler unten

VC1       3-Wege-Mischventil (bei gemischtem Heizkreis)

VR2       Heizkreispumpe (bei ungemischtem Heizkreis)

VB1       3-Wege-Mischventil Pufferspeicher

Wodtke waterPelletofen mit integriertem Wasser-Wärmetauscher

Wodtke PS04Speicherladestation

Frank1451, 2023-04-22:
[image]

MichaelDvP, 2023-04-22:
Thanks. As expected there are boiler telegrams and AM200 telegrams, but most of the values are published in both telegrams, like flowtemp, rettemp,...
We have to check if it more usefull to map as boiler and maybe add some values or as heatsource. The setting telegrams are not broadcasted, we have to read them.
Could you please open a telnet and post the output from these commands:
`read 08 E6`
`read 08 EA`
`read 08 54C`

Frank1451, 2023-04-22:
ems-esp:$ read 08 E6
000+22:57:37.369 I 3: [emsesp] heatsource(0x08) -W-> Me(0x0B), ?(0xE6), data: 01 50 00 00 64 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 00 00
000+22:57:37.637 I 4: [emsesp] heatsource(0x08) -W-> Me(0x0B), ?(0xE6), data: <empty> (offset 26)

ems-esp:$ read 08 EA
000+22:58:06.842 I 5: [emsesp] heatsource(0x08) -W-> Me(0x0B), ?(0xEA), data: 01 3C 00 00 00 00 00 00 00 00 00 3C 00 2D 00 3C 00 00 00 00 00 00 0A (offset 5)

ems-esp:$ read 08 54C
000+22:58:24.255 I 6: [emsesp] heatsource(0x08) -W-> Me(0x0B), ?(0x054C), data: 02 00 01 00 00 00 3C 3C 00 78 00 3C 01 01 05 1E 01 00 5A 00 01 23 01 64 23
000+22:58:24.481 I 7: [emsesp] heatsource(0x08) -W-> Me(0x0B), ?(0x054C), data: 0F 05 (offset 25)

Frank1451, 2023-04-22:
maybe that will help too.
These were the data fields for coding 1:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Frank/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Frank/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Entity Name | value | UoM
-- | -- | --
AHQ1 System Vorlauftemperatur | 19,1 | Â°C
AHQ1 System RÃ¼cklauftemperatur | 19,2 | Â°C
AHQ1 Alternativer WE Vorlauftemperatur | 21 | Â°C
AHQ1 Speichertemperatur Oben | 40,7 | Â°C
AHQ1 Speichertemperatur Unten | 31,7 | Â°C
AHQ1 Puffer-Ventil | 0 | %
AHQ1 RÃ¼ckfluss-Ventil | 50 | %
AHQ1 Alternativer WE Pumpenmodulation | 0 | %
AHQ1 VR2 Konfiguration | bypass |  
AHQ1 Alt. WÃ¤rmeerzeuger aktiviert | an |  
AHQ1 Konfig. Hauptpumpe | aus |  
AHQ1 Ausgang Pumpe PR1 | pwm |  
AHQ1 Minimale Pumpenansteuerung | 35 | %
AHQ1 RÃ¼cklauf TemperaturerhÃ¶hung | aus |  
AHQ1 Soll-RÃ¼cklauftemperatur | 60 | Â°C
AHQ1 Mischer-Laufzeit | 120 | seconds
AHQ1 Sollwert Vorlauftemperatur | 65 | Â°C
AHQ1 Puffer-Bypass Konfig. | mixer |  
AHQ1 Speicher-Mischer-Laufzeit | 60 | seconds
AHQ1 Konfig. Warmwasserspeicher | monovalent |  
AHQ1 Konfig. Sperr-Modus | aus |  
AHQ1 Konfig. Sperrterminal | n_o |  
AHQ1 Hysterese Sperrmodus | 5 | Â°C
AHQ1 Wartezeit Kessel-Freigabe | 30 | minutes
AHQ1 Brenner | aus |  
AHQ1 Alternativer WE Pumpe | aus |  
AHQ1 WÃ¤rmeanforderung | 0 | %
AHQ1 verbleibende Blockzeit | 0 | minutes
AHQ1 verbleibende Blockzeit WW | 0 | minutes

</body>

</html>

MichaelDvP, 2023-04-22:
At first look it seems better to map as heatsource with the AM200 telegram. I'll need some time to check all values. For now i've updated my dev build to read/show the known AM200 values from switch position 1 also for switch 10. Try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Frank1451, 2023-04-22:
Great, that works very well - thanks!

When I can still support, please let me know.

closed 2023-07-03 by proddy