emsesp/EMS-ESP32 #1163: Wocester Comfort RF Thermostat

richardprior, 2023-04-23:
I have a BBQKees Gateway S32 running the latest dev firmware connected to a Worcester Greenstar 25Si Boiler with a Worcester Comfort Rf remote thermostat. The boiler side appears to work fine but the thermostat isn't recognised.

I've attached the output of /api/system and the EMS watch logs when changing the value on the thermostat. Is it possible to add this thermostat?

[system-info.txt](https://github.com/emsesp/EMS-ESP32/files/11305025/system-info.txt)
[ems-outputs.md](https://github.com/emsesp/EMS-ESP32/files/11305020/ems-outputs.md)

MichaelDvP, 2023-04-25:
Please try with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Sadly the RF thermostats are not writeable, they don't have a setting telegram. We have seen this for Comfort II RF and your log also shows after change only the publish of the monitor-telegram, which is not writeable.

richardprior, 2023-04-25:
That works perfectly, thank you.

I'm trying to implement call-for-heat with my TRVs, do you have any insights as to how this _may_ be possible with this setup or does the read-only nature of the thermostat make this an impossibility?

richardprior, 2023-04-25:
OK, no worries - thanks for your help!

closed 2023-04-25 by richardprior