emsesp/EMS-ESP32 #1164: Alle WW-Werte sind statisch bei Buderus Logamatic 4000

reindeer-web, 2023-04-27:
Ich bekomme keine Aktualisierungen bei den Warmwasser-Werten.
Buderus Heizkessel GB 162
Verwendete Heizungs-Regelung: Buderus Logamatic 4000
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/11341766/emsesp_info.txt)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/11341767/log.txt)

MichaelDvP, 2023-04-27:
The Logamatic 4000 seems to be a modular system with different modules. I think the device 0x0C  on the bus is one of this modules, Without a product id and reading any single value from the monitor 0x18 is very strange behavior. Also device-id 0x0C is normally the cascade. What modules do you have installed? Is there anything other connected to the bus?

The boiler reports the dhw settings in 0x33 and dhw values in 0x34. Both telegrams are recognized by ems-esp.
```
2023-04-27 11:05:31.235 TRACE 7652: [emsesp] boiler(0x08) -W-> connect(0x0C), UBAParameterWW(0x33), data: 00 FF 3C FB 00 28 00 02 46 00 FF
2023-04-27 11:05:31.340 TRACE 7654: [emsesp] boiler(0x08) -W-> connect(0x0C), UBAMonitorWW(0x34), data: 3C 80 00 80 00 01 02 00 03 00 01 28 30 00 25 A0
```
But 0x33 offset 0: 00 means that there is not dhw on the bus and 0x34 offset 1: 8000 means there is not temperature sensor. Seems the dhw system is deactivated or not present on the bus.

reindeer-web, 2023-04-28:
many thanks for the quick response

These devices are connected to the bus:
- boiler GB162
- Logamatic 412x with ZM424, FM455 and FM442
- Logamatic 412x with FM442
- web KM300

ZM424 is for boiler + 2 heating circuits + 1 warm water
FM442 is for 2 heating circuits
FM455 is for KSE 1 boiler (FM456 would be for 2 and FM457 for 4 boilers)

Please let me know if you need further informations.

MichaelDvP, 2023-04-28:
I've searched for the 4000 system. On https://buderus-de-de.boschtt-documents.com/download/pdf/file/7747014856.pdf page 34 the ZM424 have all connections and temperature sesnors. The 4000 system communicates vis ECOCAN-Bus, EMS is only used for the boiler, but boiler control functions and sensors are not used. I think the parameters and sensorvalues are not on the ems bus and can not be read with ems-esp. 
The KM300 have ems and eco-bus connections, i guess it is connected via ecobus, because the log does not show a KM300 on ems.

reindeer-web, 2023-05-01:
Thanks for the information.

3 of 24 objects for the hot water are updated in my system:
- ems: dhw charging
- ems: dhw starts
- ems: dhw active time

proddy, 2023-07-03:
I don't think there are more EMS messages we can support for this 4000. @MichaelDvP close it?

MichaelDvP, 2023-07-03:
Yes, seems most information are only on the CAN-bus, in EMS log there is nothing we can add.

proddy, 2023-07-03:
ok - will close.

closed 2023-07-03 by proddy