emsesp/EMS-ESP32 #1167: Feature request: button to turn off Ethernet in E32

bbqkees, 2023-05-03:
The E32 and some other boards use a GPIO pin (16 for E32) to turn on (the oscillator of) the Ethernet chip.
Lot's of E32 owners use only WiFi, so in those cases it can be beneficial to turn off Ethernet to save some of the precious RAM.

Currently you can turn it off by selecting a custom board profile and deselect the Ethernet NIC.
An on/off or checkmark button would be more convenient here.

proddy, 2023-05-03:
when EMS-ESP starts up it will always load the Eth library and tries to connect (costing RAM) to see if the ethernet is connected, so plug&play with an E32 works first time. But with this addition, it'll save memory so a good thing to implement.

MichaelDvP, 2023-05-05:
We disable Wifi by leaving ssid blank, why not use the same logic?
Enable eth is wifi-ssid is blank and wifi is disabled, disable eth if wifi is enabled.
The network status page already shows only one connection and two active connections cost a lot of ram.
This is only a text change: `NETWORK_BLANK_SSID: 'leave blank to disable WiFi and enable ETH',` and some lines in `system::netowrk_init()`
```
    bool disableEth;
    EMSESP::esp8266React.getNetworkSettingsService()->read([&](NetworkSettings & settings) {
        disableEth = settings.ssid.length() > 0;
    });

    // no ethernet present or disabled
    if (phy_type_ == PHY_type::PHY_TYPE_NONE || disableEth) {
        return;
    }
```

I think most users will not recognize an extra checkbox and leave it on default.

proddy, 2023-05-05:
that will work, and prevent the costly ETH.begin() from being called in `WebSettings::update()`

MichaelDvP, 2023-05-05:
Hmm, the ETH.begin in WebSettings is only called if load_board_profile failed: once on the first start. Then the board profile is set to E32 or S32 and this fallback is never executed again (only after factory reset). With board profile set the ETH.begin is only called in system::network_init., where the above ssid check prevents it.

proddy, 2023-05-27:
this is in dev12

closed 2023-05-27 by proddy