emsesp/EMS-ESP32 #1168: Need fresh test data

proddy, 2023-05-05:
@MichaelDvP I need fresh test data to test the WebUI and it's new functionality. Today I'm using my old setup from years ago with just 2 devices and it'll be good to get some more EMS devices and data in, like the Solar and Mixer from your setup:

[image]

What you'll need to do is grab the output from `/rest/deviceData` endpoint for each of those id's. It will be compressed, so either modify the code to not use `MsgpackAsyncJsonResponse` in `WebDataService::device_data()` or just send me the binary files capturing direct from the browsers dev window (after a ctrl-r in Network tab). Or you can use a script like on Linux/Windows WSL:
```sh
curl --output - -X POST -H "Content-Type: application/json" -d '{"id": 1}' http://ems-esp.local/rest/deviceData | msgpack2json -d
```

Then I can add to `server.js` and continue testing the new web changes.

thanks

MichaelDvP, 2023-05-05:
[Devices.zip](https://github.com/emsesp/EMS-ESP32/files/11404203/Devices.zip)

proddy, 2023-05-06:
Thanks Michael. I've updated the test data and added to my open PR

closed 2023-05-06 by proddy