emsesp/EMS-ESP32 #1171: MS100 is shown as SM100, as a result all states are shown under the solar tree in the IOBroker

Helmut1965, 2023-05-08:
EMS-ESP Version v3.5.1
Device (Platform / SDK) ESP32 / v4.4.2

Current behavior:
Device:  Solar Module | SM100/MS100 | Device ID 0x2A | Product ID 163 | Version 27.05
Device:  Solar Module | SM100/MS100 | Device ID 0x30 | Product ID 163 | Version 27.05

Expected behavior:
Device:  HotWater Module | MS100 | Device ID 0x2A | Product ID 163 | Version 27.05
Device:  Solar Module | SM100 | Device ID 0x30 | Product ID 163 | Version 27.05

Most entities below the MS100 starts with a DHW, this can maybe used to create this as the right devices

MichaelDvP, 2023-05-08:
EMS-ESP maps the devices by product-id. But you are using the ioBroker adapter, i think with KM200 data structure. Than it is better to ask here: https://github.com/tp1de/ioBroker.ems-esp/issues, @tp1de can map the dhw circuit of the solar module to dhwCircuits as he maps the mixers and boilers with dhw here: https://github.com/tp1de/ioBroker.ems-esp/blob/a1f961b5d4b70792dff421678314285b23e935e4/lib/ems.js#L365-L369

Helmut1965, 2023-05-08:
@MichaelDvP, thanks for the fast reply. Yes, I saw that both devices has the same ID, this is really crazy. I'm not using the KM200 structure, but for the Hotwater Devices it's maybe better.

I also think there is a bug in the entity "dhw max power". In case I remember it right, this is the max power for the warm up the water storage. I can change what ever I want, this not reflected by the entity "dhw max power" and also not buy one of the other entities. I will do some further checks before I open a ticket for this one.
Thanks again

Helmut1965, 2023-05-09:
@tp1de 
Hi Thomas, thanks, I'll check.

For Dhw, I'm able to change this on the display to eg. 65% and it works, in case this is helpfull I can record the log or I can search for something special. 
I also steer currently the modulation of the boiler with an external 0-10V signal, is there a way to steer it directly with the esp-ems?
Thanks for all the god work.
Helmut

tp1de, 2023-05-09:
@Helmut1965 
I do have a KB192i gas boiler with a RC310 thermostat. Modulation is controlled by the thermostat for ww and for heating circuits. Don't you have a thermostat to control boiler modulation? Why do you want to overwrite it? The entity boiler.selburnpow is there but setting it to a specific value does not work for my system.
Maximum modulation can be set by boiler.burnmaxpower but this is only active on my system for heating circuits.
For the first 2 minutes after ignition there is 52% modulation fixed.

Helmut1965, 2023-05-09:
@tp1de 
Hi, there are lot of reasons to do so - first if you run a device at the maximum level it breaks earlier - for sure this isn't the driver here.
I've the same device connect to a bigger water storage tank behind. This partly heated up by the solar systems. In case the boiler needs to heat up for the heating system a modulation between 11%-40% is enough, this gives me a maximum of "delta T" in the storage temp which results in the lowest "Ruecklauf" Temperature, this is best for the effiency of the burner and gives you to lowest start of the flame. At winter time I'm able to run 24/7  without burner restart.
For the warm water is the same, why I should go up to 100% when 70% gives me enough an this is better for my efficiency. 
Why I want to steer this, this is also easy, in case the storage temp for the solar system is low it makes sense to work with 70%, in case the storage is high, it can be run with 40% for examaple

tp1de, 2023-05-09:
@Helmut1965 I do understand. Heating can be set by boiler.burnmaxpower. 
For ww using wwflowtempoffset should give a similar result when supported by your thermostat.
Just use a low offset temperature in relation to the target ww temp. Then the boiler uses low power/modulation steered by the thermostat. That's the way I do it. 
You could even consider to give temporarly a higher offset to load ww faster if needed (fastload by script)

proddy, 2023-05-29:
can you trace the EMS messages when you change it in RC310 so we can simulate it?

closed 2023-08-13 by proddy