emsesp/EMS-ESP32 #1172: Logavent HRV176 - Unrecognized EMS device (deviceID 0x51, productID 231)

Timon321, 2023-05-08:
### PROBLEM DESCRIPTION

I have a new device on my EMS-Bus, which is not recognized today running the latest dev-release 3.6.0-dev.9.

The device is the new Buderus Logavent HRV176. 

```lua
2023-05-08 14:06:05.444 I 17: [emsesp] Recognized new boiler with deviceID 0x08
2023-05-08 14:06:06.141 I 18: [emsesp] Recognized new thermostat with deviceID 0x10
2023-05-08 14:06:22.506 N 19: [emsesp] Unrecognized EMS device (deviceID 0x51, productID 231). Please report on GitHub.
2023-05-08 14:06:48.781 I 20: [emsesp] Recognized new mixer with deviceID 0x20`
```

What information is needed to adapt the ventilation system with the EMS-ESP?

[image]

MichaelDvP, 2023-05-08:
Yes, ventilation systems are completly unknown. I think we should make a new device class for this.
Could you please tell what values this system reports and what settings it have.
And use telnet with `watch 51` for some minutes to see the telegrams from/to this device. 
If you change some paramters/settings while logging we could also see the setting telegrams. Please note when and what you have changed and tell it. 
Could you also add the `support info` from help-page.

Timon321, 2023-05-08:
Support-Info:

```YAML
{
  "System Info": {
    "version": "3.6.0-dev.9",
    "platform": "ESP32",
    "uptime": "000+01:22:17.380",
    "uptime (seconds)": 4937,
    "free mem": 122,
    "max alloc": 39,
    "free app": 3,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "IPv4 address": "192.168.178.33/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 2252,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": true
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 15053,
    "bus reads (tx)": 3106,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 6,
    "bus reads failed": 5,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 5
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "03.03",
      "entities": 148,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xF7 0x04A0 0x02D6 0x0489 0x04AC 0x02CD 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "Rego 3000/UI800",
      "device id": "0x10",
      "product id": 253,
      "version": "47.06",
      "entities": 51,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031E",
      "handlers ignored": "0xE7 0x02E1 0x02EB 0x0507 0x0508 0x0509 0x050A 0x056B 0x061E 0x059D 0xF7 0x0589 0x0291 0xBF 0x0593"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0xD2 0xE2 0x0255 0xBF 0x02CD"
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x51",
      "product id": 231,
      "version": "53.02",
      "entities": 0,
      "handlers ignored": "0x0583 0xF9 0x0575 0x056B 0x0585 0x0586 0x0587 0x0588 0x058D 0x058E 0x058F 0x05CF 0x05D9 0x05E3 0xBF 0x0590"
    }
  ]
}
```

I will analyze the rest of the information and submit it by the end of the week at the latest. We have not yet moved into the house.

MichaelDvP, 2023-05-08:
I've looked into the manual on https://buderus-de-de.boschtt-documents.com/download/pdf/file/6721841222.pdf
page 37/38 shows a lot of settings and monitor values. 
Also the `handlers ignored": "0x0583 0xF9 0x0575 0x056B 0x0585 0x0586 0x0587 0x0588 0x058D 0x058E 0x058F 0x05CF 0x05D9 0x05E3 0xBF 0x0590"` shows a lot of telegrams to proced. Some of the thermostats unknown telegrams could be settings for the ventilation.
You use Ethernet gateway with 4MB flash. I think with this addition code will be larger and need repartitioning to fit. Let see what values you need and what we find.

Timon321, 2023-05-10:
Yesterday i had 5 minutes to take a look into the communication.

At first here are the sensor values, i had at recording (maybe they changed a little bit while doing this):

- outdoor fresh air temperature (Außenlufttemperatur) 19.0°C
- indoor fresh air temperatur (Zulufttemperatur) 21.4°C
- indoor exhaust air temperatur (Ablufttemperatur) 21.9°C
- outdoor exhaust air temperature (Fortlufttemperatur) 18.2°C
- voc sensor (Raumluftqualität) 980ppm
- bypass ventil (Beipassklappe) closed
- filter remaining time (Filter-Restlaufzeit) 179 days

I tried to cut it into pieces.

1. Manual Control

```lua
##Selected Level 1##
000+21:26:42.813 N 21: [emsesp] thermostat(0x10) -W-> unknown(0x51), ?(0x056B), data: 01
000+21:26:43.314 N 22: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x056B), data: 01
000+21:26:43.495 N 23: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0575), data: 01 (offset 1)
000+21:26:43.679 N 24: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0585), data: 1E 1E
000+21:26:44.176 N 25: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0575), data: 01
000+21:26:44.741 N 26: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 C7 00 00 00 00 00 C9 (offset 47)
000+21:26:45.220 N 27: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 07 00 06 (offset 30)
000+21:26:46.095 N 28: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 0F 00 11 (offset 30)
000+21:26:46.895 N 29: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 13 00 13 (offset 30)
000+21:26:47.445 N 30: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 13 00 13 (offset 30)
000+21:26:47.628 N 31: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 01 51 (offset 17)
000+21:26:47.816 N 32: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 5D 00 C8 (offset 45)
000+21:26:48.854 N 33: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 04 74 (offset 2)
000+21:26:49.041 N 34: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 0C 00 0B (offset 30)
000+21:26:49.225 N 35: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 04 8A (offset 17)
000+21:26:49.753 N 36: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 04 7A (offset 2)
000+21:26:49.941 N 37: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 0C 00 07 (offset 30)
000+21:26:50.148 N 38: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 04 A4 (offset 17)
000+21:26:50.658 N 39: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 08 00 07 (offset 30)
000+21:26:50.841 N 40: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 5C (offset 45)
000+21:26:51.025 N 41: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 DB (offset 55)
000+21:26:51.503 N 42: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 03 C5 (offset 2)
000+21:26:51.691 N 43: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 06 00 03 (offset 30)
000+21:26:51.874 N 44: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 03 D7 (offset 17)
000+21:26:52.566 N 45: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 03 12 (offset 2)
000+21:26:52.749 N 46: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 05 (offset 30)
000+21:26:52.933 N 47: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 00 (offset 32)
##Selected Level 2##
000+21:26:53.326 N 48: [emsesp] thermostat(0x10) -W-> unknown(0x51), ?(0x056B), data: 02
000+21:26:53.991 N 49: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 02 FA (offset 17)
000+21:26:54.173 N 50: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 45 (offset 40)
000+21:26:54.356 N 51: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 03 (offset 30)
000+21:26:55.441 N 52: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 C6 00 00 00 00 00 C8 (offset 47)
000+21:26:55.625 N 53: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0585), data: 46 46
000+21:26:56.766 N 54: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0575), data: 02 02
000+21:26:56.947 N 55: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x056B), data: 02
000+21:26:57.130 N 56: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 04 21 (offset 2)
000+21:26:58.836 N 57: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 80 00 00 68 00 67 00 0C 00 0A FF FF FF 4A 3D 41 45 00 E5 00 D1 00 5B (offset 24)
000+21:26:59.065 N 58: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 C9 04 D4 FF FF FF 00 DB FF 03 9C 00 00 C7 00 DB 05 46 7F FF 00 80 00
##Selected Level 3##
000+21:26:59.441 N 59: [emsesp] thermostat(0x10) -W-> unknown(0x51), ?(0x0589), data: 02 FF
000+21:26:59.588 N 60: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 5B 00 BD 00 00 00 00 00 C8 00 DB (offset 45)
000+21:26:59.788 N 61: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0575), data: 02 02 46 46 00 00 FF 80 00 01
000+21:27:00.778 N 62: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0585), data: 46 46 00 00 00 77 00 03 F4 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:27:00.959 N 63: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0585), data: 00 (offset 25)
##Selected Level 4##
000+21:27:01.351 N 64: [emsesp] thermostat(0x10) -W-> unknown(0x51), ?(0x056B), data: 03
000+21:27:01.541 N 65: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0586), data: 00 64 64 00 00 00 00 00
000+21:27:01.772 N 66: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0587), data: 00 00 64 00 64 01 00 02 20 01 00 01 00 00 00 46 00 00 00 02 00 B4 00 B4
000+21:27:02.267 N 67: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0588), data: 11 01
[... cut ...]
000+21:33:08.839 N 269: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 65 00 2F (offset 45)
000+21:33:09.022 N 270: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 B2 (offset 53)
000+21:33:11.111 N 271: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 2E (offset 47)
000+21:33:13.885 N 272: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 66 (offset 45)
000+21:33:17.486 N 273: [emsesp] unknown(0x51) -B-> All(0x00), ?(0x0583), data: 00 2D (offset 47)
```

I hope there are some useful informations. I can select more, if you want to.

MichaelDvP, 2023-05-10:
I've added the ventilation device class and the first entities. For 3.6.0-dev9 this exceeds the current partition limit. 
You can repartition with emsesp-loader and +700k for emsesp, or i can compile to an en language only version. But since it's a testing version i have merged proddys changes from PR #1116, which gives 14kB free space in flashand much faster web interface. The breaking changes does not affect you, it's only renaming dallas/analog sensors, you have no connected.
Please check https://github.com/MichaelDvP/EMS-ESP32/releases

MichaelDvP, 2023-05-25:
The decimal point is corrected now 
If the remaining time goes to 178 days, read the 0x583 telegram again, then we can see which data is the right value.

proddy, 2023-05-25:
@MichaelDvP ready to merge your changes back into the dev branch?

MichaelDvP, 2023-05-26:
Yes, i'll open a PR. It contains some work for #1158 (HIU), #1161 (AM200), #1167 (eth/wifi switching) and this ventilation.
These issues are still open, maybe there is some more work to do, but actual state works so far.

proddy, 2023-05-26:
> Yes, i'll open a PR. It contains some work for #1158 (HIU), #1161 (AM200), #1167 (eth/wifi switching) and this ventilation. These issues are still open, maybe there is some more work to do, but actual state works so far.

ok thanks

proddy, 2023-05-27:
merged to dev12

MichaelDvP, 2023-05-27:
@proddy Bypass-valve and filter remaining time are still not included, maybe leave this open until we have all entities included.
@Timon321  You can use the custom entiies to create reads to device 51, type 583 offset 26, valtype ushort and one for offset 28. Then you can see/record  what value changes to 178 days.

Timon321, 2023-06-04:
I just found an option for manually opening and closing the bypass-valve:

`006+00:38:49.148 N 40: [emsesp] thermostat(0x10) -W-> ventilation(0x51), ?(0x055C), data: 01 (offset 1) | OPEN VALVE
006+00:38:49.623 N 41: [emsesp] ventilation(0x51) -B-> All(0x00), Blowerspeed(0x0585), data: 64 (offset 23)
006+00:38:49.804 N 42: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0587), data: 01 (offset 1)
006+00:38:49.985 N 43: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x055C), data: 01 (offset 1)
006+00:38:50.512 N 44: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 11 (offset 30)
006+00:38:50.695 N 45: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 93 (offset 45)
006+00:38:52.149 N 46: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 24 (offset 47)
006+00:38:52.333 N 47: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0D (offset 32)
006+00:38:52.813 N 48: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 12 (offset 30)
006+00:38:52.997 N 49: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 DA (offset 53)
006+00:38:53.500 N 50: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0E (offset 32)
006+00:38:54.548 N 51: [emsesp] thermostat(0x10) -W-> ventilation(0x51), ?(0x055C), data: 00 (offset 1) | CLOSE VALVE
006+00:38:55.073 N 52: [emsesp] ventilation(0x51) -B-> All(0x00), Blowerspeed(0x0585), data: 00 (offset 23)
006+00:38:55.254 N 53: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 1E (offset 40)
006+00:38:55.437 N 54: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0587), data: 01 00
006+00:38:55.910 N 55: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x055C), data: 00 (offset 1)
006+00:38:56.413 N 56: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 DB (offset 53)
006+00:38:56.963 N 57: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0D (offset 32)
006+00:38:57.438 N 58: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 06 14 (offset 2)
006+00:38:58.938 N 59: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 92 (offset 45)
006+00:38:59.335 N 60: [emsesp] thermostat(0x10) -W-> ventilation(0x51), ?(0x055C), data: 01 (offset 1) | OPEN VALVE
006+00:38:59.811 N 61: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x055C), data: 01 (offset 1)
006+00:38:59.994 N 62: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0E (offset 32)
006+00:39:00.176 N 63: [emsesp] ventilation(0x51) -B-> All(0x00), Blowerspeed(0x0585), data: 64 (offset 23)
006+00:39:00.700 N 64: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0587), data: 00 01
006+00:39:00.883 N 65: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 23 (offset 47)
006+00:39:01.975 N 66: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0D (offset 32)
006+00:39:02.450 N 67: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 91 (offset 45)
006+00:39:03.625 N 68: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0E (offset 32)
006+00:39:04.151 N 69: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 E5 (offset 43)
006+00:39:04.497 N 70: [emsesp] thermostat(0x10) -W-> ventilation(0x51), ?(0x055C), data: 00 (offset 1) | CLOSE VALVE
006+00:39:04.998 N 71: [emsesp] ventilation(0x51) -B-> All(0x00), Blowerspeed(0x0585), data: 00 (offset 23)
006+00:39:05.181 N 72: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0587), data: 01 00
006+00:39:05.362 N 73: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x055C), data: 00 (offset 1)
006+00:39:05.863 N 74: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 11 (offset 30)
006+00:39:06.717 N 75: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 12 00 0D (offset 30)
006+00:39:07.588 N 76: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 E4 (offset 43)
006+00:39:08.062 N 77: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 90 (offset 45)
006+00:39:08.938 N 78: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 0E (offset 32)
006+00:39:09.412 N 79: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 22 (offset 47)
006+00:39:10.685 N 80: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0587), data: 00
006+00:39:10.869 N 81: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 DA (offset 53)
006+00:39:16.217 N 82: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 8F 00 21 (offset 45)
006+00:39:18.600 N 83: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 11 (offset 30)
006+00:39:20.613 N 84: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 8E (offset 45)
006+00:39:21.763 N 85: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 20 (offset 47)
`

The temperatures are now showing correct. But the VOC PPM now is wrong. Its showing 121,9 but it is 1219.

Also i have the problem with mqtt in HomeAssistant, that there are all values unavailable in the v3.6.0-dev.12:
[image]
The other devices (boiler, mixer, thermostat) are available.

Now the readings for type 583 with the offsets:
`ems-esp:$ read 51 583 26
006+00:45:19.013 I 90: [emsesp] ventilation(0x51) -W-> Me(0x0B), VentilationMonitor(0x0583), data: 1E 2A 20 00 E9 00 E3 00 6D 00 1F 00 00 00 00 00 D7 00 DC (offset 38)
ems-esp:$ read 51 583 28
006+00:45:26.321 I 91: [emsesp] ventilation(0x51) -W-> Me(0x0B), VentilationMonitor(0x0583), data: 1F 00 E9 00 E3 00 6D 00 1F 00 00 00 00 00 D7 00 DC (offset 40)`

MichaelDvP, 2023-06-04:
Looks like telegram 55C is the command and telegram 587 offset 1 gives the state, in the first logs i see that 587 is repeated broadcasted. I add the bypass with reading from 587 and writing to 55C.
> Now the readings for type 583 with the offsets

Sorry, a `read 583` always gives all offsets. I meant 26/28 dez, the read command uses hex.
Please just use `read 583` and write what is the actual filter remaining time.

I think the voc and HA issues are solved. 
My actual dev uses a different mqtt lib (see #1178), but should work. Please try with the new build.

Timon321, 2023-06-05:
The filter remaining time is 162 days today:

```
ems-esp:$ read 51 583
000+21:30:08.911 I 3: [emsesp] ventilation(0x51) -W-> Me(0x0B), VentilationMonitor(0x0583), data: 00 E6 06 54 FF FF FF 00 E8 FF 03 86 00 00 E9 00 E3 06 F4 7F FF 00 80 00 80
000+21:30:09.700 I 4: [emsesp] ventilation(0x51) -W-> Me(0x0B), VentilationMonitor(0x0583), data: 80 00 00 B4 00 B4 00 12 00 0E FF FF FF 2F 1E 2D 1C 00 F2 00 F3 00 5A 00 33 (offset 24)
```

My HomeAssistant says the following log about the ventilation

```
Logger: homeassistant.components.mqtt.models
Source: components/sensor/__init__.py:583
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 11:58:08 (1888 occurrences)
Last logged: 17:12:38

Exception raised when updating state of sensor.ventilation_in_blower_speed, topic: 'ems-esp/ventilation_data' with payload: b'{"outfresh":23.3,"infresh":22.7,"outexhaust":23.0,"inexhaust":23.2,"ventinspeed":"auto","ventoutspeed":100,"airquality":920}'
Exception raised when updating state of sensor.ventilation_in_blower_speed, topic: 'ems-esp/ventilation_data' with payload: b'{"outfresh":23.3,"infresh":22.7,"outexhaust":23.0,"inexhaust":23.2,"ventinspeed":"auto","ventoutspeed":100,"airquality":902}'
Exception raised when updating state of sensor.ventilation_in_blower_speed, topic: 'ems-esp/ventilation_data' with payload: b'{"outfresh":23.3,"infresh":22.7,"outexhaust":23.0,"inexhaust":23.2,"ventinspeed":"auto","ventoutspeed":100,"airquality":880}'
Exception raised when updating state of sensor.ventilation_in_blower_speed, topic: 'ems-esp/ventilation_data' with payload: b'{"outfresh":23.3,"infresh":22.7,"outexhaust":23.0,"inexhaust":23.1,"ventinspeed":"auto","ventoutspeed":100,"airquality":880}'
Exception raised when updating state of sensor.ventilation_in_blower_speed, topic: 'ems-esp/ventilation_data' with payload: b'{"outfresh":23.3,"infresh":22.7,"outexhaust":23.0,"inexhaust":23.1,"ventinspeed":"auto","ventoutspeed":100,"airquality":903}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: 'auto'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.ventilation_in_blower_speed has device class power_factor, state class measurement unit % and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: auto (<class 'str'>)
```

proddy, 2023-06-24:
closing this as the issue has been resolved with the addition of the new device.

closed 2023-06-24 by proddy