emsesp/EMS-ESP32 #1179: [v.3.5.0 +] Tx writes not repeated until successful

yazz007, 2023-05-29:
### PROBLEM DESCRIPTION

I've upgraded to v.3.5.1 and notice very unreliable behavior for changing temperature on thermostat. Lots of request was 'rejected by host'. Unfortunately logs doesn't say much to me, I can only read and understand what it says, so:
v.3.5.1:
```
2023-05-29 07:05:22.212 D 4465: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 19.5 and id 1 on device 0x18
2023-05-29 07:05:22.385 D 4466: [telegram] Sending write Tx [#184], telegram: 0B 18 FF 0A 01 B9 27
2023-05-29 07:05:22.508 E 4467: [emsesp] Last Tx write rejected by host
2023-05-29 07:08:52.009 D 4638: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 19.5 and id 1 on device 0x18
2023-05-29 07:08:52.200 D 4639: [telegram] Sending write Tx [#228], telegram: 0B 18 FF 0A 01 B9 27
2023-05-29 07:08:52.232 D 4640: [emsesp] Last Tx write successful
2023-05-29 07:08:52.232 D 4641: [telegram] Sending post validate read, type ID 0x2A5 to dest 0x18
2023-05-29 07:26:04.243 D 5588: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 19.0 and id 1 on device 0x18
2023-05-29 07:26:04.309 D 5589: [telegram] Sending write Tx [#229], telegram: 0B 18 FF 0A 01 B9 26
2023-05-29 07:26:04.309 D 5589: [telegram] Sending write Tx [#229], telegram: 0B 18 FF 0A 01 B9 26
2023-05-29 07:26:04.399 E 5590: [emsesp] Last Tx write rejected by host
```

As you can see, Tx writes are sent at the same time, I can assume that the reason is to make sure it will be delivered and executed. But it's not.

Again I upgraded to v.3.6.0 dev 12, besides that DEBUG log doesn't work (seems like INFO) and system was noticeable slower, the log (ALL) shows:
```
2023-05-29 09:14:58.879 T 393: [telegram] Incomplete Rx: 0B 18 FF 0A 01 B9 28 B9
2023-05-29 09:14:58.887 E 394: [emsesp] Last Tx write rejected by host
```

I didn't notice any retry on that log. 

I had to downgrade to v.3.4.4 and logs looks like that:
```
2023-05-29 09:39:42.854 D 92: [command] Calling command 'thermostat/manualtemp' (manual temperature) with value 19.5
2023-05-29 09:39:43.098 D 93: [telegram] Sending write Tx [#113], telegram: 0B 18 FF 0A 01 B9 27
2023-05-29 09:39:43.098 D 94: [emsesp] No telegram type handler found for ID 0x2B9 (src 0x0B)
2023-05-29 09:39:43.135 D 95: [emsesp] Last Tx write successful
2023-05-29 09:39:43.135 D 96: [telegram] Sending post validate read, type ID 0x2B9 to dest 0x18
2023-05-29 09:39:43.135 D 96: [telegram] Sending post validate read, type ID 0x2B9 to dest 0x18

2023-05-29 09:42:26.178 D 288: [command] Calling command 'thermostat/manualtemp' (manual temperature) with value 20
2023-05-29 09:42:26.327 D 289: [telegram] Sending write Tx [#169], telegram: 0B 18 FF 0A 01 B9 28
2023-05-29 09:42:26.414 E 290: [emsesp] Last Tx write rejected by host
2023-05-29 09:42:26.725 D 291: [telegram] Sending write Tx [#170], telegram: 0B 18 FF 0A 01 B9 28
2023-05-29 09:42:26.725 D 292: [emsesp] No telegram type handler found for ID 0x2B9 (src 0x0B)
2023-05-29 09:42:26.763 D 293: [emsesp] Last Tx write successful
2023-05-29 09:42:26.763 D 294: [telegram] Sending post validate read, type ID 0x2B9 to dest 0x18
2023-05-29 09:42:27.190 D 295: [telegram] Sending read Tx [#171], telegram: 0B 98 FF 00 19 01 B9
2023-05-29 09:42:27.246 D 296: [emsesp] Last Tx read successful

2023-05-29 09:43:42.426 D 354: [command] Calling command 'thermostat/seltemp' (selected room temperature) with value 19.0
2023-05-29 09:43:42.759 D 355: [telegram] Sending write Tx [#184], telegram: 0B 18 FF 0A 01 B9 26
2023-05-29 09:43:42.849 E 356: [emsesp] Last Tx write rejected by host
2023-05-29 09:43:43.093 D 357: [telegram] Sending write Tx [#185], telegram: 0B 18 FF 0A 01 B9 26
2023-05-29 09:43:43.186 E 358: [emsesp] Last Tx write rejected by host
2023-05-29 09:43:44.718 D 359: [telegram] Sending write Tx [#186], telegram: 0B 18 FF 0A 01 B9 26
2023-05-29 09:43:44.718 D 360: [emsesp] No telegram type handler found for ID 0x2B9 (src 0x0B)
2023-05-29 09:43:44.751 D 361: [emsesp] Last Tx write successful
2023-05-29 09:43:44.751 D 362: [telegram] Sending post validate read, type ID 0x2A5 to dest 0x18
2023-05-29 09:43:46.845 D 363: [telegram] Sending read Tx [#187], telegram: 0B 98 FF 00 19 01 A5
```
You can see the Tx write is repeated at least twice in queue not at the same time. Again I assume, that it will repeat till Tx write is successful.

Quite strange is that despite Tx writes failed many times, the QUALITY was always 100% and FAIL was 0 with v.3.5.0+

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
[... 136 lines cut ...]
      "version": "03.05",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2023-05-29:
Yes there was a case where writing was not possible, with endless tx-repeats blocking the bus. 
I thought the reject always means that it is not writable at all. Seems i was wrong and retrys are usefull.
Should we change to 3 retrys as for tx-reads?

proddy, 2023-05-29:
I think so yes.

Also remove DEBUG from the Web Log, as debug messages are only shown when the code is compiled with the `EMSESP_DEBUG` flag and specifically for developers (not end users).

proddy, 2023-07-02:
closing this.

commit e4a07991 (2023-05-30): 3 retrys on rejected tx, fixes #1179

please delete branch after merge. My dev branch contains the mqtt lib, so i edit on github directly.
closed 2023-07-02 by proddy