emsesp/EMS-ESP32 #1187: Setting Request: Control of heating circuit mode

MichScha, 2023-06-03:
### PROBLEM DESCRIPTION

To enable cooling as a possible mode for the HP you have to set in the service-menu of the HP the Option "Heating Circuit Function" to one of the following values that it is actually possible to cool.

- Heating, Cooling, Heating / Cooling

I have logged a few attempts to change the entry but couldn´t track them down whats meant by the values.

```
Heizen
001+11:00:06.850 N 489: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 1F (offset 4)
001+11:00:07.039 N 490: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 15 19 00
001+11:00:07.527 N 493: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 25 04
001+11:00:07.711 N 494: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 7B 00
001+11:00:07.894 N 495: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 E0 1F
001+11:00:08.469 N 498: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 0C F6 7F (offset 1)
001+11:00:08.652 N 499: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 9B F3
001+11:00:09.152 N 500: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF6), data: FF 01 A5 0A 0F
001+11:00:09.336 N 501: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 67 00

Heizen #2
001+11:03:39.150 N 605: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 E0 1F
001+11:03:39.334 N 606: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 7F (offset 3)
001+11:03:39.519 N 607: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF6), data: FF 01 A5 0A 0F
001+11:03:39.912 N 610: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 67 00
001+11:03:40.095 N 611: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 19 (offset 1)
001+11:03:40.282 N 612: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 25 04
001+11:03:40.747 N 613: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 0C F6 7F 1F (offset 1)
001+11:03:40.936 N 614: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 15 19 00
001+11:03:41.612 N 615: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 7B 00
001+11:03:41.797 N 616: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 25
001+11:03:41.980 N 617: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 9B F3

Kühlen
001+11:02:19.025 N 545: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 87 (offset 3)
001+11:02:19.209 N 546: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 67 1D
001+11:02:19.392 N 547: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 00 (offset 1)
001+11:02:19.990 N 548: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 00 00
001+11:02:20.173 N 549: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 F2 1F
001+11:02:20.366 N 550: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 00 F6 87 3F (offset 1)
001+11:02:20.755 N 551: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 00 00 02
001+11:02:20.938 N 552: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 7B 03
001+11:02:21.400 N 553: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 9B F2

Kühlen #2
001+11:04:23.162 N 639: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 00
001+11:04:23.346 N 640: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 00
001+11:04:24.063 N 641: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 00 (offset 1)
001+11:04:24.247 N 642: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 9B F2
001+11:04:24.430 N 643: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 F2 1F
001+11:04:25.453 N 644: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 87 3F (offset 3)
001+11:04:25.637 N 645: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 67 1D
001+11:04:25.822 N 646: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 00 02 (offset 1)
001+11:04:26.862 N 647: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 00 (offset 1)
001+11:04:27.047 N 648: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 03 7B 03

Kühlen + Heizen
001+11:04:56.900 N 668: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 19 (offset 1)
001+11:04:57.713 N 671: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 04 (offset 1)
001+11:04:57.897 N 672: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 E0 1F
001+11:04:58.081 N 673: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 B9 15
001+11:04:58.612 N 676: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 AF 25
001+11:04:58.803 N 677: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 A5 0C F6 FF (offset 1)
001+11:04:59.187 N 678: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 9B F3
001+11:04:59.374 N 679: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0xF6), data: FF 01 A5 0A 0F
```

```
System: RC300 in WLW196i
    "version": "3.6.0-dev.12",
    "platform": "ESP32",

"Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "01.16",
      "entities": 118,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x049F 0x04A0 0x04A3 0xF9"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "13.07",
      "entities": 51,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031E",
      "handlers ignored": "0xF9 0x0291 0xE7 0x02E0 0x02EA 0x061E 0xBF"
    },
    {
      "type": "thermostat",
      "name": "RC100H",
      "device id": "0x38",
      "product id": 200,
      "version": "40.04",
      "entities": 4,
      "handlers received": "0x042B 0x047B",
      "handlers fetched": "0x0273",
      "handlers ignored": "0xBF"
    },
    {
      "type": "gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.07",
      "entities": 0
    }
  ]
```

MichaelDvP, 2023-06-04:
Sorry, the F7 telegram is only to show/hide menu items in the RC300 menu. 
e.g the `thermostat(0x10) -B-> All(0x00), ?(0xF7), data: FF 01 E0 1F` means:
telegram FF 01 E0 (our counting 0x2E0), and 1F is a bitfileld, one bit for each offset, 
=> here paramters for 0x2E0 offsets 0-4 shown, 5-7 hidden in menu.
But F7 has not a function that ems-esp can use.

MichaelDvP, 2023-06-06:
> Can we say that the whole item is not interesting as it just switches some menu items?

The telegram 0xF7 is not interesting for us. 
> Found out that telegram 0x0291 is changing depending on the item.

For testing please use the dev build, go to settings->custom entities and add a entity with device 0x10, type 0x291,offset 5, UINT, writable. 
Check if it works. Then we can add it to the thermostat code as selectable heating/cooling/heating+cooling.
Please suggest a name for the setting (mqtt and localized UI: hpmode?
> In the same settings menu there is another setting that allows the following settings for Heating/Cooling. 

Isn't this already shown as ' hc1/hpoperatingmode' (telegrams hc1:0x471,hc2: 0x472, ...)

MichScha, 2023-06-07:
For the function type heating;cooling;heating/cooling - device 0x10, type 0x291,offset 5, UINT, writable I have added a new custom item
[image]

It seems that the write command doesn´t go out. I have logged via telnet the 'watch 291' and by changing the value via ems-esp it doesn´t show up. Am I missing something?

**Isn't this already shown as ' hc1/hpoperatingmode' (telegrams hc1:0x471,hc2: 0x472, ...)**
For at least my system no

```
{
  "name": "hpoperatingmode",
  "fullname": "hc1 heatpump operating mode",
  "circuit": "hc1",
  "type": "enum",
  "enum": [
    "off",
    "auto",
    "heating",
    "cooling"
  ],
  "readable": true,
  "writeable": true,
  "visible": true,
  "value": "not set"
}
```

**All are in telegram 0x467. Are the settings device specific, or hc specific?**
Jep, they depend all on the hc
Ref.: https://www.meinhausshop.de/media/pdf/meinhausshop-6720821143.pdf P.10 all items under 'Heizkreis 1 ... 4'
A few other entries from that service menu tree are also interesting and in my opinion worth adding: heating at and cooling at. Lets do those which we are discussing currently and add the other later - what do you think? Logging them is no problem.

**Can you do a read on the telegrams to see all parameters and for the following telegrams,:**
```
read 10 291
003+23:34:59.841 N 289: [emsesp] Me(0x0B) -R-> thermostat(0x10), ?(0x0291), length: 0x20
003+23:34:59.871 I 290: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0291), data: 01 00 00 03 FF 00

read 10 292
003+23:35:44.234 I 295: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0292), data: 00 00 00 00 FF 00

read 10 293
003+23:35:59.046 I 296: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0293), data: 00 00 00 00 FF 00

read 10 467
003+23:36:10.282 I 297: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0467), data: 0A 15 00 01 04

read 10 468
003+23:36:21.632 I 300: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0468), data: 0A 11 00 01 03
003+23:36:22.867 N 301: [emsesp] Me(0x0B) -R-> thermostat(0x10), ?(0x0291), length: 0x19 (offset 5)
003+23:36:22.886 N 302: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0291), data: 00 (offset 5)

read 10 469
003+23:36:41.545 I 303: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0469), data: 0A 11 00 01 03

```
**Are the temperature vaues all scale 1?**
Yes

MichaelDvP, 2023-06-08:
> It seems that the write command doesn´t go out. 

Yes seems a bug that UOM in needed to write the value out, havn't found the cause for this.

proddy, 2023-06-09:
> > It seems that the write command doesn´t go out.
> 
> Yes seems a bug that UOM in needed to write the value out, havn't found the cause for this.

are you fixing this or shall i my next web update (which is still 1-2 weeks out)

MichaelDvP, 2023-06-09:
I can fix it, already found the bug. But i'm out of home for next 10 days.

MichaelDvP, 2023-06-25:
fixed and added the thermostat entitiies of the following issues from @MichScha to thermostat. 
But it's in my dev, which uses the (fixed) espMqttClient. To merge in dev with with asyncMqtt is some manual work.

MichScha, 2023-06-25:
Can I do something to help or shall I just wait?

proddy, 2023-06-25:
wait for a while. There's some big PRs coming that need to be manually merged...Michael's dev branch with lots of fixes and new goodies and then my alova-dev branch has a lot of Web changes in it, plus some other enhancements in the C++ code.

proddy, 2023-07-03:
should be fixed in dev2 (or the test builds)

MichScha, 2023-07-04:
There seems to be something wrong with the HP Mode - will dig a little bit into it and report back. Basically it shows me always cooling even if I am changing the values. I will also test #1190 and #1192 next time. Thanks for integrating - makes live much easier when changing to summer-mode with cooling when those items are working!

MichScha, 2023-07-05:
Thank you very much!

Works like expected for all - tested R/W
Itemname: hpmode

Completes also
#1190 Dew Point Temperature Difference
- Itemname: dewoffset

#1192 Minimal flow temperature
- Itemname: hpminflowtemp

I think we are close to the point that all cooling related items for HPs are included :-)
@proddy you may want to add 3.6

closed 2023-07-05 by MichScha