emsesp/EMS-ESP32 #1188: Not all costomizations are saved, eg WW.... is only saved partitially

harvey637, 2023-06-04:
Setting:
I use EMS-ESP connected Bosch Heatpump 7400i ONLY for heating, not for pool and not for WARMWATER.

When I customize the heatpump and the thermostat I select all "WW" values and deselect them from gui and api, ok.
After update/reboot SOME of the WW.... values are back again.

I see them in the downloaded costomization configuration:
```
         "03damping",
          **"03/wwsettemp",**
          "03/wwmode",
          "03/wwsettemplow",
          "03/wwcircmode",
          "03/wwchargeduration|<3810",
          "03/wwcharge",
          "03/wwextra1",
          "03/wwextra2",
          "03/wwdisinfecting",
          "03/wwdisinfectday",
          "03/wwdisinfecttime|<1431",
          "03hybridstrategy",
          "03switchovertemp",
          "03energycostratio|>0<20",
          "03fossilefactor",
          "03electricfactor",
          "03delayboiler",

```
but they all start with "03/ww ......" not like the other eg "03fossilefactor".
These "/ww..." values are not related to "hc1" like "03hc1/wwprio", which is handled correctly (not shown in gui and api) after reboot.

So all this "**03/ww.....**" WITH THE **SLASH** are NOT evaluated for the gui and api customization.

Selecting "WW" in customization dialog lets me disable gui/api and push the save button. Until the next reboot everything is ok.
But it is not saved (and read) in the customization down/upload file.

No big deal (no usefull infos there), but it annoingly initiates useless creating the ww-variables in iobroker with iobroker.ems-esp adapter.

Thank You
Harvey

proddy, 2023-06-04:
yes, this looks incorrect. On which version are you on?

harvey637, 2023-06-04:
I am on 3.6.0-dev12, but it is present since "a long time" :-)

MichaelDvP, 2023-06-04:
There is a wrong "ww"-tag generated here
https://github.com/emsesp/EMS-ESP32/blob/46578dbe699987c66382cb9b014ce915e2f8a09e/src/emsdevice.cpp#LL1133C16-L1133C16
it should be 
`if (dv.tag >= DeviceValueTAG::TAG_HC1) {`

harvey637, 2023-06-04:
Hi at superfast MichaelDvP and Proddy,
tested dev-12 after fix 1188 - looking very good!!!!!

No more "03/ww...." in thermostat and no more "03ww/ww..." in boiler configuration file.

have a very nice (last hours somewhat :-) of this weekend!
cu
Harvey

commit 34046795 (2023-06-04): fix #1188
closed 2023-06-04 by harvey637