emsesp/EMS-ESP32 #1193: Buderus GB182 - HC1 mode change not work

ralf-koe, 2023-06-06:
### PROBLEM DESCRIPTION

GB182i.2 with MX300 and the EMS-ESP E32 adapter is running 
Detected:
Condens 5000i/Greenstar 8000/GC9800IW
UI800
Wifi module

Unfortunately I can not switch the HC1 mode. In the EMS-ESP frontend the value is editable. Temperatures I can set. Via Buderus app and the MX300 I can change the mode and it also updates the values via MQTT.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.6.0-dev.12",
    "platform": "ESP32",
    "uptime": "001+02:08:16.075",
    "uptime (seconds)": 94096,
    "free mem": 117,
    "max alloc": 65,
    "free app": 14,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "IPv4 address": "172.16.100.191/255.255.255.0",
    "IPv4 gateway": "172.16.100.1",
    "IPv4 nameserver": "172.16.100.232",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "172.16.100.1",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 29859,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
[... 133 lines cut ...]
change within MyBuderus App bring the change also in EMS-ESP, MQTT and the following debug log
```
2023-06-05 17:16:34.412   TRACE 1179:   [emsesp]     gateway(0x48) -W-> thermostat(0x10), ?(0x016E), data: FF
2023-06-05 17:16:35.018   TRACE 1180:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CB 00 01 1E 00 00 00 02 01 30 00 00 00 F8 00 00 0F 00 01 1E 00 00
2023-06-05 17:16:35.194   TRACE 1181:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 99 00 00 (offset 35)
2023-06-05 17:16:35.384   TRACE 1182:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 1E 00 64 41 00 00 00 00
2023-06-05 17:16:35.633   TRACE 1183:   [emsesp]     thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 02 00
2023-06-05 17:16:35.714   TRACE 1184:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 80 00 11 22 00 00 22 22 05 A0 00 03 01 01 1C 02 A4 00 00 11 02 00 FF FF
2023-06-05 17:16:35.943   TRACE 1185:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 00 FF 00 41 00 3C 01 FF 01 02 00 00 00 00 58 (offset 24)
2023-06-05 17:16:36.161   TRACE 1186:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 32 01 D3 01 D3 00 00 00 00 46 3C 00 00 02 00 01 C3 00 00 19 00 00 00 32 02 00
2023-06-05 17:16:36.475   TRACE 1187:   [emsesp]     thermostat(0x10) -W-> boiler(0x08), ?(0x02E0), data: 01 00 00 00 01
2023-06-05 17:16:36.503   TRACE 1188:   [emsesp]     thermostat(0x10) -W-> boiler(0x08), ?(0x02EA), data: 00
2023-06-05 17:16:36.553   TRACE 1189:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 00 00 00 00 00 00 00 06 5C
2023-06-05 17:16:36.738   TRACE 1190:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300Floordry(0x0267), data: 00 00
2023-06-05 17:16:36.916   TRACE 1191:   [emsesp]     thermostat(0x10) -B-> All(0x00), ?(0xC4), data: 00
2023-06-05 17:16:37.278   TRACE 1192:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 22 00 00 22 22 05 A0 (offset 3)
2023-06-05 17:16:37.461   TRACE 1193:   [emsesp]     thermostat(0x10) -B-> All(0x00), ?(0x016E), data: FF
2023-06-05 17:16:37.983   TRACE 1194:   [emsesp]     thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 00 00
2023-06-05 17:16:38.069   TRACE 1195:   [emsesp]     thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 (offset 8)
2023-06-05 17:16:45.012   TRACE 1196:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 CB 00 01 1E 00 00 00 02 01 30 00 00 00 F8 00 00 0F 00 01 1E 00 00
2023-06-05 17:16:45.189   TRACE 1197:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 99 00 00 (offset 35)
2023-06-05 17:16:45.379   TRACE 1198:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 1E 00 64 41 00 00 00 00
2023-06-05 17:16:45.583   TRACE 1199:   [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 05 01 D3 01 D3 00 00 00 00 46 3C 00 00 02 00 01 C3 00 00 19 00 00 00 05 02 00

```

proddy, 2023-06-06:
Clear write-up, thanks. We'll look into it.

MichaelDvP, 2023-06-07:
I think `absence` is not the same as `mode`. The thermostats also sets a lot of dhw settings after receiving the command.
It is a bit strange, 0x16E is listed on Norberts list as Junkers hc10 settings with offset 0 the type of hc.

Could you read the telegram 2B9 (terminal `read 10 2B9`) and give us the values in different modes. Also read the 16E telegram.

Use the settings custom entities to add a entitiy absence: device 0x10, type 0x16E, offset 0, INT, writable.
Then you can test different values, use `-1` to set 0xFF to absence mode.

theFeverDog, 2023-06-07:
I have a GB192i.2, currently still without the MX300, which is detected as the same device. I can also test this if needed.

ralf-koe, 2023-06-07:
Read 10 2B9 with absence deactivated
`000+00:03:32.877 I 1: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: 00 00 2A 00 26 01 00 00 FF 00 2A 01 E1 20 01 0F 05 2E 00 01 22 02 2A 00 01
000+00:03:33.132 I 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: 30 22 00 00 2E 00 00 (offset 25)
`
Read 10 2B9 with absence activated
`000+00:05:28.872 I 3: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: 00 00 2A 00 26 01 00 00 FF 00 2A 01 E1 20 01 0F 05 2E 00 01 22 02 2A 00 01
000+00:05:29.829 I 4: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: 30 22 00 00 2E 00 00 (offset 25)
`
and the value change also
[image]

and back to 0, after switch in the app.

Changing in EMS-ESP to -1 seems to be sendet, but nothing change in the app or MQTT. Change to 0 is the same.

Is this correct?
[image]

Read 10 16E in auto mode
`000+00:01:46.847 I 1: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x016E), data: 00
`
and after setting over the app to manual
`000+00:03:05.715 I 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x016E), data: FF
`

Test directly with the UI800 within the heater show the same results.

theFeverDog, 2023-06-07:
> without MX300 the app is not useable

I know, but this way I can verify if the issue originates with the MX300 or not... and hopefully have everything ready and working for when my MX300 arrives ;-)

So after creating the custom entity, I also see the value change from 0 to -1 if I switch the from Present to Absent via my remote control*. However, changing the value in the EMS-ESP interface is sent correctly, but is not reflected in the value, or the remote control display

* In my setup I also have a Buderus RC220 (recognized as an RT800), but this only seems to be a fancy temperature sensor with a display to control the settings exposed by the BC400/UI800.

ralf-koe, 2023-06-08:
Very good hint and cool, it works now.

How I can use the "User definied entity" with MQTT?

It looks like the value is not send to the MQTT server.

theFeverDog, 2023-06-08:
I can confirm that it works after setting the UoM.

In MQTT the custom device is shown under the EMS-ESP itself (not under the boiler or thermostat, at least in Home Assistant), and the status updates correctly. However it's only a sensor and cannot be edited.

MichaelDvP, 2023-06-08:
> However it's only a sensor and cannot be edited.

Yes, the custom entities are new in v3.6 and not completly implemented/tested. Mqtt-publish is in /custom_data/.

ralf-koe, 2023-06-08:
Thanks again and sorry, I should have checked it before in MQTT Explorer, reading works, publishing I will check later.

closed 2023-06-24 by proddy