emsesp/EMS-ESP32 #1196: Support multiple EMS-ESP's in HA

proddy, 2023-06-24:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1195

<div type='discussions-op-text'>

<sup>Originally posted by **g6094199** June 23, 2023</sup>
Hi,

i have to ems-esp (with different names an different heating systems) in the network and try to connect them to 1 HA instance.
i can see both ems in the mosquitto broker and they update nicely but HA seems to get fucked up. i see a mixture of settings ob both ems in the HA mqtt plugin.

so is there something wrong.  has this ever been tested? is this a ems or HA problem? or both in combination?

sash</div>

proddy, 2023-06-24:
I'll see if changing `ids` in the device entry for HA's discovery topic helps here
```json
  "dev": {
    "ids": [
      "ems-esp32"
    ]
```

proddy, 2023-06-24:
Ok implemented this and will continue testing. Then will merge into the dev build.

g6094199, 2023-06-25:
plz report when there is a testable beta :-)

proddy, 2023-07-03:
@g6094199 you can pick up the build & firmware with this fix at https://github.com/emsesp/EMS-ESP32/releases/tag/test

g6094199, 2023-07-04:
hi

i think we are almost there...

i flashed both units. one gets displayed as indendet (junkers-heatronic-wdb) but the other gets displayed as "unknown" (should be buderus-heatronic-da)). btw vendor should be prody, in any case.

[image]

.

proddy, 2023-07-04:
Strange, I'll see if I can reproduce. Do make sure you remove all the MQTT before flashing, either from within HA or using something like MQTTExplorer

g6094199, 2023-07-04:
in mosquitto i see:

[image]

proddy, 2023-07-04:
It looks like you're missing the main device called 'junkers-heatronic-wdb' in HA.

Can you use MQTTExplorer and show me the payload for the topic `homeassistant/binary_sensor/junkers-heatronic-wdb/system_status/config`.

If that doesn't exist find the topic with the word `proddy` in it, using the search.

If also may be good to check the HA logs.

g6094199, 2023-07-05:
`{"uniq_id":"system_status","obj_id":"system_status","stat_t":"junkers-heatronic-wdb/status","name":"EMS-ESP status","pl_on":"online","pl_off":"offline","stat_cla":"measurement","dev_cla":"connectivity","ent_cat":"diagnostic","dev":{"name":"junkers-heatronic-wdb","sw":"v3.6.0-dev.13b","mf":"proddy","mdl":"EMS-ESP","cu":"http://192.168.1.144","ids":["junkers-heatronic-wdb"]}}`

proddy, 2023-07-06:
Sorry, can't reproduce. If I use `junkers-heatronic-wdb` as a MQTT base it adds the EMS-ESP system device correctly. I would need to see the system settings for both EMS-ESP devices to recreate your environment and see what is happening. 

Did the HA logs tell you anything?

g6094199, 2023-07-12:
> 

`2023-07-08 17:08:15.002 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Integration numbeq is not supported
2023-07-08 17:08:15.099 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Integration numbeq is not supported
2023-07-08 17:08:17.574 ERROR (SyncWorker_1) [openwebif.api] There was a connection error calling http://192.168.2.66:80/api/bouquets Please check the network connection to the Enigma2 box is ok and enable debug logging in Enigma2 if required. Error: HTTPConnectionPool(host='192.168.2.66', port=80): Max retries exceeded with url: /api/bouquets (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6a3b6d10>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2023-07-08 17:08:18.380 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID system_uptime is already used by sensor.system_uptime - ignoring sensor.system_uptime_2
2023-07-08 17:08:18.436 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_heating_pump_modulation is already used by sensor.boiler_heating_pump_modulation - ignoring sensor.boiler_heating_pump_modulation_2
2023-07-08 17:08:18.442 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_current_flow_temperature is already used by sensor.boiler_current_flow_temperature - ignoring sensor.boiler_current_flow_temperature_2
2023-07-08 17:08:18.446 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_return_temperature is already used by sensor.boiler_return_temperature - ignoring sensor.boiler_return_temperature_2
2023-07-08 17:08:18.454 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_system_pressure is already used by sensor.boiler_system_pressure - ignoring sensor.boiler_system_pressure_2
2023-07-08 17:08:18.461 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_burner_current_power is already used by sensor.boiler_burner_current_power - ignoring sensor.boiler_burner_current_power_2
2023-07-08 17:08:18.467 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_burner_starts is already used by sensor.boiler_burner_starts - ignoring sensor.boiler_burner_starts_2
2023-07-08 17:08:18.473 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_total_burner_operating_time is already used by sensor.boiler_total_burner_operating_time - ignoring sensor.boiler_total_burner_operating_time_2
2023-07-08 17:08:18.479 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_burner_stage_2_operating_time is already used by sensor.boiler_burner_stage_2_operating_time - ignoring sensor.boiler_burner_stage_2_operating_time_2
2023-07-08 17:08:18.484 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_total_heat_operating_time is already used by sensor.boiler_total_heat_operating_time - ignoring sensor.boiler_total_heat_operating_time_2
2023-07-08 17:08:18.491 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_burner_starts_heating is already used by sensor.boiler_burner_starts_heating - ignoring sensor.boiler_burner_starts_heating_2
2023-07-08 17:08:18.496 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_total_uba_operating_time is already used by sensor.boiler_total_uba_operating_time - ignoring sensor.boiler_total_uba_operating_time_2
2023-07-08 17:08:18.505 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_last_error_code is already used by sensor.boiler_last_error_code - ignoring sensor.boiler_last_error_code_2
2023-07-08 17:08:18.510 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_service_code_number is already used by sensor.boiler_service_code_number - ignoring sensor.boiler_service_code_number_2
2023-07-08 17:08:18.518 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_maintenance_message is already used by sensor.boiler_maintenance_message - ignoring sensor.boiler_maintenance_message_2
2023-07-08 17:08:18.521 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_dhw_set_temperature is already used by sensor.boiler_dhw_set_temperature - ignoring sensor.boiler_dhw_set_temperature_2
2023-07-08 17:08:18.529 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_dhw_type is already used by sensor.boiler_dhw_type - ignoring sensor.boiler_dhw_type_2
2023-07-08 17:08:18.535 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_dhw_charging_type is already used by sensor.boiler_dhw_charging_type - ignoring sensor.boiler_dhw_charging_type_2
2023-07-08 17:08:18.539 ERROR (MainThread) [homeassistant.components.sensor] Platform mqtt does not generate unique IDs. ID boiler_dhw_current_intern_temperature is already used by sensor.boiler_dhw_current_intern_tempera [... cut ...]
2023-07-08 17:08:18.794 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID boiler_heating_active is already used by binary_sensor.boiler_heating_active - ignoring binary_sensor.boiler_heating_active_2
2023-07-08 17:08:18.799 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID boiler_tapwater_active is already used by binary_sensor.boiler_tapwater_active - ignoring binary_sensor.boiler_tapwater_active_2
.......`

proddy, 2023-07-13:
sorry what is the kesseltemperatur ? Is that coming from EMS-ESP?

g6094199, 2023-07-13:
"kesseltemperatur" is german for boiler temperature.

[image]

it must come from ems cause where should it alternativly come from?  there are no not EMS devices in the network.

the  "temperatursensor 1"  has many WW entries and seems to hold the rest of the stuff.

[image]

g6094199, 2023-07-14:
missed that info.....will try and report back

g6094199, 2023-07-14:
nope. there is still something wrong,

i did:
-unpowered both ems-esp units
-removed EVERY (also under HA topic) topic on mosquitto regardin one of the units.
-removed both units in HA
-rebootet HA
-powered buderus-heatronic-da only
-only buderus-heatronic-da showed up in HA (
-   buderus-heatronic-da
-   buderus-heatronic-da Thermostat
-   buderus-heatronic-da Boiler
)
-set up the unit in HA

everything fine so far.

but then:

-powerded junkers-heatronic-wdb

and instantly HA gets scrambled like so :

[image]

[mqtt-ae1d8fb3e7846b8056f3cc976520beae-buderus-heatronic-da Thermostat-6a5ba23b82f457dd306a7ca52cce2e38.json.txt](https://github.com/emsesp/EMS-ESP32/files/12053696/mqtt-ae1d8fb3e7846b8056f3cc976520beae-buderus-heatronic-da.Thermostat-6a5ba23b82f457dd306a7ca52cce2e38.json.txt)

proddy, 2023-07-15:
Ok, that's not good. Thanks for taking the time to remove everything and test it again.

I'll need reproduce your setup to see what is happening. Can you send me the config for both EMS-ESPs please? This is the JSON output from `http://<the DNS or IP address>/api/system`. Don't worry, no passwords are saved in this file. Then I'll test today and try and see what is happening.

g6094199, 2023-07-15:
[emsesp_settings.json.txt](https://github.com/emsesp/EMS-ESP32/files/12061129/emsesp_settings.json.txt)

[emsesp_settings(1).json.txt](https://github.com/emsesp/EMS-ESP32/files/12061130/emsesp_settings.1.json.txt)

MichaelDvP, 2023-07-15:
`"entity_format": 0,` makes the uniqu_id identical, because it does not contain the base. You need set to multiple.

proddy, 2023-07-15:
> `"entity_format": 0,` makes the uniqu_id identical, because it does not contain the base. You need set to multiple.

I think that's it! I was just about to configure 2 EMS-ESPs and test but believe this is the problem.

@g6094199 in MQTT settings set the Discovery setting to 'Multiple Instances'

g6094199, 2023-07-15:
ok,didnt know about this setting...

 both ems are now set to "Multiple Instances, short name" and it seems to show correctly.

[image]

proddy, 2023-07-15:
fantastic!

We'll update the documentation with 3.6.0 is released.

Enjoy the free software! :)

commit 158617f5 (2023-07-02): fixes #1196 - use basename as ids in HA to support multiple EMS-ESP's
closed 2023-07-15 by proddy