emsesp/EMS-ESP32 #1197: control and view the state of the bypass valve in HA

proddy, 2023-06-24:
Yes. Now it works with HomeAssistant and MQTT. It looks very good.:
[image]

Now it would be nice, if i can control and view the state of the bypass valve. My wife already starts to open the windows in the evening at these temperatures to let the cold air in. Maybe I could prevent that with smart bypass ventilation. :)

_Originally posted by @Timon321 in https://github.com/emsesp/EMS-ESP32/issues/1172#issuecomment-1577366450_

proddy, 2023-06-24:
anything to add here @Timon321 ?

Timon321, 2023-06-24:
I found out, that there is also a **room humidity** measurement. Activated it in the installer settings.
But dont now, how to give you an extraction of the values. The current Value was **56%**.
And bypass-valve was closed. It would be nice, if there was manual control and view of state of the bypass-valve.

Here are some telegrams from that above named values:
```ems-esp:$ watch 51
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x51
018+16:33:08.245 N 3: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 31 00 00 00 00 00 FD 00 F3 (offset 47)
018+16:33:08.428 N 4: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 41 (offset 45)
018+16:33:08.803 N 5: [emsesp] thermostat(0x10) -W-> ventilation(0x51), ?(0x0589), data: 03 FF
018+16:33:09.891 N 6: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 30 (offset 47)
018+16:33:10.209 N 7: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 31 (offset 40)
018+16:33:11.095 N 8: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FC 00 F4 (offset 53)
018+16:33:13.916 N 9: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 43 (offset 45)
018+16:33:14.099 N 10: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 F3 (offset 55)
018+16:33:16.628 N 11: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 42 (offset 45)
018+16:33:18.453 N 12: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 2F (offset 47)
018+16:33:20.207 N 13: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FD 00 F4 (offset 53)
018+16:33:22.720 N 14: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FC 00 F3 (offset 53)
018+16:33:25.541 N 15: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 43 (offset 45)
018+16:33:25.724 N 16: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 F4 (offset 55)
018+16:33:27.554 N 17: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 2E (offset 47)
018+16:33:29.054 N 18: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 42 (offset 45)
018+16:33:29.238 N 19: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FD (offset 53)
018+16:33:30.927 N 20: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 32 (offset 40)
018+16:33:31.828 N 21: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 41 (offset 45)
018+16:33:34.616 N 22: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 42 (offset 45)
018+16:33:34.800 N 23: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 F3 (offset 55)
018+16:33:38.316 N 24: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 41 (offset 45)
018+16:33:38.503 N 25: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FC 00 F4 (offset 53)
018+16:33:40.316 N 26: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 2F (offset 47)
018+16:33:42.827 N 27: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 33 (offset 40)
018+16:33:45.692 N 28: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 30 (offset 47)
018+16:33:46.724 N 29: [emsesp] Me(0x0B) -R-> ventilation(0x51), VentilationMode(0x056B), length: 0x19
018+16:33:46.749 N 30: [emsesp] ventilation(0x51) -W-> Me(0x0B), VentilationMode(0x056B), data: 05
018+16:33:47.254 N 31: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 F3 (offset 55)
018+16:33:49.692 N 32: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 31 (offset 47)
018+16:33:49.880 N 33: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FD 00 F4 (offset 53)
018+16:33:52.705 N 34: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FC (offset 53)
018+16:33:54.529 N 35: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 32 (offset 47)
018+16:33:55.052 N 36: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 32 (offset 40)
018+16:33:56.204 N 37: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FD (offset 53)
018+16:33:56.771 N 38: [emsesp] ventilation(0x51) -B-> All(0x00), ?(0x0575), data: 02 02 56 56 00 00 FF 80 00 01
018+16:33:57.001 N 39: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 FC 05 8F FF FF FF 00 F8 38 02 94 00 01 01 00 F3 06 0F 7F FF 00 80 00
018+16:33:57.524 N 40: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 80 00 00 9A 00 9A 00 0D 00 0A FF FF FF 3A 39 38 33 01 02 01 0B 00 41 (offset 24)
018+16:33:57.724 N 41: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 32 00 00 00 00 00 FD 00 F4 (offset 47)
018+16:33:58.290 N 42: [emsesp] ventilation(0x51) -B-> All(0x00), Blowerspeed(0x0585), data: 56 56 00 00 CB 4E 00 03 29 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
018+16:34:34.230 N 74: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 32 (offset 47)
018+16:34:36.653 N 75: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 30 (offset 40)
018+16:34:36.837 N 76: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 31 (offset 47)
018+16:34:39.792 N 77: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 2F (offset 47)
018+16:34:41.143 N 78: [emsesp] ventilation(0x51) -B-> All(0x00), VentilationMonitor(0x0583), data: 00 F3 (offset 55)```

proddy, 2023-07-03:
it could be the 2nd byte of telegram 583
```
VentilationMonitor(0x0583), data: 3A 38 33 01 02 01 0B 00 41 00 34 (offset 38)
```

As hex for 56 is 38. Could you `watch` that telegram and see if it matches with the bypass value

MichaelDvP, 2023-09-29:
I think we already have the state of the bypass valve. 
0x0587, offset 1 shows the state and 
0x055C, offset 1 is used to set it. 

The humidity was not known, i think it is offset 2 in `Airquality(0x05D9), data: 02 94 38` @Timon321 please check for other values. You can also use a custom entity to monitor.

moriax, 2025-08-31:
Hey, sorry for posting in the old issue.

But I have a problem with reading the Bypass value.

587 offset 1 switches shortly to 1, and then back to 0

```
ems-esp:$ read 51 587
000+07:07:44.546 N 3: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: 00 00 64 00 64 0A 00 01 54 01 00 01 00 00 18 46 00 00 00 02 00 A3 00 A3
000+07:07:44.579 N 4: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: <empty> (offset 24)

000+07:08:16.138 I 7: [command] Called command ventilation/bypass with value on
ems-esp:$ read 51 587
000+07:08:23.312 N 8: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: 00 01 64 00 64 0A 00 01 54 01 00 01 00 00 00 46 00 00 00 03 00 A3 00 A3
000+07:08:23.345 N 9: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: <empty> (offset 24)
ems-esp:$ read 51 587
000+07:08:30.746 N 10: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: 00 00 64 00 64 0A 00 01 54 01 00 01 00 00 00 46 00 00 00 02 00 A3 00 A3
000+07:08:30.779 N 11: [emsesp] ventilation(0x51) -> me(0x0B), W, Bypass(0x0587), data: <empty> (offset 24)

```

It seems, that for me 55C seems to be the correct adress for reading and writing ?
55C always shows the correct value:
First row is after opening bypass, second after closing again

```
ems-esp:$ read 51 55C
000+07:10:27.434 N 15: [emsesp] ventilation(0x51) -> me(0x0B), W, ?(0x055C), data: 08 01 11 17
ems-esp:$ read 51 55C
000+07:12:44.269 N 16: [emsesp] ventilation(0x51) -> me(0x0B), W, ?(0x055C), data: 08 00 11 17
```

MichaelDvP, 2025-09-16:
0x55C is a command telegram and not broadcasted. We can fetch it, but this updates only once a minute. I think there should be a state value, broadcasted  on change. I also don't know if the bypass only switches on command or could be triggered internally (without using 0x55C). I searched the logs and found https://github.com/emsesp/EMS-ESP32/issues/1172#issuecomment-1575552994 Seems that 0x587 offset 1 shows opening and offset 0 closing state of the bypass and both goes to 0 after state is reached. I guess 0x585 offset 23 shows a percent value of opening, on command open it goes to 0x64, on close it shows 0. Maybe this shows the state?
@moriax Please add a custom entitiy fpor 0x585, offset 23 and check if it shows the bypass state.

closed 2025-10-21 by proddy