emsesp/EMS-ESP32 #1198: set cooling mode [WSW196i.2-6 + RC220]

Bingo2023, 2023-06-28:
Buderus Logatherm WSW196i.2-6 with passive Coolingstation

Turn cooling on/off is not working via EMS-ESP32.
So I logged changes on MQTT & System-log. I changed setting via MX300 (inside the App).

It seams that changes come from thermostat in my case "Rego 3000/UI800" (Buderus RC220).

MQTT (all other values don't change)
[image]

[Kühlen_ausschalten.txt](https://github.com/emsesp/EMS-ESP32/files/11899332/Kuhlen_ausschalten.txt)
[Kühlen_einschalten.txt](https://github.com/emsesp/EMS-ESP32/files/11899333/Kuhlen_einschalten.txt)

Also funny that modetype:"Eco" stands for off/Stop

EMS-ESP Version
v3.6.0-dev.9

MichaelDvP, 2023-06-30:
Seems to be `RC300Set(0x02B9), data: 01 (offset 28)`
I've added to my dev build, please test: https://github.com/MichaelDvP/EMS-ESP32/releases

Bingo2023, 2023-06-30:
Hello Michael,

thanks for the fast response.

Seems like the latest build has some general problem. After flashing the EMS device are not showing up. Reflashing also does not help. See screenshot below. Some minutes later the ESP32 crashed and does not respond to webinterface. Later will will reassemble it and try USB flashing...
[image]

Bingo2023, 2023-07-01:
I remember a message like unexpected message end (or similar)...
Is the EPS running out of memory?

MichaelDvP, 2023-07-01:
> Seems like the latest build has some general problem. 

Yes, the min/max fix for mqtt does not work as expected, i've fixed it.

Bingo2023, 2023-07-01:
new version is running :)
Also cooling function is working now (turn on/off).

There is just a typo in the UI, but technical working - thanks for the quick fix - you guys are Incredible fast !

[image]

Bingo2023, 2023-07-01:
I would expect that "WP Kühlen" from Heatpump should be the status of the above setvalue.
It is not working at the moment. I my assumption is right, how can I catch the log for it to provide it here?

[image]

MichaelDvP, 2023-07-01:
Translation is fixed now,
The "WP Kühlen" is a generated value from hpActivity "Kompressoraktivität", telegram 0x48D offset 7. 
Just log all and change the cooling setting on thermostat, then we should see the thermostat change, the thermostat->boiler command and the boiler monitor message.

Bingo2023, 2023-07-01:
here is the log for it.
Hope I captured it right.
[log.txt](https://github.com/emsesp/EMS-ESP32/files/11926574/log.txt)

Bingo2023, 2023-07-01:
"Kompressoraktivität" is not working as this is a passive Cooling station.
 **Buderus PKS9** [Wärmetauscher Sole<->Heizkreis]

Bingo2023, 2023-07-13:
The Version of @MichaelDvP, leads to ESP32 freeze every few days (2-3d).

Is this fixed already fixed in latest release (including cooling fix of this threat)?
[Test Build v3.6.0-dev.13c](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test)
[Development Build v3.6.0-dev.12](https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest)

MichaelDvP, 2023-07-14:
Could you please exlain what you mean with "freeze". I don't see anything like that and no user have reported something like that.

The cooling is implemented in my dev/dev2 builds and the official test-build: https://github.com/emsesp/EMS-ESP32/releases/tag/test

Bingo2023, 2023-07-14:
I updated to "ESP-3_6_0-dev_13c-ESP32.bin"
I'll report if I see it again.

Bingo2023, 2023-07-17:
I want to report that the ESP32 with above SW-Version freezes every 2-3 days.
By freeze I mean that it is not responding to Webinterface or MQTT. I can't check more because it is built in Heat Pump - I can only give it a power reset).

Bingo2023, 2023-07-17:
> > Seems like the latest build has some general problem.
> 
> Yes, the min/max fix for mqtt does not work as expected, i've fixed it.

could this improovement be the issue? Just extended from some minutes to 2-3d ?

proddy, 2023-07-17:
it could be a memory leak - do you use something like Home Assistant or iobroker? Then keep a history of the free mem and available block from the heartbeat MQTT topic. Also a copy of the output from http://ems-esp.local/api/system would show us the memory and how many entities are loaded.

MichaelDvP, 2023-07-18:
Without info about the system it could be a lot of reasons. 
- wifi disconnect, see #1073
- mqtt broker disconnect, mqtt queue memory increasing
- mqtt qos issue, queue increasing
- ...

I have 2 emsesp, one (S3) for testing, updated often, and one production system (MH-ET), running stable, normally uptimes of some days before updating. I did not see a memory leak in normal operation.

proddy, 2023-07-24:
@Bingo2023 do we still need to fix

> "Kompressoraktivität" is not working as this is a passive Cooling station.
Buderus PKS9 [Wärmetauscher Sole<->Heizkreis]

?

Bingo2023, 2023-07-24:
@proddy good point !
With our current calc methode this is not possible (HP state is never showing cooling-state).
I'm not sure what is the right signal, but MX300/Buderus app is showing the cooling state from somewhere.

Can we find this in the logs I posted above (first post)?
Otherwise I could make new logs.

proddy, 2023-07-24:
The log you posted I think has all the right telegrams in it. Michael is the expert in finding easter eggs.

MichaelDvP, 2023-07-24:
I don't think the log shows the real switching. I can see the command to set cooling off here:
`17:51:37.648 TRACE 636: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Set(0x02B9), data: 00 (offset 28)`
but difficult to see a state changes. Possible candidate is:
`17:51:40.194 TRACE 649: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 (offset 32)`
but this byte was also zero before in the last periodic poll.
`17:51:13.245 TRACE 509: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 0D 00 28 00 00 00 32 00 00 00 00 01 (offset 25)`
May a test to add this offset 32 to custom entities and look if it changes.

Bingo2023, 2023-07-24:
[image]

@MichaelDvP good job !!!
the attached custom entitie, monitors the status of cooling correct :)

MichaelDvP, 2023-07-24:
I'll add it to dev2.

proddy, 2023-07-24:
love it! The custom entity is such a nice way to find values in a telegram. We should write something about this technique in the Wiki.

proddy, 2023-07-31:
This is now in the dev build https://github.com/emsesp/EMS-ESP32/releases/tag/latest

Bingo2023, 2023-08-04:
I've installed latest dev built.
Is there a duplicate entry (marked red)?
[image]

proddy, 2023-08-04:
> I've installed latest dev built. Is there a duplicate entry (marked red) in MQQT message? [image]

The values come from 2 different telegrams, and one is writable (i.e. can switch on/off).

MichaelDvP, 2023-08-05:
> The values come from 2 different telegrams, and one is writable (i.e. can switch on/off).

Yes, as mentioned here: https://github.com/emsesp/EMS-ESP32/issues/1198#issuecomment-1647850588
Enable/disable cooling in `RC300Set(0x02B9), offset 28` and the actual state of cooling in `RC300Monitor(0x02A5), offset 32`

Can we close this issue?

closed 2023-08-08 by proddy