emsesp/EMS-ESP32 #1210: Validate file name extension on file upload

proddy, 2023-07-08:
As reported by @swa72 when you upload a file with unrecognized extension (not bin, json or md) it will report back an Unauthorized HTTP error. It's better to use something like '406 Not Acceptable' and have the web code pick this up and report back. The upload should also be immediately stopped (Upload.abort()).

MichaelDvP, 2023-07-08:
> The upload should also be immediately stopped (Upload.abort()).

It already stops [here](https://github.com/emsesp/EMS-ESP32/blob/b57bc1107052595beba61ada679c9a4ece6f7531/lib/framework/UploadFileService.cpp#L43-L45) and does not start the Update, you only need to add `handleError(request, 406);`before return. Maybe set for a wrong bin-file also 406 instead of 503-service unavailable

proddy, 2023-07-08:
yes, I tried adding a handleError(request, 401) at that exact place but it doesn't abort the POST. Looking through the internet I see other people have similar problems cancelling the upload.

proddy, 2023-07-08:
Ah, found a workaround. I added a `request->client()->close()` after the send.

commit 362d837b (2023-07-10): Merge pull request #1211 from proddy/dev2

report on invalid extension - fixes #1210
closed 2023-07-10 by proddy