emsesp/EMS-ESP32 #1213: JSON formatting problem

tp1de, 2023-07-09:
I had an user of my ioBroker Adapter reporting crashes while reading his heatpump values.
I recognized that this was due to formatting within the JSON values for the field:

boiler.uptimecomppool 
{ "name": "uptimecomppool", "fullname": "operating time compressor pool", "circuit": "", "value": 71582784., "type": "number", "uom": "minutes", "readable": true, "writeable": false, "visible": true }

Javascript JSON.parse is crashing on the value field ending with a decimal point.  
I changed my ioBroker Adapter code to avoid crashing but it would be better to change the formatting ems-esp code either to avoid a decimal point the end or add 1 decimal place ... e.g. 71582784.0

proddy, 2023-07-09:
which version of EMS-ESP is this?

tp1de, 2023-07-09:
Originally 3.5.1 as delivered from BBQKees. But the User tested with latest 3.6 dev as well. 
Formatting was the same ... I can't judge

tp1de, 2023-07-09:
https://github.com/tp1de/ioBroker.ems-esp/issues/20#issuecomment-1627384694

MichaelDvP, 2023-07-09:
The ending 0 is cut because render_value uses char[10]. Easy fix.

proddy, 2023-07-11:
fixed

closed 2023-07-11 by proddy