emsesp/EMS-ESP32 #1215: tempdiffcool range parameter in JSON wrong

basalblas, 2023-07-10:
The current JSON for tempdiffcool is producing the error below in HA log:

`ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_temp_diff_tc3_tc0_cool: 2.0 (range 3.0 - 10.0)`

The latest recommended value for this setting on a Nefit EnviLine All-Electric Split is 2K as can be found [here](https://s3.eu-west-1.amazonaws.com/np-nefit-pro-production/Documenten/Advies-instellingen-Nefit-Bosch-EnviLine-ALL-ELECTRIC-V01-05-2022.pdf)

The JSON at http://ems-esp/api/boiler/tempdiffcool defines the range to be min 3 to max 10, and should be changed to min 2 max 10.

```
{
  "name": "tempdiffcool",
  "fullname": "Temp.vers. TC3/TC0 koel.",
  "circuit": "",
  "value": 2.0,
  "type": "number",
  "min": 3,
  "max": 10,
  "uom": "K",
  "readable": true,
  "writeable": true,
  "visible": true
}

```

proddy, 2023-07-10:
Thanks for reporting - we'll change in the dev version

proddy, 2023-07-11:
will be in dev2 when Michael merges his PR

closed 2023-07-11 by proddy