emsesp/EMS-ESP32 #1216: LAN-Gateway E32 stürzt häufig ab-Keine Verbindung

Volker2912, 2023-07-10:
### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT) Nein, aber Mail mit Kees
- [X ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

Gerät ist unregelmäßig im LAN nicht mehr erreichbar: Dann Unterbrechung Stromversorgung. Zusätzlich erforderlichenfalls Neustart Software, weil nicht alle Topics sichtbar sind.  MQTT - Signalqualität 100%

### EXPECTED BEHAVIOUR

Keine Abstürze erwartet

### SCREENSHOTS
keine

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2023-07-10:
https://emsesp.github.io/docs/Support/

proddy, 2023-07-10:
Looks like running out of memory. Try v3.6.0 and also look at the FAQ in the documentation to see what you further can optimize

Volker2912, 2023-07-10:
V3.6.0 ist eine Entwicklungsversion – nutze ich (noch) nicht.

FAQ dazu nichts gefunden

Wenn ein Speicher voll läuft: Wie kann ich den automatisch leeren?

Macht es vielleicht Sinn, in regelmäßigen Abständen neu zu starten?

proddy, 2023-07-10:
You will need to wait for the official build then.