emsesp/EMS-ESP32 #1233: Buderus Logano plus GBH212 Hybrid System - kWh meters

lurchii, 2023-07-24:
Hello,

Recently I did just integrate EMS-ESP32 into HomeAssitant. Buderus Gb212 with Heatpump. I was hoping to get the power consumption values of the boiler (kWh or m³ for the gas) and the heating pump (kWh for electrical power) to use Energy Dashboard. But I am surprised that these values do not show up as entities.

Dashboard shows boiler and heat pump group with a number of entities. See screenshots. Each providing power in %. But as already said I was hoping to have kWh consumption (gas kWh for boiler and electrical kWh for heat pump) available as entities.
[image]

Find available entities attached.
[device_entities_heatpump.csv](https://github.com/emsesp/EMS-ESP32/files/12146854/device_entities_heatpump.csv)
[device_entities_termostat.csv](https://github.com/emsesp/EMS-ESP32/files/12146866/device_entities_termostat.csv)
[device_entities_boiler.csv](https://github.com/emsesp/EMS-ESP32/files/12146867/device_entities_boiler.csv)

Any idea how I could get kWh values out of the system?

Thanks!
Uli

MichaelDvP, 2023-07-24:
For master heatpumps we have this info in telegram 0x495, for the hybrid heatpumps this is unknown. 
The gas boilers does not provide this info, you have to read burner on, modulation and calculate with the burner rated power. I think this was discussed a few times here.
Try to update to newest software, make a complete log of some minutes and tell what values (name and value) we have to search for. Use the thermostate to read the energy values while logging.

lurchii, 2023-08-06:
Thanks for reponse. 

Modulation is always 100%, burner on is not available. Rated power is 22kW.

In the meanwhile I've managed to calculate burner power in kW using following code
    # Errechnet aus Brennerleistung in [%] die Brennerleistung in [kW}
      - name: Brennerleistung kW
        unique_id: "Brennerleistung_kW"
        unit_of_measurement: 'kW'
        state: >-
            {{(((float(states('sensor.boiler_curburnpow'))) / 100) * 22 ) | round(2)}}

Then I was using a helper to get in kWh using Integration - Riemann Summenintegralsensor.
Then I was using that entity to set up counter to add in enery dashboard.
[image]

When checking values for one heating cycle uunfortunately it doesn't match with values shown in Buderus App.

HA Energy Dashboard:
[image]

Buderus App
[image]

So at the moment I am not sure I can trust what I can see...

tp1de, 2023-08-27:
@lurchii  what are your publishing intervalls within mqtt?

I just added the energy consumption calculation within my ioBroker adapter for the ems-esp gateway. Pll-Time for api read is set 15 seconds. Then calculated consumtion is even more accurate than recordings from Buderus gateway.

There is one "tricky" part within the Buderus App logic:
Buderus takes samples every minute (60 seconds). So hourly values should have  60 samples. I recognized that sometimes samples are missed. When this happens, than Buderus interploates the values. E.g. 10% samples missing, then value is increased by 10%.  This might be right if

bbqkees, 2023-08-28:
I wonder where the RC310 etc get the max burner power info in kW from?

I do get emails often about this so lot's of people expect the kWh of the gas boiler to be present in ems-esp.
Maybe something to have a better look at.

tp1de, 2023-08-28:
I recommend to discuss this here and start with setting max burner power within the ems-esp settings.
https://github.com/emsesp/EMS-ESP32/discussions/1268

closed 2024-02-22 by proddy