emsesp/EMS-ESP32 #1243: [homeassistant] adjust naming of MQTT entities to new guidelines

Sirs0ri, 2023-08-02:
### PROBLEM DESCRIPTION

The latest HomeAssistant version (2023.8) has changed the guidelines on how it expects MQTT entities to be named: https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/
tl;dr: entities associated with a device aren't supposed to have the device's name as part of the entity name.

Since the update, there's now a warning displayed in HomeAssistant about the `system_status` entity that used to be called `EMS-ESP status`, which HomeAssistant has automaticelly renamed to `status`.

This will become a breaking problem in 6 months, with HA 2024.2.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
  "System Info": {
    "version": "3.5.1",
    "platform": "ESP32",
    "uptime": "026+22:34:41.865",
    "uptime (seconds)": 2327681,
    "free mem": 97,
    "max alloc": 42,
    "free app": 1,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "C4:DE:E2:B1:7A:53",
    "IPv4 address": "192.168.188.99/255.255.255.0",
    "IPv4 gateway": "192.168.188.1",
    "IPv4 nameserver": "192.168.188.65",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 544163,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 6,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 60,
    "publish time solar": 60,
    "publish time mixer": 60,
    "publish time other": 60,
    "publish time sensor": 60,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 2014947,
    "bus reads (tx)": 465571,
    "bus writes (tx)": 6,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": false,
    "max web log buffer": 50,
    "web log buffer": 29
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "BK13/BK15/Smartline/GB1x2",
      "device id": "0x08",
      "product id": 64,
      "version": "02.05",
      "entities": 63,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC30",
      "device id": "0x10",
      "product id": 67,
      "version": "02.08",
      "entities": 49,
      "handlers received": "0xA3 0x06 0xA2 0x12 0x48 0x47 0x38 0x39",
      "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
      "handlers pending": "0x13 0x42 0x49 0x4C 0x52 0x51 0x53 0x56 0x5C 0x5B 0x5D 0x60",
      "handlers ignored": "0x1A 0x35 0x05"
    },
    {
      "type": "controller",
      "name": "BC10/RFM20",
      "device id": "0x09",
      "product id": 68,
      "version": "02.00",
      "entities": 0,
      "handlers ignored": "0x29"
    }
  ]
}

```

### TO REPRODUCE

1) Set up a HomeAssistant instance on at least version 2023.8
2) Enable EMS-ESP's HA autodiscovery feature for HomeAssiant
3) make sure the EMS-ESP device gets discovered, there should be a `system_status` entity automatically.
4) The warning will show up in HomeAssistant's logs and as a possible repair at HA's main settings page.

### EXPECTED BEHAVIOUR

The `system_status` entity created through MQTT discovery should not have `EMS-ESP` at the start of its name, so that HA stops complaining.

### SCREENSHOTS

n.a.

### ADDITIONAL CONTEXT

The log HomeAssistant displays:

```
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294
Integration: MQTT (documentation, issues)
First occurred: 23:06:32 (1 occurrences)
Last logged: 23:06:32

MQTT entity name starts with the device name in your config {'unique_id': 'system_status', 'object_id': 'system_status', 'state_topic': 'ems-esp/status', 'name': 'EMS-ESP status', 'payload_on': 'online', 'payload_off': 'offline', 'device_class': <BinarySensorDeviceClass.CONNECTIVITY: 'connectivity'>, 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'device': {'name': 'EMS-ESP', 'sw_version': 'v3.5.1', 'manufacturer': 'proddy', 'model': 'EMS-ESP', 'configuration_url': 'http://192.168.188.99', 'identifiers': ['ems-esp'], 'connections': []}, 'payload_available': 'online', 'force_update': False, 'availability_mode': 'latest', 'payload_not_available': 'offline', 'qos': 0, 'encoding': 'utf-8', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Status'
```

tfriberg, 2023-08-03:
Same issue for me; can confirm.

proddy, 2023-08-03:
Thanks for reporting. I'll upgrade to 2023.8 too and check.

spetrie13, 2023-08-03:
Same for me:
Discovered entities with a name that starts with the device name
This stops working in version 2024.2.0. Please address before upgrading.
Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

sensor.ems_esp_status
binary_sensor.system_status
Warning - 03/08/2023

zigomatichub, 2023-08-03:
the guide is here https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

fugazzy, 2023-08-04:
I find three of them:
```
binary_sensor.system_status
sensor.ems_esp_status
binary_sensor.ems_esp_status
```

MichaelDvP, 2023-08-04:
I think what HA calls `device` is the the mqtt-base, which is editabel in mqtt settings. Renaming should avoid the conflict.

rpi-fan, 2023-08-05:
Worked for me:
in MQTT-settings in ems-esp I changed the Client-ID from default "ems-esp" just to "buderus-123", restartet EMS-ESP and 
HA.
Problem seems to be solved.

piltous, 2023-08-06:
Thank you for the fix!

Stefan28L, 2023-08-06:
Not for me :-(

[image]

Is still recognized as "EMS-ESP" in the HA after restart !?

[image]

Is there anything else to do ?

proddy, 2023-08-06:
> Not for me :-(
> 
> [image]
> 
> Is still recognized as "EMS-ESP" in the HA after restart !?
> 
> [image]
> 
> Is there anything else to do ?

It's the MQTT base name that you need to change, not the client id.

But I'll work on a proper fix when I'm back home. Until then try changing the base from the MQTT settings page.

MichaelDvP, 2023-08-06:
Change the **`Base`**, not the client-ID. 
We have in the config of status:
https://github.com/emsesp/EMS-ESP32/blob/6e653eea7ee5fe502a95448be966d943642fcfc5/src/mqtt.cpp#L529
and the device is set here:
https://github.com/emsesp/EMS-ESP32/blob/6e653eea7ee5fe502a95448be966d943642fcfc5/src/mqtt.cpp#L539-L543
The reported error message is `MQTT entity name starts with the device name in your config`
so we have to change the device name `dev["name"]` not the hardcoded model name `mdl`
But i think for future it is easier to change the entity name  `EMS-ESP status` to `System  status`, which is also more in line with the `System NTP status`, etc.

proddy, 2023-08-06:
I upgraded my HA to 2023.8.1 and tested with the latest dev and don't get any warnings in the HA Logs so not sure we need to change anything. Has anyone tried the dev build?

Roos-AID, 2023-08-06:
Just upgraded to latest dev version and after cleaning the MQTT broker it is fixed. Restarted HA and EMS-ESP.
I have Base and Client-id as ems-nefit.

piltous, 2023-08-06:
No more message after the restart. I'm in dev4

Stefan28L, 2023-08-06:
sorry, but now i'm totally confused.
I tried changing both: First the "Base" (Restart EMS-ESP + HA) and then later the "Client ID" (Restart EMS-ESP + HA) .Unfortunately both without success. I had updated to the latest version, but not (yet) to the "dev".
In the home assistant, the device is still displayed as "EMS-ESP"!?
However, I now have some duplicate entries for the entities.
New entities were created and my existing ones are now "unknown". Unfortunately, a problem that is not quite as important is now an acute problem, since my automations are no longer running.

What do you mean by "cleaning the MQTT broker"? Maybe I forgot a step?

proddy, 2023-08-06:
> sorry, but now i'm totally confused. I tried changing both: First the "Base" (Restart EMS-ESP + HA) and then later the "Client ID" (Restart EMS-ESP + HA) .Unfortunately both without success. I had [...]

I would recommend you upgrade to the dev branch. It'll replace 3.5.x in a few days anyway and we'll no longer be supporting older versions. I can help you get it all setup when you're on 3.6.0

Stefan28L, 2023-08-06:
Thank you for the offer. I'm on V3.6.0-dev.14 now.

proddy, 2023-08-06:
> Thank you for the offer. I'm on V3.6.0-dev.14 now. I'm not sure how to get rid of my duplicate entries in HA. Does that have to be done manually, or is there an easier way?

It really depends on your setup. What I usually do is
 - use MQTT Explorer and delete everything under 'homeassistant'. This clears out any topics that are 'retained'
 - to be sure this worked, I go to HA, into the MQTT integrations (/config/integrations/integration/mqtt) and manually delete anything that starts with ems-esp
 -  restart EMS-ESP

then you have a 100% clean install.

Stefan28L, 2023-08-06:
Is it pssible to change the conversation to an other chanel, because i think this is a little bit "off topic" now.
I'm afraid i don't have enough knowledge to carry out all the steps correctly.

If i delete everything under "homeassistant" won't that affect my other installations ?

proddy, 2023-08-06:
ok, create a new discussion topic for this

MichaelDvP, 2023-08-06:
If you go to mqtt settings, disable HA discovery, but leave mqtt enabled, click `apply settings` and wait some minutes, ems-esp will remove all HA configs associated with the actual base. Then change base (if required) and enable HA discovery again to build new configs.

Stefan28L, 2023-08-06:
@proddy : I think i fixed it now. useed MQTT Explorer and delete everything under 'homeassistant' with ems-esp (or what ever i tryed) in the name. Now the dobble entities diseperead.

I still have the warning in HA.

@MichaelDvP : This is to avoid the warning ? After your steps - i will get the same problems in HA again ?

MichaelDvP, 2023-08-06:
> @MichaelDvP : This is to avoid the warning ? After your steps - i will get the same problems in HA again ?

This is the ems-esp internal way to remove the homeassistant configs like you have done with mqtt-explorer. If you disable HA discovery in ems-esp and leave mqtt on, ems-esp subscribes to `homeassistant/<base>` and removes any entry the broker sends. Reenable ha discovery newly creates all active configs. This is a way to cleanup the discovery from old entities.

Stefan28L, 2023-08-06:
Thank you for your help and your patience ;-)
After I renamed the "base" the warning is gone now too. Now everything seems to be working properly again.

proddy, 2023-08-09:
fixed in dev17. As Michael said, there was only one entry in v3.6.0

commit 8dfff293 (2023-08-09): fixes #1243
commit b96cede1 (2023-08-09): Merge pull request #1252 from proddy/dev

rename EMS-ESP Status to System Status so HA doesn't complain (fixes #1243)
closed 2023-08-09 by proddy