emsesp/EMS-ESP32 #1246: Greenstar Comfort I (RF) Controller and Thermostat not recognised

to0kan, 2023-08-06:
### PROBLEM DESCRIPTION

The boiler was recognised successfully, however, neither controller nor thermostat were found. I have Greenstar Comfort I
which, judging from the WIKI should be supported: https://bbqkees-electronics.nl/ems-device-compatibility/ (Comfort RF)

[image]

000+00:00:06.696 I 12: [emsesp] Recognized new boiler with deviceID 0x08
000+00:00:07.235 I 13: [emsesp] Recognized new controller with deviceID 0x09
000+00:00:08.138 N 14: [emsesp] Unrecognized EMS device (deviceID 0x18, productID 215). Please report on GitHub.
000+00:00:10.000 I 15: [emsesp] Stopping Access Point
000+00:00:24.926 I 16: [emsesp] Scanning devices...
000+00:01:06.711 E 17: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 88 26 00 1B

### REQUESTED INFORMATION

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i-30i",
      "device id": "0x08",
      "product id": 123,
      "version": "06.02",
      "entities": 72,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "controller",
      "name": "Controller",
      "device id": "0x09",
      "product id": 152,
      "version": "02.09",
      "entities": 0
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x18",
      "product id": 215,
      "version": "01.03",
      "entities": 0,
      "handlers ignored": "0x06 0x02A5 0x35 0x1A"

```

MichaelDvP, 2023-08-06:
In the dev build this thermostat is already implemented. Please update to dev. 
The Controller is recognized but not shown in dashboard because it has no entities.

to0kan, 2023-08-06:
I have updated the firmware and can indeed see the thermostat now, thank you so much!

Is the controller somewhere on the roadmap by any chance, or is it not compatible at all?

MichaelDvP, 2023-08-06:
Yes, the RF thermostats do not have a setting telegram, there is no way to write to.
The controller is already full supported, but there is nothing to read or set in it. We list as device in the api-settings, but not in dashboard, because there are no entites to show.

to0kan, 2023-08-06:
Got it, thanks a lot

closed 2023-08-06 by to0kan