emsesp/EMS-ESP32 #1247: HM200 (53) v3.6.0-dev.14 no values in entities (hpcompspd) (heatpumpoutput) but compressor is running

TobiRabid, 2023-08-06:
System: KB195-15 / RC310 (74.03) / HM200 (OF06.02) / WLW196-11 (04.19)

Heatpump / ODU (outdoor unit)  is running in heating mode, compressor is running  (30%)
The entities (hpcompspd) (heatpumpoutput) show no values.

In the ems-dashbord the entities are not listed (in summary 28 entrys)
In settings/custommizations the entities are listed (in summary 31 entrys)

Also checked with mqtt explorer, this both entities were never sent.

[image]
[image]

[watch53.txt](https://github.com/emsesp/EMS-ESP32/files/12270399/watch53.txt)

MichaelDvP, 2023-08-07:
The data for HM200 come from #500. 
`heatpumpoutput` is a setting from function test, maybe not set in normal operation. 
`hpcompspeed` seems to be a typo number rotation: offset 51 in a telegram of length 37 couldn't be, should be offset 15. 
I'll fix it in dev after proddys PR is merged and we get working binaries again.

TobiRabid, 2023-08-07:
@MichaelDvP thx for your fast response ! The compressor state is one f the most importand monitoring values of the hp.
The current cold days in germany allows me a little bit to test the hm200/ems-esp integration.
I see some other interessting but missing hm200 values/states/entities but not so importand like the hpcompspd. Durring the winter time i will try to record and describe this states/values. Unfortunately im not a developer so that i´m not able to commit here :-(

MichaelDvP, 2023-08-07:
The compressor speed should work in https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest, please check.
The HM200 have a lot of unknown telegrams. The methode to find the right telegrams/offsets is easy for commands, difficult for measurement-values.
For commands log the telegrams (log all) and change the value on the thermostat to different values. If we know the old and new value we can find the command in the log.
For measurements it's the same, we need a change of the value and need to know the old/new value, then we can find it in the log. But it's difficult to find the right time where the value changes and the bus updates some telegrams only in 10sec or one minute intervals,

TobiRabid, 2023-08-07:
tested, works, thx

closed 2023-08-10 by proddy