emsesp/EMS-ESP32 #1255: EMS bus disconnected on Bosch 2500W

ahhoj, 2023-08-11:
### PROBLEM DESCRIPTION

After a power cycle of the boiler (Bosch 2500W), EMS-ESP cannot connect to the boiler. EMS bus disconnected error message is present on Dashboard. System log contains nearly only "Incomplete RX: ..." error messages. This board and boiler combination worked well for some monthes, same kind of power cycle also happened during this period. The board is E32 from https://bbqkees-electronics.nl/

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

{
  "System Info": {
    "version": "3.6.0-dev.17",
    "platform": "ESP32",
    "uptime": "000+00:03:59.069",
    "uptime (seconds)": 239,
    "free mem": 148,
    "max alloc": 103,
    "free app": 8,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -76,
    "IPv4 address": "192.168.99.99/255.255.255.0",
    "IPv4 gateway": "192.168.99.1",
    "IPv4 nameserver": "0.0.0.0",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 20,
    "API fails": 20
  },
  "Bus Info": {
    "bus status": "disconnected"
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": true,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "connect",
      "name": "RS232",
      "device id": "0x04",
      "product id": 0,
      "version": "00.00",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

Nothing particular, just turn on boiler and ems-esp board.

### EXPECTED BEHAVIOUR

EMS bus is connected, boiler is visible for the ems board.

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

Tried changing power supply, factory reset. Changed the cable conencting the boiler and board. Tried v3.5.1 and latest dev (v3.6.0-dev17)

System log from the web ui:
```lua
000+00:01:14.991 WARNING 147: [telegram] Incomplete Rx: FF 06 FF 0C
000+00:01:17.816 WARNING 151: [telegram] Incomplete Rx: FF 04 0C 08 FF 04 0C FF
000+00:01:21.855 WARNING 158: [telegram] Incomplete Rx: 0C 04 0C 0C FF
000+00:01:26.335 WARNING 167: [telegram] Incomplete Rx: FF 0C FF 0C
000+00:01:26.475 WARNING 168: [telegram] Incomplete Rx: FF 04 FF 0C 0C FF
000+00:01:39.328 ERROR 182: [emsesp] Call failed with error: unknown device (Error)
000+00:01:43.029 WARNING 189: [telegram] Incomplete Rx: 04 0C 0C FF
000+00:01:45.671 WARNING 193: [telegram] Incomplete Rx: FF 04 04 04 0C 04 FF 0C 0C 0C 0C 0C
000+00:01:46.552 WARNING 198: [telegram] Incomplete Rx: FF 0C 0C 04 0C FF
000+00:01:49.279 WARNING 201: [telegram] Incomplete Rx: 08 04 08 08 08 0C 0C
000+00:01:49.632 WARNING 203: [telegram] Incomplete Rx: 0C 0C 0C 0C 0C
000+00:01:50.352 WARNING 207: [telegram] Incomplete Rx: FF 0C 0C 04 08 0C 0C 04 0C 0C 0C 0C 0C 0C FF 0C
000+00:01:50.691 WARNING 209: [telegram] Incomplete Rx: 0C FF 0C 0C FF 0C FF
000+00:01:53.611 WARNING 211: [telegram] Incomplete Rx: 08 0C 0C 0C 08 0C 0C 0C 0C 08 0C 0C 0C 0C 0C 0C 0C 0C 04 08 0C 0C
000+00:01:53.891 WARNING 212: [telegram] Incomplete Rx: 08 0C 0C 0C 04 0C 0C 0C 0C 0C 0C 0C FF 0C
000+00:01:57.437 WARNING 214: [telegram] Incomplete Rx: 04 04 0C 0C 04 04 FF 04 0C 04 FF 04 FF 0C 08 0C 0C 0C 0C 0C 04
000+00:01:59.050 WARNING 216: [telegram] Incomplete Rx: 0C 0C 0C 0C 0C 0C 0C 0C 0E FF 0C FF 0C 0C FF 04 08 0C 0C 0C 0C 0C 0E 0C 0C 0C 0C 0E
000+00:02:01.097 WARNING 217: [telegram] Incomplete Rx: 08 08 0C 0C
000+00:02:01.490 WARNING 219: [telegram] Incomplete Rx: 08 0C 0C 04 0C 0C 0C 0C
000+00:02:01.490 WARNING 219: [telegram] Incomplete Rx: 08 0C 0C 04 0C 0C 0C 0C
000+00:02:01.950 WARNING 220: [telegram] Incomplete Rx: 04 0C 0C 0C 0C 0C 0C 08 0C FF 0C FF 0C 0C 0C 0C 0C FF 06 0C 0C 04 FF 0C 0C 0C
```

proddy, 2023-08-11:
It's most definitely hardware. Either a faulty cable, a dirty input connection or the board is broken. If you get the same errors with both v3.6 and v3.5 it can't be the software.

MichaelDvP, 2023-08-11:
Unstable power can have this effect. You have already tried changing the power supply, Try to power from the internal usb instead of 12V supply. Check if 5V and 3.3V is stable.

ahhoj, 2023-08-11:
Thank you for your replies. I tried changing cable, tried the internal usb, but the behaviour remain the very same. Itis just strange that this whole setup worked well, and got faulty suddenly. But I guess this will be the case after all.

bbqkees, 2023-08-14:
A new replacement Gateway is already on its way so this issue can be closed.

closed 2023-08-14 by proddy