emsesp/EMS-ESP32 #1256: loading configuration file after upgrade from 3.4.4 to 3.6.0 fails

florian-reus, 2023-08-11:
### PROBLEM DESCRIPTION

after update from 3.4.4 according to homepage (saving configuration, flashing with esp flasher) I can not load the old configuration. I can select the .json file, device is automatically restarting, but coming up without the configuration again.
[image]

(it even seems to forget that I configured 100 log entries).

tested with attached boiler and without.

log after restart:
000+00:00:00.000 I 1: [emsesp] Starting EMS-ESP version 3.6.0-dev.17
000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
000+00:00:00.000 I 3: [emsesp] Loading board profile S32
000+00:00:00.000 N 4: [system] Upgrading to version 3.6.0-dev.17
000+00:00:00.000 I 5: [emsesp] Starting Scheduler service
000+00:00:00.000 I 6: [emsesp] Starting Custom entity service
000+00:00:00.000 I 7: [temperaturesensor] Starting Temperature sensor service
000+00:00:00.000 I 8: [emsesp] Loaded EMS device library (116 records)
000+00:00:00.000 I 9: [emsesp] Starting Access Point with captive portal on 192.168.4.1
000+00:00:05.096 I 10: [emsesp] WiFi connected with IP=192.168.9.241, hostname=ems-esp
000+00:00:05.129 I 11: [emsesp] mDNS responder service started
000+00:00:10.000 I 12: [emsesp] Stopping Access Point
000+00:00:14.870 I 13: [emsesp] Recognized new boiler with deviceID 0x08
000+00:00:16.068 I 14: [emsesp] Recognized new controller with deviceID 0x09
000+00:00:17.022 I 15: [emsesp] Recognized new thermostat with deviceID 0x10
000+00:00:19.838 I 16: [emsesp] Recognized new mixer with deviceID 0x20
000+00:00:20.345 I 17: [emsesp] Recognized new mixer with deviceID 0x21

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :
{
  "System Info": {
    "version": "3.6.0-dev.17",
    "platform": "ESP32",
    "uptime": "000+00:03:26.921",
    "uptime (seconds)": 206,
    "free mem": 125,
    "max alloc": 75,
    "free app": 8,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -78,
    "IPv4 address": "192.168.9.241/255.255.255.0",
    "IPv4 gateway": "192.168.9.1",
    "IPv4 nameserver": "192.168.9.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 429,
    "bus reads (tx)": 123,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 6,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 96
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 17
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 69,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.01",
      "entities": 83,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlers ignored": "0x02E1 0x02E2 0x02EB 0x02EC 0x35"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x1A 0x0255"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 3,
      "handlers received": "0x02D7",
      "handlers ignored": "0x1E 0x1A 0x0255"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}
```lua
  System information output here:

```

### TO REPRODUCE

- saved configuration from 3.4.4
- updated with ESP flasher to 3.6.0 (

### EXPECTED BEHAVIOUR
- config is loaded and working

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

added configuration file

**(Please, remember to close the issue when the problem has been addressed)**
[emsesp_settings_3.4.4.json.zip](https://github.com/emsesp/EMS-ESP32/files/12322615/emsesp_settings_3.4.4.json.zip)

proddy, 2023-08-11:
thanks for reporting. I never tested upgrading from 3.4 to 3.6 and will look into it (although its like 10 months old).

MichaelDvP, 2023-08-11:
Yes, v3.6 does not load all saved settings (only Network and emsesp-Settings), the names here are wrong:
https://github.com/emsesp/EMS-ESP32/blob/ac889d921fb245576b8b3f1aa82f9f50c72d2b66/src/system.cpp#L1020C15-L1023
The saved file in 3.6 is compatible, only the loading looks for "NTP Settings" section instead of "NTP", etc.. OTA section is completly missing. I'll make a PR.

proddy, 2023-08-11:
Thanks Michael. Thought you were going on vacation!

MichaelDvP, 2023-08-11:
~2 weeks, starting sunday.

florian-reus, 2023-08-11:
Thanks!
Is there a workaround?
Using a saved file from 3.6 and putting settings from 3.4.4 in it?
Mit freundlichen Grüßen,
Florian Reus

On 11 August 2023 19:15:34 CEST, MichaelDvP ***@***.***> wrote:
> Yes, v3.6 does not load all saved settings (only Network and emsesp-Settings), the names here are wrong: [...]

MichaelDvP, 2023-08-11:
Workaound: 
Update to https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest or wait for the mergeand load it from here, then upload the v3.4.4 settings. 
The saved/edited v3.6. file is also only working after the fix.

proddy, 2023-08-11:
Michale's update merged. I haven't tested but it's in the latest dev 3.6.0-dev.18

closed 2023-08-13 by proddy

florian-reus, 2023-08-13:
Tested, working here
Mit freundlichen Grüßen,
Florian Reus

On 13 August 2023 14:52:00 CEST, Proddy ***@***.***> wrote:
>Closed #1256 as completed.
>