emsesp/EMS-ESP32 #1262: v3.6.0 lost the "force heating off" option

proddy, 2023-08-13:
v3.6.0 lost the "force heating off" option. Downgrading to v3.5.1 fixes the issue.

_Originally posted by @martini1992 in https://github.com/emsesp/EMS-ESP32/issues/951#issuecomment-1676432735_

martini1992, 2023-08-13:
Specifically makes the device unusable when using it as an interface for more complex control through Home Assistant. Due to not using the new scheduler functionality in EMS-ESP.

ahhoj, 2023-08-18:
I checked how the scheduler works in v3.6.0. I think the functionality itself is clear. Most probably it is only my fault but didn't quite find out what messages should I send every minute. I know it is present in the #951 issue, but still couldn't figure out (admitedly I didn't spend much time with it :) ). It could be nice to have a wiki guide how to configure to mimic the earlier functioanlity: forceheatingoff (including the startup option).

martini1992, 2023-08-18:
So I need to add and toggle 3 one minute timer schedules (selflowtemp > 0, burnerpower > 0, heating > 0) when I want to stop/start my heating? Seems like a poor solution to me for a well known set of commands that need to be sent to specific models of boilers to implement required functionality.

I don't understand what the issue was with having the "force heating off" virtual entity available.

MichaelDvP, 2023-08-28:
I think the freezes are memory related. If mqtt disconnects the queue is filled consuming memory until a minimum is reached, 
1. this https://github.com/emsesp/EMS-ESP32/blob/86e29515e789b26384eed4dae210b65d50f6adc4/lib/espMqttClient/src/Config.h#L32 is to small, i think in v3.5 we had ~40k as minimum free before deleting queued messages.
2. We queue messages while mqtt is disconnected: https://github.com/emsesp/EMS-ESP32/blob/86e29515e789b26384eed4dae210b65d50f6adc4/lib/espMqttClient/src/Config.h#L40 in v3.5. publish/queue was stopped on disconnect. 

Maybe better to change both settings. 
(BTW. due to vacations and not updating my emsesp, it is running >16 days now on dev18, no freeze, no memory leak.)

MichaelDvP, 2023-08-28:
I think only setting selflowtemp is enough, maybe it takes a few seconds until the boiler stops the burner, but it should work.
If not, you can set the whole telegram in the scheduler
[image]
or create a own entity as ULONG and schedule this entiy.

jrb80, 2023-09-15:
I also found this issue after upgrading to v3.6.1.

Is this entity permanently removed now?

philipherbert, 2023-10-04:
Hello,

why is forceheatingoff no longer available ?
It is possible to set this via Buderus Controller to disable heating.

proddy, 2023-10-07:
@MichaelDvP is it worth adding back this feature?

philipherbert, 2023-10-07:
.just a short update from my side: 

I needed to revert to 3.5.1 to be able to change this back to forceheatingoff = off. 

This setting does not seem to have anything todo with setting on the UI of the boiler. (GB192i/2 BC400).

Emergency Mode overwrites ignores forceheatingoff anyway.

--Philip

MichaelDvP, 2023-10-07:
It's just a bit copy/paste, not much code. I'll make a PR, less work than discussing this further.

MichaelDvP, 2023-10-07:
@philipherbert 
> This setting does not seem to have anything todo with setting on the UI of the boiler. (GB192i/2 BC400).

Yes, see #951, it is a special ems-esp function to schedule a command for setting flowtemp and burner to zero. This is a workaround for boilers that can not set heating off by bus. It's not a entity of the boiler!

closed 2023-11-18 by proddy