emsesp/EMS-ESP32 #1265: Loss of Configuration after update (3.6.0)

bpcrocke, 2023-08-18:
After updating via file upload in the web interface device restarted and lost all settings (wifi, mqtt). This could only be fixed by logging in to the ems-esp wifi and restoring a backup. Loss of all data also sometimes happens if device crashes. It at least needs a robust storage of wifi data otherwise maintenance is a hassle if one is not physically on site to connect to standard wifi.

proddy, 2023-08-18:
upgrading from which version? Also need more details. See https://emsesp.github.io/docs/Support/

MatzeFro, 2023-08-31:
I have the same Problem, after update from one Version to an new one (today from 3.7.0-dev.1 to 3.7.0-dev.2) the system makes an FactoryReset or lost all configuration.
Sometime after 1 hour or 1 day or 3 day a factory reset also happened.
if I try to download the settings - the file have only the version number in it or version and OTA but never the hole settings.

bpcrocke, 2023-09-06:
> upgrading from which version? Also need more details. See https://emsesp.github.io/docs/Support/

From previous beta version. But this also happens randomly. When I was away for three weeks the device suddenly stopped working and lost all settings. Was on version 3.6.0.

proddy, 2023-09-06:
Interesting. I'll keep an eye on if his happens with or Michael, who are doing many upgrades a week,

MichaelDvP, 2023-09-08:
Have seen this very, very rarely and last time is long ago, somewhere in 3.0. Maybe a corrupted file system, a not finshed write on reboot? We can add a `LittleFS.end()` before doing a `ESP.restart()`, maybe it helps..

proddy, 2023-09-08:
I've actually seen it happen a few times, but only the MQTT settings disappear. I haven't been able to find the root cause.

MichaelDvP, 2023-10-04:
Seems in the today update of core 2.0.13 / 3.0.0 there is a change in LittleFS, After upload the FS is formatted and ems-esp is set to defaults. Happens today on two ems-esp, one S3, one C3. After first initialization all is ok, can flash older versions and go back to new without losing the settings.

bpcrocke, 2023-10-05:
Had a full loss of config data again, updating from 3.6.0 to 3.6.2. Sorry, no logs due to the loss of all data. 

Support Info (after reconfiguring): 
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "000+00:35:07.065",
    "uptime (seconds)": 2107,
    "free mem": 110,
    "max alloc": 47,
    "free app": 6581,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -60,
    "IPv4 address": "192.168.1.219/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 280,
    "MQTT queued": 0,
    "MQTT publish fails": 21,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 3299,
    "bus reads (tx)": 926,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 4,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 40
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "06.03",
      "entities": 73,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "18.06",
      "entities": 50,
      "handlers received": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlers ignored": "0x23 0x35 0x1A 0xBF"
    },
    {
      "type": "solar",
      "name": "SM100/MS100",
      "device id": "0x30",
      "product id": 163,
      "version": "27.05",
      "entities": 36,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x0380 0x038E 0x0391",
      "handlers pending": "0x035D 0x035F 0x035C 0x0361",
      "handlers ignored": "0x0368 0x03A9 0x03AA 0x35 0xBF"
    },
    {
      "type": "controller",
      "name": "Controller",
      "device id": "0x09",
      "product id": 152,
      "version": "03.01",
      "entities": 0
    }
  ]
}

proddy, 2023-10-05:
@mtbsteve if all settings have been lost, then it'll start up with an Access Point and you can follow the instructions on the EMS-ESP wiki or BBQKees wiki.

To help narrow down the problem, I have a few questions:
- how did you perform the upgrade? via the WebUI, Flash tool, CLI?
- have you ever done an upgrade before, or is this the first time?
- is this a BBQKees gateway board?

mtbsteve, 2023-10-05:
@proddy thanks for getting back. Unfortunately my EMS-ESP seems to be dead after the update. I can’t connect remotely, nor I get an access point. The blue led is solid on after a reboot. Please help I updated via the webpage in the System tab of the EMS ESP interface. And yes, I performed several updates successfully  over the past year since I purchased the device directly from Bbqkees. It’s an original board.

MichaelDvP, 2023-10-05:
Solid led means connected to bus and wifi, so this is not a loss of config-data. Sounds more like the "freeze" users report in #1264. Do you get blank page or timeout?
- Try clear your browser cache 
- Check Wifi-AP if it reports ems-esp as connected
- powercycle ems-esp

mtbsteve, 2023-10-05:
@MichaelDvP Thanks for your fast response. A further power cycle of the eMS ESP fixed it. The device is up and running again, BUT it reverted back on my previous Version 3.5.0. (!)so yes,  it sounds similar to what @gwilford reported ia few days ago n the issue you mentioned.

proddy, 2023-10-07:
@MichaelDvP do you think the fall-back to the backup partition is now fixed?

MichaelDvP, 2023-10-07:
Yes, i think this case is a failed update. 
The fallback was imo a write to wrong partition on a (first time) working app1, next reboot it does not find valid bootloader data and fall back to app0. But only in one of my first dev with power entities, and no loss of configuration (to be on topic again).

ziporah, 2023-10-30:
I am facing the same issue here. Was on 3.5.0 and for some reason the devices just stopped working. Settings gone and all. Tried updating to 3.6.2 with the UI, upload succeeded, reconfigured settings, rebooted and settings gone again, device seems also to be stuck in boot/startup, no EMS-ESP access point available. Strange enough, if I connect the device with micro-usb cable direct to the esp32, it did boot and wasn't stuck, settings however were still gone. Manually connected trough serial cable the wifi, reset and removed the usb cable. Device starts up again (after a very long time) and I could reconfigure it. Tried restarting the device in the web UI and now restarting does work without loosing settings. I'll see how it goes and if it stays online, I'll report back if it looses configuration again.

proddy, 2023-11-05:
I think this is a result of a bad dev version we created that messed up the partitions. The cleanest way is to make a copy of the settings, erase the ESP32 and re-flash with the latest dev. If you see the issue again please re-open.

closed 2023-11-05 by proddy