emsesp/EMS-ESP32 #1267: Add Shower Duration to MQTT Discovery topic

proddy, 2023-08-26:
Add the shower_duration as an MQTT topic which makes it easier to add as an HA entity

proddy, 2023-08-29:
See https://emsesp.github.io/docs/Home-Assistant/#shower-integration

proddy, 2023-08-29:
Added a new command called boiler/coldshot which can be called via the API and HA

rene865, 2023-08-30:
can you see what i am doing wrong? I have just started with home assistant and yaml.
EMS-ESP Version 3.7.0-dev.0

i added your example in configuration.yaml, at first I thought the code here was wrong _ems_esp_shower_DATA_duration but what I try it doesn't work.
[image]
[image]

but for the new entity the shower duration is not availible
[image]

proddy, 2023-08-30:
Sorry! I hadn't made the build yet. Doing it now.

proddy, 2023-08-30:
> Sorry! I hadn't made the build yet. Doing it now.

https://github.com/emsesp/EMS-ESP32/releases/tag/latest

closed 2023-08-31 by proddy