emsesp/EMS-ESP32 #1274: Gateway S32 (Standard WiFi Edition V2.0) - Restart and loss config

MatzeFro, 2023-08-31:
i flashed the V3.7.0-dev.2 with EMS-ESP-Flasher.
after config everything - it runs a while.
i restart the system - that looks ok.
but after a while the system ist down and restarts without any config.
and i do everything again.
i config the system again and safe the settings. after I look in the json-file I wonder.
every export have different content in it.

MatzeFro, 2023-08-31:
after flashing with the EMS-ESP-Flasher I see:
[00:05:47]
[00:05:47]┌────────────────────────────────────────┐
[00:05:47]│ EMS-ESP version 3.7.0-dev.2            │
[00:05:47]│ https://github.com/emsesp/EMS-ESP32    │
[00:05:47]│                                        │
[00:05:47]│ type help to show available commands   │
[00:05:47]└────────────────────────────────────────┘
[00:05:47]
[00:05:47]ems-esp:$ ./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
[00:05:47]E (62) esp_littlefs: mount failed,  (-84)
[00:05:47]E (62) esp_littlefs: Failed to initialize LittleFS
[00:05:49]E (1600) esp.emac: emac_esp32_init(349): reset timeout
[00:05:49]E (1601) esp_eth: esp_eth_driver_install(214): init mac failed
[00:05:49]./components/esp_littlefs/src/littlefs/lfs.c:3169:debug: Bad block at 0x5
[00:05:49]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x2, 0x3}
[00:05:49]./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x2, 0x3}
[00:05:51][00:05:51]

how can I fix it ?

proddy, 2023-09-01:
From which version did you upgrade from? Was everything working efficiently before? Do you have any dump of the system settings so we can see if there is anything strange? I would erase the ESP32 using the flasher and start again, and if there are still issues then perhaps the ESP32 and onboard flash RAM is corrupt and the board needs replacing.

MatzeFro, 2023-09-01:
I try many versions - and I think it never works perfekt.
maybe I try a Windows tool or a linux tool.
Waranty over BBQKees Electronics

bbqkees, 2023-09-01:
If it happened after an update, then probably something went wrong in that process.
Please load 3.6.0 firmware from a Windows computer and see if the problem is resolved.
Also after each update you need to clear the browser cache otherwise it may appear it is not working correctly.

If it's really a hardware defect then I will send a new Gateway of course but I don't think that's the issue here.

MatzeFro, 2023-09-01:
I try it again with a Linux Computer:
the problem is the same :(
after I connect to ems-esp WLAN - i configure it - I check an export-settings twice and get twice different content.

here is the Flash-Log: (I just wondered about the warning)
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 EMS-ESP-3_6_0-ESP32.bin
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ec:62:60:4c:d3:04
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001fefff...
Warning: Image file at 0x1000 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 17104 bytes to 11191...
Wrote 17104 bytes (11191 compressed) at 0x00001000 in 0.4 seconds (effective 336.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 127...
Wrote 3072 bytes (127 compressed) at 0x00008000 in 0.0 seconds (effective 530.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 696.2 kbit/s)...
Hash of data verified.
Compressed 2024320 bytes to 1280288...
Wrote 2024320 bytes (1280288 compressed) at 0x00010000 in 24.2 seconds (effective 670.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...Connecting....

proddy, 2023-09-01:
thats suck. Sorry that you're having issues. Do you have another spare ESP32 dev board you can test with?

I would also use esptool to erase the flash completely first using `python -m esptool --chip esp32 erase_flash`

MatzeFro, 2023-09-01:
i will search and test

bbqkees, 2023-09-01:
Have you also tried the EMS-ESP flash tool?

MatzeFro, 2023-09-01:
i tried: 
1.  python -m esptool --chip esp32 erase_flash - then flash again ---- same result
2. flash with the EMS-ESP flash tool ---- same result
3. I found a ESP32-WRoom-32 and try it to flash. ----- that works. the download settings ist 3 KB and have the correct content. A reboot starts correct.

I will but it to an power adapter and observe it

bbqkees, 2023-09-01:
Well it looks like you are one of the first who fried a S32 :-)
Maybe a corrupt flash chip or something.

Can you reply on your order confirmation email or send me your order number
via the contact form in the webshop?
Then I can send you a new Gateway right away.

MatzeFro, 2023-09-01:
done

closed 2023-09-01 by MatzeFro

len-sla, 2024-07-25:
Just not  to  start new issue as it looks similar
first background I bought EMS interface board V3.1
then tried to flash esp on Linux ESP32 4MB board but without success as it seemed extra * bin's from webpage were not appropriate size and overlapping occurred.

Anyway I sent email to Kees mail with question and got nice quick response  :simply use flashing tool and at best  "... an  ESP32-S3 module with 16MB flash and some PSRAM on it.A good candidate would be the T7-S3. "

I bought that board but result was as below I've  tried version 3.6.5 and 3.7.0 
( in the meantime I flashed ESP32 4MB board without any issues and whole system is working)

Though Now I have a bit of time  to make T7-S3 working to have capacity for the future. It is nothing urgent.

Log after flashing T7-S3 is as follows

Auto-detected serial port: COM5
Using 'COM5' as serial port.
Connecting...
Detecting chip type... ESP32-S3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S3
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: NO
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: NO
 - MAC Address: 34:85:18:8F:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 16MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.5s
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00226fff...
Flash params set to 0x024f
Compressed 13936 bytes to 9695...
Writing at 0x00000000... (100 %)
Wrote 13936 bytes (9695 compressed) at 0x00000000 in 0.2 seconds (effective 680.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 165...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (165 compressed) at 0x00008000 in 0.0 seconds (effective 806.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1188.7 kbit/s)...
Hash of data verified.
Compressed 2191008 bytes to 1383684...
Writing at 0x00010000... (1 %)
Writing at 0x0001bdf1... (2 %)
Writing at 0x00025b22... (3 %)
Writing at 0x00030e04... (4 %)
Writing at 0x00039ad1... (5 %)
Writing at 0x00041a2b... (7 %)
Writing at 0x00045958... (8 %)
Writing at 0x00049953... (9 %)
Writing at 0x0004d94e... (10 %)
Writing at 0x00051949... (11 %)
Writing at 0x00055944... (12 %)
Writing at 0x0005993f... (14 %)
Writing at 0x0005d93a... (15 %)
Writing at 0x00061935... (16 %)
[... 85 lines cut ...]
[15:13:29]│ su to access admin commands              │
[15:13:29]│                                          │
[15:13:29]│ https://github.com/emsesp/EMS-ESP32      │
[15:13:29]│                                          │
[15:13:29]└──────────────────────────────────────────┘
[15:13:29]
[15:13:29]ems-esp:$ E (9) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}
[15:13:29]
[15:13:29]E (9) esp_littlefs: mount failed,  (-84)
[15:13:29]E (11) esp_littlefs: Failed to initialize LittleFS
[15:13:29][   533][E][vfs_api.cpp:105] open(): /littlefs/config does not exist, no permits for creation
[15:13:29][   583][E][vfs_api.cpp:105] open(): /littlefs/config/networkSettings.json does not exist, no permits for creation
[15:13:29][   585][E][vfs_api.cpp:105] open(): /littlefs/config does not exist, no permits for creation
[15:13:29][   622][E][vfs_api.cpp:105] open(): /littlefs/config/apSettings.json does not exist, no permits for creation
[15:13:29][   641][E][vfs_api.cpp:105] open(): /littlefs/config/ntpSettings.json does not exist, no permits for creation
[15:13:29][   661][E][vfs_api.cpp:105] open(): /littlefs/config/otaSettings.json does not exist, no permits for creation
[15:13:29][   683][E][vfs_api.cpp:105] open(): /littlefs/config/mqttSettings.json does not exist, no permits for creation
[15:13:29][   730][E][vfs_api.cpp:105] open(): /littlefs/config/securitySettings.json does not exist, no permits for creation
[15:13:29][   822][E][vfs_api.cpp:105] open(): /littlefs/tmp_upload does not exist, no permits for creation
[15:13:29][   826][E][vfs_api.cpp:105] open(): /littlefs/config/emsespSettings.json does not exist, no permits for creation
[15:13:29][   830][E][Preferences.cpp:483] getString(): nvs_get_str len fail: boot NOT_FOUND
[15:13:29][   889][E][vfs_api.cpp:105] open(): /littlefs/config/emsespCustomization.json does not exist, no permits for creation
[15:13:29][   916][E][vfs_api.cpp:105] open(): /littlefs/config/emsespScheduler.json does not exist, no permits for creation
[15:13:29][   945][E][vfs_api.cpp:105] open(): /littlefs/config/emsespEntity.json does not exist, no permits for creation
[15:13:30][15:13:30]