emsesp/EMS-ESP32 #1275: Entities disappeared between EMS-ESP versions

guardianbs, 2023-09-01:
### PROBLEM DESCRIPTION

Just updated from version 3.4 to version 3.6.0 (and subsequently to 3.7.0-dev 2) and noticed that a couple of entities have disappeared.

They were available and updated under version 3.4. On version 3.6.0 and 3.7.0-dev 2 they appear in the Settings - Customization section but have no value. They do not appear at all in the Dashboard.

The entities are **setflowtemp** and **setburnpow**. These were sent to HA as **sensor.boiler_set_flow_temperature** and **sensor.boiler_burner_set_power** respectively.

The boiler is a Worcester Bosch Greenstar 37CDi and nothing has changed on the boiler between versions.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

### TO REPRODUCE
Update to version 3.6.0

### EXPECTED BEHAVIOUR
As prior to version 3.6.0 the entities should be made available to HA

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT
[EMS-ESP Problem.txt](https://github.com/emsesp/EMS-ESP32/files/12498363/EMS-ESP.Problem.txt)

proddy, 2023-09-01:
That's odd. Both are sent in the UBASetPoints (telegram 0x1A). Do you see any failures in the log? If you force it via a 'read 1a' from a telnet console, do they appear?

MichaelDvP, 2023-09-01:
When comming from older version it is recommended to rebuild the HA config:
- disable HA discovery in MQTT settings (leave mqtt enabled), Apply changes.
- wait ~5 minutes
- reenable HA discovery

proddy, 2023-09-01:
Thanks @MichaelDvP, forgot about that.

So @guardianbs you need to use 'burner set power (setburnpow)' and 'selected flow temperature (selflowtemp)'.

guardianbs, 2023-09-01:
Hi

Done all as suggested above. No change.

setburnpow and setflowtemp do not appear on the Dashboard and are not published to MQTT.

selflowtemp is not the same as setflowtemp 

Both setburnpow and setflowtemp  - i.e values generated from UBASetPoints (telegram 0x1A) - were created / published in version 3.4

proddy, 2023-09-01:
oh, ok, then open a telnet session, do a `watch 1a` and let's see what comes in

guardianbs, 2023-09-01:
ems-esp:$ watch 1a
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x1A
000+01:56:41.556 N 1: [emsesp] thermostat(0x18) -W-> boiler(0x08), UBASetPoints(0x1A), data: 55 06

This is the same as above as shown in the log when the Thermostat was turned on (with a different value for setburnpow obviously)

MichaelDvP, 2023-09-02:
Checked the old commits. The renaming was long ago in v2.x->v3.0 and we don't have setflowtemp anymore.
Later we added the 1A telegram with new setflowtemp und setburnpow and check telegrams also for destination and map them to the device, 
But there were some issues with this telegram handling and with [this](https://github.com/emsesp/EMS-ESP32/commit/b499d908abb05974d22f45a6657e992fa174a609) commit we revert mapping the telegram destination to device and the entities disappear. So the values or 1A telegram are not updated anymore. 
If we want to add it back we need a better logic, `if (emsdevice->is_device_id(telegram->src) || emsdevice->is_device_id(telegram->dest)) {` does not work. But i think changing to 
```
        if (emsdevice->is_device_id(telegram->src)) {
            ...
            break;
        } else if (emsdevice->is_device_id(telegram->dest)) {
            emsdevice->handle_telegram(telegram);
        }
```
will do without issues.

MichaelDvP, 2023-09-02:
@guardianbs : Please check the test build: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
It should give you the entities back.

guardianbs, 2023-09-02:
Yes, back now. All good. Thank you.

commit 2f5b7cd0 (2023-09-09): add back process dest, fixes #1275
commit 6173e94c (2023-09-20): add back process dest, fixes #1275, v3.6.1-dev0a
closed 2023-09-02 by guardianbs