emsesp/EMS-ESP32 #1278: Unknown device 0x18, Product ID 215

basak, 2023-09-02:
### PROBLEM DESCRIPTION

I have an "Unknown" device. Maybe this is the [Greenstar Comfort I](https://www.worcester-bosch.co.uk/products/boiler-controls/directory/greenstar-comfort-i-1), but I also see an "HT3" device in the output below I don't recognise. It currently shows up in EMS-ESP as Device ID 0x18, Product ID 215, Version 01.02.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.5.1",
    "platform": "ESP32",
    "uptime": "000+12:45:40.538",
    "uptime (seconds)": 45940,
    "free mem": 80,
    "max alloc": 39,
    "free app": 1,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -59,
    "IPv4 address": "REDACTED",
    "IPv4 gateway": "REDACTED",
    "IPv4 nameserver": "REDACTED",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": true,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Etc/UTC"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 9980,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 2,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 20682,
    "bus reads (tx)": 2312,
    "bus writes (tx)": 765,
    "bus incomplete telegrams": 184,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 4,
    "bool dashboard": 3,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 49
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "12.16",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A"
    },
    {
      "type": "controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "12.16",
      "entities": 0
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x18",
      "product id": 215,
      "version": "01.02",
      "entities": 0,
      "handlers ignored": "0x06 0x02A5 0x1A 0x35"
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

Nothing - just go to the Dashboard on default settings.

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

The device to be recognised.

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

I also saw some messages that don't look like they're being decoded:

```
000+00:07:21.274 N 163: [emsesp] unknown(0x18) -B-> All(0x00), ?(0x02A5), data: 00 CF 30 00 00 00 0A
000+00:07:25.053 N 164: [emsesp] unknown(0x18) -B-> All(0x00), ?(0x06), data: 17 09 17 01 26 00 04 01
000+00:07:25.265 N 165: [emsesp] unknown(0x18) -B-> All(0x00), ?(0x02A5), data: 00 CF 30 00 00 00 0A
```

The docs suggest that 0x06 is RCTime so I'm not sure why that's _not_ being decoded? Any idea about 0x02A5?

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2023-09-02:
Could you upgrade to 3.6.1? https://github.com/emsesp/EMS-ESP32/releases/tag/latest

I see the a Comfort RF was added with product ID 215, so it may pop up.

basak, 2023-09-04:
Ah, thanks. This worked. I had looked at the release notes and didn't see anything related to this, so hadn't realised support had been added.

Unfortunately the values are read-only but I suspect that's the thermostat.

closed 2023-09-04 by basak