emsesp/EMS-ESP32 #1280: MQTT enum index option leads to error in HA

MichScha, 2023-09-03:
### PROBLEM DESCRIPTION

I have set enum format to index but it seems that HA doesn´t like it. The discovery option for HA should not be affected by this setting but from logs I get the following error which seems that the index is been used instead of value?

Example
``` LUA
2023-09-03 22:01:43.610 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_dhw_disinfection_day: '1' (valid options: ['mo', 'tu', 'we', 'th', 'fr', 'sa', 'su', 'all'])
```

@proddy mentioned: Setting the enum format to Index doesn't change the MQTT Discovery topics (the options attribute). Would it work for you if we fix this and use the index's as well here instead of the real values?

I am currently not too familiar with HA if this is a good idea :-D From OH perspective I definitly love the index version as a mapping of enum is pretty simple and well implemented.

### REQUESTED INFORMATION
``` LUA
"System Info": {
    "version": "3.6.1-dev.0",
    "platform": "ESP32",
```

proddy, 2023-09-08:
fixed in 3.6.1-dev.2

MichScha, 2023-09-08:
Just tested - getting now less errors but still some. String vs Number
``` bash
2023-09-08 22:20:42.124 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_dhw_disinfection_day: '1' (valid options: [0, 1, 2, 3, 4, 5, 6, 7])
2023-09-08 22:20:42.138 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_mode: '0' (valid options: [0, 1])
2023-09-08 22:20:42.139 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_nofrost_mode: '1' (valid options: [0, 1, 2])
2023-09-08 22:20:42.139 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_control_mode: '0' (valid options: [0, 1, 2, 3, 4, 5])
2023-09-08 22:20:42.140 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_program: '0' (valid options: [0, 1])
2023-09-08 22:20:42.140 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_reduce_mode: '2' (valid options: [0, 1, 2])
2023-09-08 22:20:42.141 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_hp_mode: '1' (valid options: [0, 1, 2])
```

proddy, 2023-09-08:
Thanks for testing. I think we need to rethink the logic. Although there is probably a use case to use enum index values instead of the string value in the mqtt payload, but it messes up the loveliness of home assistant. Perhaps we should publish both enum index & string values in the _data payload?

proddy, 2023-09-09:
made a fix, so the indexes are seen as strings. I still think it looks strange, but now no errors hopefully.

MichScha, 2023-09-10:
No errors now with latest version - Thanks! Just as I am not that familiar with HA - how do you map the enum index to a string or should for HA the mqtt discovery always map the real string for the index without using the enum setting (index)?

Just another thing I currently don´t understand
In HA I see 3 devices where the first one is the device itself. But there should be at least 4 devices based on my setup (EMS-ESP, Boiler, Thermostat, Room-Thermostat. Or is it the idea to collect all thermostat devices in one HA-Device?
[image]
[image]

proddy, 2023-09-11:
> How do you map the enum index to a string or should for HA the mqtt discovery always map the real string for the index without using the enum setting (index)?

You will need to create a [template](https://www.home-assistant.io/blog/2023/09/06/release-20239/#template-sensors-from-the-ui) yourself and code the logic. But can you remind me why you wanted to use Index's in the first place?

> Just another thing I currently don´t understand In HA I see 3 devices where the first one is the device itself. But there should be at least 4 devices based on my setup (EMS-ESP, Boiler, Thermostat, Room-Thermostat. Or is it the idea to collect all thermostat devices in one HA-Device?

I'm not sure anymore. Probably you'll see all the data in HA as separate heating circuits?

closed 2023-09-18 by proddy