emsesp/EMS-ESP32 #1281: Support AnalogRead in Analog Sensors

proddy, 2023-09-04:
Davini on the Discord's hardware [channel](https://discord.com/channels/816637840644505620/816958486394830870/1147924917409558538) wants to use the Sensors to read an LDR. As LDR measures resistance (not voltage) we need an `AnalogRead()` and a new Type.

MichaelDvP, 2023-09-04:
We can't measure resistance direct, we always measure voltage and have to use ohms and kirchhofs laws.  For measure we need additionl circuits:
- use a constant current source Ic to the ldr and measure voltage R=U/Ic  => enter 1/Ic as factor. For this we can add `Ohms`as uom. But we don't have a const current source on ems-esp.
- use a constant resistor Rc in series and connect to const voltage Uc, But this is non-linear and we need Rc and Uc to calculate, 

But if someone uses a LDR he's not interrested in resistant, he want's to measure light intensity, This is extra calculation and depends on optics what uom is usefull (candela/lumen).

I think better measure voltage and make the calculation in the homeautomation-system.

Or is the LDR only for reading LED-pulses (water/gas/el.energy meters)?

davini-dev, 2023-09-04:
I would like to get this value from a gpio board controller...not from the ems-esp interface.

In this hardware:
https://github.com/Xinyuan-LilyGO/T-Display-S3

proddy, 2023-09-04:
I was thinking if one pin of the LDR was connected to +V and the other to a GPIO with a pullup you could read the values with analogRead(gpio)

MichaelDvP, 2023-09-04:
> In this hardware:

And what Sensor?

bbqkees, 2023-09-04:
Use a resistor divider connected to one of the ADC pins and read out the value.
A LDR is not accurate at all so there is no real use to have Lux as output, just send the mV value to HA and do the conversion there.
https://www.tweaking4all.com/hardware/arduino/arduino-light-sensitive-resistor/

davini-dev, 2023-09-04:
[image]

davini-dev, 2023-09-04:
I'm trying to use the sensor above with liligo s3

MichaelDvP, 2023-09-04:
Ok, connect Vcc to 3.3V and gnd. Set the gpio for D0 to Sensortype "digitial in" to get the light-on/off signal, or  A0 to sensortype "ADC".

davini-dev, 2023-09-05:
oops...all working now...thanks for the clarification and support

proddy, 2023-09-05:
great

closed 2023-09-05 by proddy