emsesp/EMS-ESP32 #1282: HA MQTT Discovery fails to recognize 't' abbreviation

oskars-viksna, 2023-09-04:
### PROBLEM DESCRIPTION

HA MQTT Discovey fails to recognize 't' abbreviation used by EMS-ESP and this causes devices not to show up in HA. Seems to be caused by this forgotten HA bug: https://github.com/home-assistant/core/issues/66244

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

Also sent email to support and from response understood I am the first one to report something like this.

```lua
"version": "3.6.0",
"platform": "ESP32",

* can provide more details upon request, does not see how those will help with this particular issue at this point
```

### TO REPRODUCE

Nothing special, just enable HA integration

### EXPECTED BEHAVIOUR

Devices show up in HA

### SCREENSHOTS

n/a

### ADDITIONAL CONTEXT

EMS for example tries to register device for HA MQTT Discovery:
```json
{
"uniq_id": "boiler_heatingactive",
"obj_id": "boiler_heatingactive",
"stat_t": "ems-esp/boiler_data",
"name": "Heating active",
"pl_on": "on",
"pl_off": "off",
"val_tpl": "{{value_json.heatingactive if value_json.heatingactive is defined else 'off'}}",
"dev": {
"ids": [
"ems-esp-boiler"
],
"name": "ems-esp Boiler",
"mf": "Buderus",
"mdl": "GBx72/Trendline/Cerapur/Greenstar Si",
"via_device": "ems-esp"
},
"avty": [
{
"t": "ems-esp/status",
"val_tpl": "{{'online' if value == 'online' else 'offline'}}"
},
{
"t": "ems-esp/boiler_data",
"val_tpl": "{{'online' if value_json.heatingactive is defined else 'offline'}}"
}
],
"avty_mode": "all"
}
```

And that causes exception in HA:

```
2023-09-03 08:31:55 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_binary_sensor_mqtt': ({'unique_id': 'boiler_heatingactive', 'object_id': 'boiler_heatingactive', 'state_
topic': 'ems-esp/boiler_data', 'name': 'Heating active', 'payload_on': 'on', 'payload_off': 'off', 'value_template': "{{value_json.heatingactive if value_json.heatingactive is defined else 'off'}}", 'device': {'identifiers': ['ems-esp-bo
iler'], 'name': 'ems-esp Boiler', 'manufacturer': 'Buderus', 'model': 'GBx72/Trendline/Cerapur/Greenstar Si', 'via_device': 'ems-esp'}, 'availability': [{'t': 'ems-esp/status', 'val_tpl': "{{'online' if value == 'online' else 'offline'}}
"}, {'t': 'ems-esp/boiler_data', 'val_tpl': "{{'online' if value_json.heatingactive is defined else 'offline'}}"}], 'availability_mode': 'all', 'platform': 'mqtt'},)
Traceback (most recent call last):
File "/opt/homeassistant/lib/python3.9/site-packages/homeassistant/components/mqtt/mixins.py", line 333, in async_discover
config = schema(discovery_payload)
File "/opt/homeassistant/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/opt/homeassistant/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
return base_validate(path, iteritems(data), out)
File "/opt/homeassistant/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['availability'][0]['topic']
```

### SUGGESTED FIX

I would suggest to fix this by just avoiding using HA abbreviations, other MQTT Discovery devices I have are using full names and everything works as expected. If there is a reason why abbreviations are used, then I would suggest implementing option to switch between abbreviated or full names.

MichaelDvP, 2023-09-04:
I think abberations are mainly used to save flash/mqtt-buffer. no conflict to change to fullnames. Thanks.

proddy, 2023-09-04:
Strange, I've never seen this error before and it hasn't been reported earlier. Which version of Home Assistant are you using?

Discovery abbreviations for HA taken from https://www.home-assistant.io/integrations/mqtt/#discovery-payload

oskars-viksna, 2023-09-04:
I see those abbreviations in HA documentation and do not get why this fails and it seems just for 't' and 'val_tpl' keys. From output (where you can see some keys are replaced) it seems, that it fails to replace deeper keys while first level keys get replaced correctly. And from linked issue it seems I am not the only one (that is another integration, not EMS-ESP) crashing into this :)
I was on HA 2022.6.6 core on Debian 11. After getting into this I tried to update to 2023.1.7 claiming to be the last one supporting Python 3.9 (available in Debian 11 with regular system packages), but it failed because of dependencies failing to install properly and I reverted. Then I went to step-by-step upgrades to 2022.6.7 -> 2022.7.0 -> 2022.7.7 and all of these have exactly the same issue. I will try to continue these step-by-step upgrades in few days, but I doubt this issue will disappear.

proddy, 2023-09-04:
i know it's not useful, but would strongly recommend you jump onto HA 2023.8.x as its a big release with plenty of cool stuff in it. It would mean upgrading your Debian distro to 'bookworm' which will contain Python 3.11

oskars-viksna, 2023-09-05:
Horray! I continued upgrade path step-by-step and this issue IS fixed with HA release 2022.10.2 - https://github.com/home-assistant/core/issues/79643 - they were really not processing abbreviations for deeper json levels.
While my issue is solved now, I would still recommend this abbreviation change for people stuck on older HA versions.

proddy, 2023-09-06:
@MichaelDvP are you sure if we switch back to the full names (not abbreviations) for MQTT Discovery attributes it won't blow up the heap, although we're not using queues?

MichaelDvP, 2023-09-06:
> @MichaelDvP are you sure if we switch back to the full names (not abbreviations) for MQTT Discovery attributes it won't blow up the heap, although we're not using queues?

I thought only the "t" should expand to "topic", thats12 bytes per HA-message, json alloc is the same and maybe we can cache one message less before reaching memory limit, that should not happen in normal environment. 
Agree: Expanding all abbreviation is not usefull.

closed 2023-09-06 by proddy