emsesp/EMS-ESP32 #1295: EMS-ESP 3.6.1/3.7.0.dev0 freeze after hours/days - no response from ems-esp

ralf-koe, 2023-09-14:
### PROBLEM DESCRIPTION
Similar and possibly related to
> EMS-ESP 3.6.0 crashes after hours/days - no response from ems-esp #1264 

As explained in discord
The discussion in the github threat from above points to WLAN issues. My EMS is connected via LAN, with version 3.6.1 and 3.7.0.dev0 it stopps working after minutes to ~2 hour. 

Memory consumption is inconspicuous. 

After freeze no MQTT data send, no Web access is possible, but no http error, just not reachable.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
possibly related to
> EMS-ESP 3.6.0 crashes after hours/days - no response from ems-esp #1264 

- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

I am currently back to v3.5.1 which running stable.

```lua
  System information output here:
{
  "type": "settings",
  "System": {
    "version": "3.5.1"
  },
  "Network": {
    "ssid": "",
    "password": "",
    "hostname": "ems-esp",
    "static_ip_config": true,
    "enableIPv6": false,
    "bandwidth20": false,
    "tx_power": 20,
    "nosleep": false,
    "enableMDNS": false,
    "enableCORS": false,
    "CORSOrigin": "*",
    "local_ip": "172.16.100.191",
    "gateway_ip": "172.16.100.1",
    "subnet_mask": "255.255.255.0",
    "dns_ip_1": "172.16.100.232",
    "dns_ip_2": "172.16.100.1"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "password": "ems-esp-neo",
    "channel": 1,
    "ssid_hidden": false,
    "max_clients": 4,
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "172.16.100.3",
    "port": 1885,
    "base": "ems-esp",
    "username": "",
    "password": "",
    "client_id": "ems-esp",
    "keep_alive": 59,
    "clean_session": false,
    "entity_format": 0,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "publish_time_heartbeat": 60,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "ha_enabled": false,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "discovery_type": 0,
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
  "NTP": {
    "enabled": true,
    "server": "172.16.100.1",
    "tz_label": "Europe/Berlin",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": false,
    "port": 8266,
    "password": "ems-esp-neo"
  },
  "Security": {
    "jwt_secret": "ems-esp-neo",
    "users": [
      {
        "username": "admin",
        "password": "Rk!603803",
        "admin": true
      }
    ]
  },
  "Settings": {
    "version": "3.5.1",
    "locale": "en",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "trace_raw": false,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "boiler_heatingoff": false,
    "shower_timer": false,
    "shower_alert": false,
    "shower_alert_coldshot": 10,
    "shower_alert_trigger": 7,
    "rx_gpio": 5,
    "tx_gpio": 17,
    "dallas_gpio": 4,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "low_clock": false,
    "telnet_enabled": true,
    "notoken_api": false,
    "readonly_mode": false,
    "analog_enabled": false,
    "pbutton_gpio": 33,
    "solar_maxflow": 30,
    "board_profile": "E32",
    "fahrenheit": false,
    "bool_format": 4,
    "bool_dashboard": 1,
    "enum_format": 1,
    "weblog_level": 6,
    "weblog_buffer": 75,
    "weblog_compact": true,
    "phy_type": 1,
    "eth_power": 16,
    "eth_phy_addr": 1,
    "eth_clock_mode": 0,
    "platform": "ESP32"
  }
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_
Use 3.6.1/3.7.0.dev0  and wait

### EXPECTED BEHAVIOUR

stable running and no freeze

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ralf-koe, 2023-09-14:
Shit, sorry, I forgot, we've been through this before.  

`{
  "System Info": {
    "version": "3.5.1",
    "platform": "ESP32",
    "uptime": "001+01:38:55.613",
    "uptime (seconds)": 92335,
    "free mem": 136,
    "max alloc": 83,
    "free app": 1,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "C4:DE:E2:B2:63:33",
    "IPv4 address": "172.16.100.191/255.255.255.0",
    "IPv4 gateway": "172.16.100.1",
    "IPv4 nameserver": "172.16.100.232",
    "static ip config": true,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": false,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "172.16.100.1",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 29323,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 59,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 131762,
    "bus reads (tx)": 25899,
    "bus writes (tx)": 4032,
    "bus incomplete telegrams": 3,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "E32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 4,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 75,
    "web log buffer": 17
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 5000i/Greenstar 8000/GC9800IW",
      "device id": "0x08",
      "product id": 195,
      "version": "05.06",
      "entities": 66,
      "handlers received": "0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xD7 0xD8 0x02D6 0x2E 0x3B 0xF9 0x17 0x36 0xF7"
    },
    {
      "type": "thermostat",
      "name": "UI800",
      "device id": "0x10",
      "product id": 4,
      "version": "49.03",
      "entities": 46,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x02D0 0x02D2 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xBF 0xE7 0x02E0 0x02EA 0xC4 0xF9 0x3B"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "06.02",
      "entities": 0,
      "handlers ignored": "0x0CBD"
    }
  ]
}`

Marcwa19197, 2023-09-26:
Hi,
im having a similar issue using 3.6.1. After upgrade the device seems to be "unstable" it works for a few days but then no data is transmitted, device webinterface is not accessable till i disconnect it from the maintenance port and reconnect it again (AUX-Cable).

Here is my device info:

`{
    "System Info": {
        "version": "3.6.1",
        "platform": "ESP32",
        "uptime": "000+00:09:43.575",
        "uptime (seconds)": 583,
        "free mem": 132,
        "max alloc": 87,
        "free app": 6149,
        "reset reason": "Power on reset / APP CPU reset by PRO CPU"
    },
    "Network Info": {
        "network": "WiFi",
        "hostname": "ems-esp",
        "RSSI": -49,
        "IPv4 address": "192.168.10.103/255.255.255.0",
        "IPv4 gateway": "192.168.10.1",
        "IPv4 nameserver": "192.168.10.104",
        "static ip config": false,
        "enable IPv6": false,
        "low bandwidth": false,
        "disable sleep": false,
        "enable MDNS": true,
        "enable CORS": false,
        "AP provision mode": "never",
        "AP security": "wpa2",
        "AP ssid": "ems-esp"
    },
    "NTP Info": {
        "NTP status": "connected",
        "enabled": true,
        "server": "192.168.10.80",
        "tz label": "Europe/Berlin"
    },
    "OTA Info": {
        "enabled": true,
        "port": 8266
    },
    "MQTT Info": {
        "MQTT status": "connected",
        "MQTT publishes": 456,
        "MQTT queued": 0,
        "MQTT publish fails": 0,
        "MQTT connects": 1,
        "enabled": true,
        "client id": "ems-esp",
        "keep alive": 60,
        "clean session": false,
        "entity format": 0,
        "base": "ems-esp",
        "discovery prefix": "homeassistant",
        "discovery type": 0,
        "nested format": 2,
        "ha enabled": true,
        "mqtt qos": 0,
        "mqtt retain": false,
        "publish time heartbeat": 60,
        "publish time boiler": 10,
        "publish time thermostat": 10,
        "publish time solar": 10,
        "publish time mixer": 10,
        "publish time other": 10,
        "publish time sensor": 10,
        "publish single": false,
        "publish2command": false,
        "send response": false
    },
    "Syslog Info": {
        "enabled": false
    },
    "Sensor Info": {
        "temperature sensors": 4,
        "temperature sensor reads": 464,
        "temperature sensor fails": 0,
        "analog sensors": 0,
        "analog sensor reads": 0,
        "analog sensor fails": 0
    },
    "API Info": {
        "API calls": 0,
        "API fails": 0
    },
    "Bus Info": {
        "bus status": "connected",
        "bus protocol": "Buderus",
        "bus telegrams received (rx)": 752,
        "bus reads (tx)": 157,
        "bus writes (tx)": 0,
        "bus incomplete telegrams": 0,
        "bus reads failed": 0,
        "bus writes failed": 0,
        "bus rx line quality": 100,
        "bus tx line quality": 100
    },
    "Settings": {
        "board profile": "S32",
        "locale": "de",
        "tx mode": 1,
        "ems bus id": 11,
        "shower timer": true,
        "shower alert": false,
        "hide led": false,
        "notoken api": false,
        "readonly mode": false,
        "fahrenheit": false,
        "dallas parasite": false,
        "bool format": 6,
        "bool dashboard": 5,
        "enum format": 1,
        "analog enabled": true,
        "telnet enabled": true,
        "max web log buffer": 10,
        "web log buffer": 10
    },
    "Devices": [
        {
            "type": "boiler",
            "name": "GB125/GB135/MC10",
            "device id": "0x08",
            "product id": 72,
            "version": "02.07",
            "entities": 71,
            "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
            "handlers fetched": "0x14 0x16 0x33",
            "handlers pending": "0xBF 0xC2 0x1A 0x35",
            "handlers ignored": "0x1F 0x20"
        },
        {
            "type": "thermostat",
            "name": "RC35",
            "device id": "0x10",
            "product id": 86,
            "version": "01.08",
            "entities": 56,
            "handlers received": "0xA3 0x06 0xA2 0x12 0x42 0x48 0x47 0x52 0x51 0x5C 0x5B 0x38 0x39",
            "handlers fetched": "0x3E 0x3D 0x3F 0xA5 0x37",
            "handlers pending": "0x13 0x49 0x4C 0x53 0x56 0x5D 0x60",
            "handlers ignored": "0x1A 0x35"
        },
        {
            "type": "controller",
            "name": "BC10/RFM20",
            "device id": "0x09",
            "product id": 68,
            "version": "02.03",
            "entities": 0,
            "handlers ignored": "0x29"
        }
    ]
}`

proddy, 2023-09-26:
Hi @Marcwa19197 , there was an issue in 3.6.1 with WiFi mesh's (specifically AVM Fritzbox) that we think is solved in 3.6.2 which is under the dev releases. Do you have have a similar setup?

Marcwa19197, 2023-09-26:
> Hi @Marcwa19197 , there was an issue in 3.6.1 with WiFi mesh's (specifically AVM Fritzbox) that we think is solved in 3.6.2 which is under the dev releases. Do you have have a similar setup?

Hi, no im using a TP-Link Access Point in the basement.

proddy, 2023-09-26:
we're tracking this in https://github.com/emsesp/EMS-ESP32/issues/1264

closed 2023-09-26 by proddy