emsesp/EMS-ESP32 #1301: minor translation issue, missing Warmwater in some names

harvey637, 2023-09-20:
Hi,

just a reminder:

some (german) translation use the same wording for eg. "total consumption" AND "total consumption warmwater"
... the additional word ".... Verbrauch Warmwasser" would clarify this without checking the internal names. Also this leads to some duplicate german names. With memory constraints this could also be an additional WW :-)

[image]

thank you
harvey

MichaelDvP, 2023-09-20:
I think they should be tagged with DHW/WW, added to my testbuild

MichaelDvP, 2023-09-29:
With the tags added we have some dhw values with double dhw in name now:
[image]
Sometimes it also depends on translation.

proddy, 2023-09-30:
I think that should be fine, unless someone complains

closed 2023-10-01 by proddy