emsesp/EMS-ESP32 #1305: Wrong assignement of hot water temperatur sensors on Buderus GB192 iT?

sanschaussure, 2023-09-21:
Dear all,

I just installed a EMS-ESP (v3.6.1) on a GB192 iT (RC300, 2x MM100, layered hot water storage). I tried to understand the different hot water temperature information by logging everything by influxdb.

I think there is a wrong naming in the intern and extern sensor (web interface). From the temperature slope the intern one seems to be outside the storage vessel while the extern one seems inside. 

According to this document: [Planungsunterlage Logamax plus GB192i/GB192iT](https://productsde.buderus.com/techdoc/Logamax-plus-GB192i/6720862178.pdf), page 16, Bild 9: Nb. 19 should be the internal, 20 the external one.

Or is there a different meaning behind intern and extern?

By the way, why are there different between web interface and mqtt message? 

Regards

proddy, 2023-09-21:
thanks for repoting, we'll check

> By the way, why are there different between web interface and mqtt message?

what do you mean? the english shortnames vs the long translated name or the values?

sanschaussure, 2023-09-21:
> what do you mean? the english shortnames vs the long translated name or the values?

In Web interface, set to Englisch, also language for device entities:

dhw current intern temperature, dhw current extern temperature, dhw storage intern temperature, dhw storage extern temperature

In MQTT message:

wwcurtemp, wwcurtemp2, wwstoragetemp1, wwstoragetemp2

"ww" vs. "dhw" is ok, "temp" and "temp1" for "intern" and "extern" is confusing.

proddy, 2023-09-21:
we use shortnames in the MQTT payload to save on memory as there can be more than 150 entities and that would break MQTT.

sanschaussure, 2023-09-22:
Ok, thank you for the explanation. That fine.

closed 2023-09-26 by proddy