emsesp/EMS-ESP32 #1320: Flashing 3.6.2 or 3.6.3 not possible (but 3.6.1 is working) -> loop "Brownout detector was triggered"

giovanne123, 2023-10-07:
Hello,
someone an idea for a "Brownout detector was triggered" I get when I try to flash 3.6.2.* or 3.6.3.*. (see below serial out)
It loops with this error.

3.6.1 is working fine (beside the #1264 problem after some days)

I have tried different usb cables, I have tried different power supply sources. Tried now already for some days with same result.
Everytime the same and with 3.6.1 everything is working.
(ESP32 with bbqboard)

I can directly flash back to 3.6.1 and it directly starts working again.

Is there something changed in >3.6.1 which may be the problem for me?

```
┌────────────────────────────────────────┐│ ␛[1mEMS-ESP version 3.6.3-dev.1 ␛[22m   
        │
│ ␛[0;92m␛[4mhttps://github.com/emsesp/EMS-ESP32␛[0m    │
│                                        ││ type ␛[4mhelp␛[0m to show available commands   │
└────────────────────────────────────────┘
␛[0;92m␛[1mems-esp␛[0m:$
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

┌────────────────────────────────────────┐│ ␛[1mEMS-ESP version 3.6.3-dev.1 ␛[22m   
        │
│ ␛[0;92m␛[4mhttps://github.com/emsesp/EMS-ESP32␛[0m    │
│                                        ││ type ␛[4mhelp␛[0m to show available commands   │
└────────────────────────────────────────┘
␛[0;92m␛[1mems-esp␛[0m:$
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

┌────────────────────────────────────────┐│ ␛[1mEMS-ESP version 3.6.3-dev.1 ␛[22m   
        │
│ ␛[0;92m␛[4mhttps://github.com/emsesp/EMS-ESP32␛[0m    │
│                                        ││ type ␛[4mhelp␛[0m to show available commands   │
└────────────────────────────────────────┘
␛[0;92m␛[1mems-esp␛[0m:$
Brownout detector was triggered

```

MichaelDvP, 2023-10-07:
Brownout means the supply voltage goes slowly down (blackout == fast decrease). 3.6.2 makes a full network scan befroe connecting, this needs more power for the wifi. Check your power supply.

giovanne123, 2023-10-07:
Thanks for the answer.
I will check again tomorrow ... but already tried with 2A power supply and others. So normal should be enough power.

For a first test I have flashed a new esp32 without bbqboard connected and with one of the same power source and this ems-esp is working with 3.6.3.*. (without bbqboard)

I need to check at productive esp32 where the bbqboard is connected, not sure how it was connected (power from esp?...) and maybe there is my fault 🤔 (for >3.6.1)

MichaelDvP, 2023-10-09:
What esp-board do you use? How old? Some USB-chargers have high ripple (doesn't mind for charging applications) and fast aging the capacitiors on the LDO.
Try to add a capacitor to 3.3V and 5V.

bbqkees, 2023-10-09:
This looks like a crappy ESP32 board or a crappy power supply or cable.

But you get this message while flashing, so I assume it is connected to the PC via a USB cable.
Try another USB port or PC. Perhaps it does not supply enough power. Some USB ports may limit the power to 100mA, this is not enough for the ESP32 to boot.

giovanne123, 2023-10-09:
Thanks for your answers.
I still in Investigation and will check your recommendations...

Maybe bad description/title, I think it is after flashing not flashing itself. Because already tried a 2A power supply anf/or flashed ota without usb connected to pc.
I have already tried a different new esp32 board and it was working but without bbqboard connected.
On the productive (old) esp32 I also tried with disconnected bbqboard and disconnected dallas sensor with same result.

Next I will replace the old esp32 with the new one and connect bbqboard to it...
maybe old is defective.. (also old esp32 is an older hw version)

giovanne123, 2023-10-09:
Update: Problem solved.
After some research and several comments in e.g. github, also after the hint of MichaelDvP 
in an first step (before replacing the esp32) I added a capacitor (used 100uF) to 3.3V and 5V.
Now the ems-esp is working again :-) (currently 3.6.3-dev1)

Will have a look if it is stable now and there is no need to switch to new esp32 ... 

For information:

- current Wemos D1 esp32 (with added capacitors) "Chip is ESP32-D0WDQ6 (revision 1)" 
- new one (so far not switched to it only tested without bbqboard, was working fine without capacitors) "Chip is ESP32-D0WD-V3 (revision 3)"

closed 2023-10-09 by giovanne123

giovanne123, 2023-11-03:
Status info. Everything fine.
(still using https://github.com/emsesp/EMS-ESP32/issues/1320#issuecomment-1753024536)
After attaching the capacitors my ems-esp is running now for 25days without issue and maybe sideeffect (or used fw is fine now) I have no longer freezes of the esp anymore 😊

proddy, 2023-11-04:
thanks @giovanne123 for reporting back.