emsesp/EMS-ESP32 #1322: Cannot write certain values to wwcirc via MQTT/API/Command line

ptwz, 2023-10-08:
### PROBLEM DESCRIPTION

When using `wwcirc` command, the values `continuous` and `off` do not seem to written properly, the value inside the boiler does not change. Writing `1x3min` or `6x3min` work as expected.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "006+07:30:55.783",
    "uptime (seconds)": 545455,
    "free mem": 121,
    "max alloc": 61,
    "free app": 437,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "NTP Info": {
    "NTP status": "disconnected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 391027,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 2,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": true
  },

  "Syslog Info": {
   "enabled": true,
    "syslog started": true,
    "syslog level": "info",
    "syslog ip": "10.0.0.123",
    "syslog queue": 1
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 2,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 1168426,
    "bus reads (tx)": 263686,
    "bus writes (tx)": 16,
    "bus incomplete telegrams": 28,
    "bus reads failed": 2,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
 "Settings": {
   "board profile": "CUSTOM",
   "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "phy type": 0,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },

  "Devices": [
    {
      "type": "boiler",
      "name": "Logano GB125/KB195i/Logamatic MC110",
      "device id": "0x08",
      "product id": 133,
      "version": "02.10",
      "entities": 66,
      "handlers received": "0xBF 0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x08E4 0xF7 0x2E 0x3B 0x17 0x36"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
     "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 48,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xE7 0x02E1 0x02EB 0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0xD2 0xE2 0x0255 0xBF"
    },
    {
      "type": "solar",
      "name": "SM200/MS200",
      "device id": "0x30",
     "product id": 164,
      "version": "25.06",
      "entities": 44,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x035D 0x0380 0x038E 0x0391 0x035F 0x035C 0x0361",
      "handlers ignored": "0xE8 0x0368 0x0369 0x03A9 0x03AA 0xBF 0x038A"
    }
  ]
}
```

### TO REPRODUCE

* Send `wwcircmode 1x3min` using MQTT
* Read back `wwcircmode`, resulting in `1x3min`
* Send `wwcircmode off`
* Read back `wwcircmode`, resulting in `1x3min`
* Send `wwcircmode 6x3min`
* Read back `wwcircmode`, resulting in `6x3min`
* Send `wwcircmode continuous`
* Read back `wwcircmode`, resulting in `6x3min`

### EXPECTED BEHAVIOUR

* Send `wwcircmode 1x3min` using MQTT
* Read back `wwcircmode`, resulting in `1x3min`
* Send `wwcircmode off`
* Read back `wwcircmode`, resulting in `off` *EXPECTED*
* Send `wwcircmode 6x3min`
* Read back `wwcircmode`, resulting in `6x3min`
* Send `wwcircmode continuous`
* Read back `wwcircmode`, resulting in `continuous` *EXPECTED*

ptwz, 2023-10-08:
`/api/boiler/wwcircmode` yields:
```json
{
  "name": "wwcircmode",
  "fullname": "dhw circulation pump mode",
  "circuit": "ww",
  "value": "6x3min",
  "type": "enum",
  "enum": [
    "off",
    "1x3min",
    "2x3min",
    "3x3min",
    "4x3min",
    "5x3min",
    "6x3min",
    "continuous"
  ],
  "readable": true,
  "writeable": true,
  "visible": true
}```

MichaelDvP, 2023-10-08:
We know from RC35 that setting off does not work (see #727), it should be en/disabled elsewhere. But continious should work, try with sending number `wwcircmode 7`

ptwz, 2023-10-09:
> We know from RC35 that setting off does not work (see #727), it should be en/disabled elsewhere. But continious should work, try with sending number `wwcircmode 7`

This indeed works, whereas the name only works in the `?x3min` Syntax.

ptwz, 2023-10-09:
I noticed the problem also exists in scheduled commands.

MichaelDvP, 2023-10-09:
Sorry, can not reproduce. tried 3.6.2 and the dev, I can change between all on modes (1-7) with number, names and translated name in telnet and mqtt. Only found that we have now a warning log message for each successfull command, i think this should be a debug message.
How do you send the command?

ptwz, 2023-10-16:
Strange, now it works flawlessly. Please disregard this issue, it seems to have sorted itself out with some firmware upgrade.

closed 2023-10-16 by ptwz