emsesp/EMS-ESP32 #1324: ems-esp device does not respond after few hours

maciejka104, 2023-10-09:
Strarting from version 3.6.3 my bbqkees ems-esp S32 after few hours is unavalivable. 
Currently I am running 3.6.3-dev.1 and it works fors up to 6-8 hurs and then it becomes dead. 
No ping no mqtt, 

I have checked syslog and mqtt connected disconnected ....

<28>1 2023-10-09T18:02:52.085523+02:00 ems-esp mqtt - - - mqtt- 000+00:02:36.323 W 172: MQTT disconnected: TCP
<30>1 2023-10-09T18:02:56.770770+02:00 ems-esp mqtt - - - mqtt- 000+00:02:41.005 I 183: MQTT connected
<28>1 2023-10-09T18:32:57.233403+02:00 ems-esp mqtt - - - mqtt- 000+00:32:41.509 W 1407: MQTT disconnected: TCP
<30>1 2023-10-09T18:33:02.576616+02:00 ems-esp mqtt - - - mqtt- 000+00:32:46.851 I 1414: MQTT connected
<28>1 2023-10-09T18:37:59.882780+02:00 ems-esp mqtt - - - mqtt- 000+00:37:44.089 W 1617: MQTT disconnected: TCP
<30>1 2023-10-09T18:38:42.374151+02:00 ems-esp mqtt - - - mqtt- 000+00:38:26.580 I 1655: MQTT connected

<28>1 2023-10-09T04:24:54.310904+02:00 ems-esp mqtt - - - mqtt- 000+01:24:38.175 W 4586: MQTT disconnected: TCP
<30>1 2023-10-09T04:24:58.280836+02:00 ems-esp mqtt - - - mqtt- 000+01:24:42.143 I 4592: MQTT connected
<28>1 2023-10-09T04:54:58.504756+02:00 ems-esp mqtt - - - mqtt- 000+01:54:42.352 W 5886: MQTT disconnected: TCP
<30>1 2023-10-09T04:55:04.300629+02:00 ems-esp mqtt - - - mqtt- 000+01:54:48.147 I 5897: MQTT connected

My setup is EMS-ESP bbqkees connected to Junkers Cerapur  + FW100. 

What could be the problem? I have checked mqtt (on HA) and looks/works fine.

proddy, 2023-10-09:
Damn, I was hoping we had fixed this issue. It looks like EMS-ESP is going into a MQTT re-connect loop. Perhaps we need to add more delay on the retry @MichaelDvP ?

@maciejka104 did you notice a WiFi glitch or anything that time?

maciejka104, 2023-10-10:
@proddy - wifi looks ok 
I will double check on WLC logs 
wifi setup - cisco 9800 wlc + 3 x C9105

MichaelDvP, 2023-10-10:
The mqtt reconnects are not very often and in one case it took 49sec to reconnect. I found mqtt reconnects with IPv6 enabled on every change of IP (dynamic IP from provider) and fixed this in my testbuild [here](https://github.com/MichaelDvP/EMS-ESP32/commit/020bb628d6bd313262290f2cf183246d060da224).

I'm always missing the support info in most new issues. Without knowing the config it's difficult to analyse possible reasons.

Since we have a lot of stable ems-esp and only a few with "freezes" it's mainly something in the environment of the users. 
#1320 reminds me that the full network scan takes more power for wifi and low/unstable power supply could have these effects. 

Users with good rssi to next AP could test with reduced tx-power.

maciejka104, 2023-10-10:
IPv6 disabled on network and on the ems-esp 
wifi coverage it is really good (one of the APs is 5-6 meters away and thin wall)
before raising a new issue I have checked usb cable and power supply. 

I have external syslog server so I can provide full log for both wifi and ems-esp.

proddy, 2023-10-10:
let's continue this discussion in https://github.com/emsesp/EMS-ESP32/issues/1321