emsesp/EMS-ESP32 #1326: Exhaust temperature entity lost after update from 3.5.0->3.6.2

glerak, 2023-10-12:
### PROBLEM DESCRIPTION

After updated from 3.5.0->3.6.2, there is not longer "exhaust temperature" entity.
After downgrade back to 3.5.0, entity is back...

Device/description is "Boiler: Bosch Logamax Plus GB122/Condense 2300".

In 3.5.0 is reported as (not inspect in 3.6.2)
Device Details
    Type    3
    Name    Logamax Plus GB122/Condense 2300
    Brand    Bosch
    Device ID    0x08
    Product ID    234
    Version    05.05

MichaelDvP, 2023-10-13:
This was changed in #1150 (issue #1147). Could you please post your telegrams 0xe4 and 0xe5, in terminal do
`read 8 e4` and `read 8 e5` or in web log all until both telegrams are in log.Then we can see how we can read both locations without conflict.

glerak, 2023-10-13:
There is the output of them.
Thanks.

```
┌────────────────────────────────────────────┐
│ EMS-ESP version 3.5.0                    │
│ https://github.com/emsesp/EMS-ESP32        │
│                                            │
│ type help to show available commands       │
└────────────────────────────────────────────┘

ems-esp:$ read 8 e4
000+16:45:59.836 I 0: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C3 00 00 00 00 01 D2 00 00 00 00 00 00 FF 00 01 C3 00 00
000+16:46:00.083 I 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
ems-esp:$ read 8 e5
000+16:46:03.589 I 2: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 01 00 00 00 00 78 00 00 00 00 00 3E 9C 05 6C 9A 00 00 00 04 C7 80 00 2F C2 00 00
000+16:46:03.793 I 3: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 05 4B 9E (offset 29)
ems-esp:$ read 8 e4
000+16:46:07.202 I 4: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 C3 00 00 00 00 01 D2 00 00 00 00 00 00 FF 00 01 C3 00 00
000+16:46:07.415 I 5: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
ems-esp:$ read 8 e5
000+16:46:10.291 I 6: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 01 00 00 00 00 78 00 00 00 00 00 3E 9C 05 6C 9A 00 00 00 04 C7 80 00 2F C2 00 00
000+16:46:10.503 I 7: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 05 4B 9E (offset 29)
ems-esp:$
```

glerak, 2023-10-13:
There is it.
I'm tested it from 3.6.3-dev.1 version - is it OK? I can go back to 3.5.0...
Near T 344 is "Boiler Heating pump modulation" - water pump started.
Near T 489 is starting burner.
Near T  675 is burner stoped.
And near  T 895 is waterpump off.
All TRACEs i have from home assistant graphs (time of state changes). 
(sorry for my "english" btw).

```
┌────────────────────────────────────────┐
│ EMS-ESP version 3.6.3-dev.1            │
│ https://github.com/emsesp/EMS-ESP32    │
│                                        │
│ type help to show available commands   │
└────────────────────────────────────────┘

000+00:03:39.153 I 0: [shell] User session opened on console pty0
ems-esp:$ read 8 e4 1f
000+00:05:26.255 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
ems-esp:$ read 8 e4 1f
000+00:05:27.876 N 2: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
ems-esp:$ log all
Log level: ALL
000+00:06:18.440 T 3: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 4B 00 64 1E 00 00 00 50
000+00:06:18.646 T 4: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 4B 00 00 00 00 02 7D 00 00 00 00 00 00 FF 00 02 4B 00 00
000+00:06:18.835 T 5: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
000+00:06:19.145 T 6: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 01 F6 01 F6 00 00 80 00 46 00 00 00 02 00 A5 2F 00 0E DC 00 00 00 2F 02
000+00:06:28.409 T 7: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 4B 00 64 1E 00 00 00 50
000+00:06:28.617 T 8: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 4B 00 00 00 00 02 7C 00 00 00 00 00 00 FF 00 02 4B 00 00
000+00:06:28.807 T 9: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
000+00:06:29.036 T 10: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 01 F8 01 F8 00 00 80 00 46 00 00 00 02 00 A5 2F 00 0E DC 00 00 00 2F 02
000+00:06:38.465 T 11: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 4A 00 64 1E 00 00 00 50
000+00:06:38.672 T 12: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 4B 00 00 00 00 02 7D 00 00 00 00 00 00 FF 00 02 4B 00 00
000+00:06:38.860 T 13: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
000+00:06:39.090 T 14: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 01 F6 01 F6 00 00 80 00 46 00 00 00 02 00 A5 2F 00 0E DC 00 00 00 2F 02
000+00:06:48.405 T 15: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 02 4B 00 64 1E 00 00 00 50
000+00:06:48.612 T 16: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 02 4B 00 00 00 00 02 7C 00 00 00 00 00 00 FF 00 02 4B 00 00
000+00:06:48.801 T 17: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
000+00:06:49.030 T 18: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 01 F6 01 F6 00 00 80 00 46 00 00 00 02 00 A5 2F 00 0E DC 00 00 00 2F 02
000+00:06:52.506 T 19: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus(0xE5), data: 40 (offset 8)
000+00:06:52.709 T 20: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus(0xE5), data: 21 00 20 00 00 78 00 00 40 00 00 3E 9F 05 6C C1 00 00 00 04 C7 92 00 2F C3 64
000+00:06:52.896 T 21: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 (offset 11)
000+00:06:56.062 T 22: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 4D (offset 23)
000+00:06:56.245 T 23: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 50 (offset 23)
000+00:06:56.414 T 24: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 54 (offset 23)
000+00:06:56.601 T 25: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 5A (offset 23)
000+00:06:56.784 T 26: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 61 (offset 23)
000+00:06:56.972 T 27: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 65 (offset 23)
000+00:06:57.232 T 28: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 71 (offset 23)
000+00:06:57.412 T 29: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 79 (offset 23)
000+00:06:57.588 T 30: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 80 (offset 23)
000+00:06:57.873 T 31: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 86 (offset 23)
000+00:06:58.053 T 32: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 8B (offset 23)
000+00:06:58.233 T 33: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 8F (offset 23)
000+00:06:58.461 T 34: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 1E 02 94 64 00 02 00 02 7C 00 00 00 00 00 00 FF 00 02 94 00 00
[... cut ...]
000+00:19:08.951 T 688: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 CB 00 01 1D 00 00 00 00 01 0B 00 00 00 00 00 00 FF 09 01 1D 00 00
000+00:19:09.139 T 689: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 00 (offset 35)
000+00:19:09.369 T 690: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 2F 01 F5 01 F5 00 00 80 00 46 00 00 00 02 00 A5 2F 00 0E DC 00 00 00 2F 02
000+00:19:12.178 T 691: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 1F (offset 23)
ems-esp:$ log off
Log level: OFF
ems-esp:$
```

MichaelDvP, 2023-10-14:
Are you sure you have a exhaust sensor intalled? v3.5 reads telegram 0xE5, offset 6 as exhaust sensor, but the value in your log is zero if burner is off and very low while burner is on. Exhaust temperatures goes up fast to high values when burner starts and goes down rapid after boiler stop, but stay at room temperature and never go to zero. So the change in #1150 was right, but your boiler does not report offset 31 of telegram 0xE4.

glerak, 2023-10-14:
Well, I'm not sure (if there is sensor installed in boiler). 
But (checked now), there is nothing about exhaust temperature in sevice manual/service menu of boiler.
I'm just see - there is "exhaust temperatue" in 3.5.0 and in 3.6.2 not.
There is some of home assistant (MQTT) graphs from history - it looks believable for me. But...? I never used it for some automation or something...

[image]

glerak, 2023-10-25:
Well, it looks - its not exhaust temperature - the vaule displayed in 3.5 as a exhaust temperature.
[image]
It correspond with actual power of burner..

Snotmann, 2023-10-26:
For me the exhausttemp is missing too ... i can see the value in the service menu so far, but its not reabable anymore since the update

proddy, 2023-10-26:
> For me the exhausttemp is missing too ... i can see the value in the service menu so far, but its not reabable anymore since the update

can you run the same test as in https://github.com/emsesp/EMS-ESP32/issues/1326#issuecomment-1760891467

and also post your system info https://emsesp.github.io/docs/Support/

Snotmann, 2023-10-29:
@proddy for sure:

Here my systeminfo:
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32",
    "uptime": "002+17:01:31.450",
    "uptime (seconds)": 234091,
    "free mem": 136,
    "max alloc": 75,
    "free app": 6581,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -48,
    "IPv4 address": "192.168.2.72/255.255.255.0",
    "IPv4 gateway": "192.168.2.1",
    "IPv4 nameserver": "192.168.2.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.2.1",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 97775,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 20,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 3,
    "temperature sensor reads": 138875,
    "temperature sensor fails": 670,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 416488,
    "bus reads (tx)": 70276,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 14,
    "bus reads failed": 1,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": true,
    "notoken api": true,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 50
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GC7000F",
      "device id": "0x08",
      "product id": 132,
      "version": "02.10",
      "entities": 65,
      "handlers received": "0xBF 0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0x02D6 0x08E4 0xF7 0xF9 0x2D"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 48,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xE7 0x02E0 0x02EA 0xBF 0x1D 0xF7 0x2D"
    }
  ]
}

here the output from the registers:
ems-esp:$ read 8 e4
002+17:00:55.285 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 59 00 CC 1F 01 4F 3A 00 02 00 00 00 00 00 80 00 00 00 FF 00 01 4F 80 00
002+17:00:55.289 N 2: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 01 5B 80 00 80 00 01 36 (offset 27)
ems-esp:$ read 8 e5
002+17:01:03.192 N 3: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorSlowPlus(0xE5), data: 20 00 00 00 00 00 00 00 00 89 F5 01 E7 E5 00 16 B9 01 99 8D 00 83 48 64 (offset 2)

proddy, 2023-10-30:
@Snotmann we read the exhaust temp from telegram 0xE4 offset 31 which shows in your output that it doesn't exist. Can you run those two `read` commands again and at the same time take a photo of what Service Menu is showing you?

Snotmann, 2023-10-30:
Yeah the servicemenu shows --- at exhaust temperature ... thats crazy as i got the exhaust temperature before 4 weeks

proddy, 2023-10-31:
do have you examples of the the readings look like, so we can find them in the telegrams?

MichaelDvP, 2023-10-31:
I think reading this thread makes clear:
- We had a temperature from 0xE5, offset 6 show as `exhausttemp`
- We learned that the exhausttemp is in 0xE4 offset 31
- We do not know what the temperature in E5 offset 6 is  and doe not show it anymore-

@Snotmann 
You can add a custom entity for a SHORT, telegram 0xE5, offset 6 and name it exhausttemp. But it is a different temperaturemeasurement.

proddy, 2023-10-31:
> @Snotmann You can add a custom entity for a SHORT, telegram 0xE5, offset 6 and name it exhausttemp. But it is a different temperaturemeasurement.

it'll be 0?

Snotmann, 2023-11-01:
@MichaelDvP should i add a entity myself ... i dont know how to do so ... can you explain a bit more ?

proddy, 2023-11-01:
> @MichaelDvP should i add a entity myself ... i dont know how to do so ... can you explain a bit more ?

Have a look at https://emsesp.github.io/docs/Configuring/#creating-custom-entities and let us know if something needs improving or isn't clear.

Snotmann, 2023-11-01:
Soooo like this ?
[image]

MichaelDvP, 2023-11-01:
[image]

Snotmann, 2023-11-01:
Seems to be correct: {"exhausttemp":30.70}

Snotmann, 2023-11-01:
@MichaelDvP seems top bei the correct value. Heater is running the temperaure went up to a close value to my exhaust measurement

MichaelDvP, 2023-11-01:
When looking at https://github.com/emsesp/EMS-ESP32/issues/1326#issuecomment-1763120974 i see that if follows the burner power delayed, a point for exhaust temperature, But the temperature level is much below the flowtemp, i think exhaust should be higher, and it goes to zero if burner switches off, that's strange. But if the exhaust temperature in E4/31 is not published, we can show this instead.

proddy, 2023-11-02:
still interesting to match these values against the readings on the boiler's front panel...can we check @Snotmann ?

Snotmann, 2023-11-02:
@proddy i got a technican here these days doing a measurement and got 20-60 degree temperature on the exhaust with a external device. So the reading from the entity could be realistic. But on the frontpanel its not shown as far i could see .... maybe i will check this again if the heater is burning again in a couple of hours

Snotmann, 2023-11-02:
@proddy ok i looked to my CW400 and there was no value for exhausttemperature ...

proddy, 2023-11-04:
@MichaelDvP do we know enought to add to dev ?

MichaelDvP, 2023-11-04:
It's back in the dev2 with priority to E4/31 and show e5/6 only if e4 ist missing.

proddy, 2023-11-04:
great! I'll close this then.

closed 2023-11-04 by proddy