emsesp/EMS-ESP32 #1327: A32/3102 error - lack communication with mixer on Buderus GB072

andrzejzukowski, 2023-10-13:
### PROBLEM DESCRIPTION

I have bought gateway (Gateway S3-LR (Standard WiFi Edition V3 Long Range)) and connected it to my Buderud GB072. Everything was working fine until I turned on my floor heating. Then boiler started throwing A32/3102 error about problems with communication between the mixer. After few minutes the error dissapears. 

It happens when mixer wants to decrease the valvel position. It happens 2-3 per hour. After unplugging the gateway everything is fine, so clearly it is related to this device.

EMS-ESP Version
3.6.3-dev.1

MichaelDvP, 2023-10-13:
Please attach the support info from help page and describe exactly how the gateway is connected and how the other modules are connected to the bus, Used clamps, cable type, diameter, length. How is the gateway powered?

andrzejzukowski, 2023-10-14:
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/12904631/emsesp_info.txt)

Gateway is connected by mini jack cable provided by bbqkees, so I believe it is good quality. There is no additional power suply.

Both modules (mixed and thermostat) are connected by wire to EMS port on the boiler.

MichaelDvP, 2023-10-14:
The jack-cable from BBQKees normally works good if both connectors are fully plugged in. Check that.
Is the thermostat mounted on boiler or with long cable in another room? Check cables and connectors (clamp-screws) to mixer.
Try to separate the bus cables from power lines. That there are disturbances when the mixer-motor starts points to influence from power lines.
If that does not help it's worth a try to change polarity of the buslines on one device mixer or thermostat.

andrzejzukowski, 2023-10-16:
Thermostat is mounted in another room with quite a long cable. I checked connectors, they are ok.
Mixer in connected to power independently, how can I separate it in a different way?

I tried to connect DC12V in addition to jack, but it didn't help.

andrzejzukowski, 2023-10-16:
[image]

bbqkees, 2023-10-16:
Can you switch the EMS wires on the mixer module?

andrzejzukowski, 2023-10-17:
I tried to:
- change the polarity
- change to the other bus connectors (on MM100 there are 2).
Nothing helped.

andrzejzukowski, 2023-10-17:
Sometimes I see those errors:
000+00:00:38.202 T 118: [telegram] Incomplete Rx: 0B 90 FF 00 19 01 AC FE
But it isn't  correlated with occurance of A32 error.

andrzejzukowski, 2023-10-18:
Now I see errors don't appear only when something happens with valve. Those peaks on the second chart depitc this events, because MM100 switches to "standalone" mode, without value driven from RC300.

[image]

MichaelDvP, 2023-10-18:
Some errors in bus communication are normal. The bus is unshielded, not twisted pair, unterminated, etc. to make it plumber compatible. But this can cause some errors on the bus.  The more devices and the longer the cables the more errors on the bus. Bosch sells bus-repeaters that can be used for hight device-count.
 General rules to reduce disturbance:
- avoid Y-connections on the bus clamps if there are free clamps. E.G. try to connect the thermostat to the out-clamp of the MM100 and ems-esp to the service Jack. The the bus is in line: ems-esp->boiler->mm100->thermostat 
- use thick cables, min 0.5mm^2
- don't use single wires or twitst the wires, the 2 bus lines should not be seperated.
- try to leave a distance between power cables and bus lines 
- Interconnection bus cables between modules should be min. 20cm, shorter cables could cause reflections.

bbqkees, 2023-10-18:
I can send a new Gateway to rule out defective hardware but the S3-LR is not available now. Will take about another month.

proddy, 2023-10-18:
> Some errors in bus communication are normal. The bus is unshielded, not twisted pair, unterminated, etc. to make it plumber compatible. But this can cause some errors on the bus. The more devices [...]

@bbqkees this would be good input on your Wiki?

andrzejzukowski, 2023-10-20:
> I can send a new Gateway to rule out defective hardware but the S3-LR is not available now. Will take about another month.

I tried all possibilities:
- replaced cables (1mm2)
- changed connection to: gateway (jack) -> boiler -> mixer  -> thermostat. Boiler and thermostat connected to independent clamps on the mixer
- variety of polarization configuration

After verification I think I don't need an LR. I'll send you an e-mail.

enkido, 2023-10-22:
I got the same problem, every time the mixer tries to set a valve position i get the A32/3102 error.

andrzejzukowski, 2023-10-22:
Which gateway do you use?

enkido, 2023-10-22:
> Which gateway do you use? niedz., 22 paź 2023, 19:29 użytkownik enkido ***@***.***> napisał:

I think its the S3 (Ethernet + WiFi Edition) Gateway (Invoice syays E32 Gateway), Version 3.6.0

andrzejzukowski, 2023-10-22:
> > Which gateway do you use? niedz., 22 paź 2023, 19:29 użytkownik enkido _**@**_.***> napisał:
> 
> I think its the S3 (Ethernet + WiFi Edition) Gateway (Invoice syays E32 Gateway), Version 3.6.0

Ok, so it it also the new S3 version.
@bbqkees If enkido notices the same problem, it may be related to hardware somehow. It would be good to compare how it works on the V2.

bbqkees, 2023-10-22:
If it has Ethernet than its the E32 and not the S3.
The E32 has a very different circuit board design.

Can you disconnect the valve or pump on the MM100 for a short while and see
if the error is still there?

enkido, 2023-10-22:
>If it has Ethernet than its the E32 and not the S3.
>The E32 has a very different circuit board design.

It has Ethernet and  WiFi.

>Can you disconnect the valve or pump on the MM100 for a short while and see
>if the error is still there?
Sorry, but i can't disconnect anything. Everything is permanently installed.
The MM100 control light starts to flash and the diagnose section states there is communication error between the mixer modul and the control panel.

bbqkees, 2023-10-23:
Is this problem recent like after a certain update? Each one appears to run 3.6.x.

Some things to try out:

- If anyone can disconnect the problematic valve or pump from the MM100 for a bit until the MM100 turns it on or off. This way we can check for hardware interference.
OR
- In the EMS-ESP settings, Enable read-only mode. This will cause the Gateway NOT to send any data on the bus. This way we can check if EMS-ESP may interfere when the MM100 switches.

andrzejzukowski, 2023-10-23:
> * If anyone can disconnect the problematic valve or pump from the MM100 for a bit until the MM100 turns it on or off. This way we can check for hardware interference.

Before I turned on floor heating this problem, hadn't appeared.
I can turn off the floor heating and leave only radiators.

> * In the EMS-ESP settings, Enable read-only mode. This will cause the Gateway NOT to send any data on the bus. This way we can check if EMS-ESP may interfere when the MM100 switches.
I'll start with this.

bbqkees, 2023-10-23:
The MM100 will be controlling the floor heating circuit.

The MM100 has a port for a three-way valve VC1 and a heating circuit circulation pump PC1.
[image]

Depending on the configuration of the heating circuit, the MM100 will turn on the three way valve and/or the pump when the heating circuit needs heat. Temporarily disconnection the valve and then the pump could give more info whether the error still occurs or not.

But first trying out the read only mode is easier off course.

MichaelDvP, 2023-10-23:
Communication errors while motor is running points to emc interference between power lines and the bus.
Again, this time from a module manual:
[image]

andrzejzukowski, 2023-10-23:
@MichaelDvP I've got it. But after all changes (thicker cables, different connection roder, polarity, moving thermostat closer), ntothing helped.
The problem doesn't appear only when the gateway is disconnected, so clearly gateway impacts the whole system somehow. 

Right now I try to verify the read only mode.

andrzejzukowski, 2023-10-23:
> The MM100 has a port for a three-way valve VC1 and a heating circuit circulation pump PC1. [image]

Accorting to arrows directions. The EMS input to MM100 shall be the cable from thermostat, the output should go to the boiler.
Right?

MichaelDvP, 2023-10-23:
> Right?

No, the boiler is busmaster and connceted to the left ems port, the right goes to additional modules/thermostat/etc.

andrzejzukowski, 2023-10-23:
> > Right?
> 
> No, the boiler is busmaster and connceted to the left ems port, the right goes to additional modules/thermostat/etc.

Thanks. I ended up with this setup.

andrzejzukowski, 2023-10-23:
> * In the EMS-ESP settings, Enable read-only mode. This will cause the Gateway NOT to send any data on the bus. This way we can check if EMS-ESP may interfere when the MM100 switches.

read-only mode didn't helped. Still the error appears.

enkido, 2023-10-23:
> * In the EMS-ESP settings, Enable read-only mode. This will cause the Gateway NOT to send any data on the bus. This way we can check if EMS-ESP may interfere when the MM100 switches.

still got the error in read only mode.

andrzejzukowski, 2023-10-23:
> * If anyone can disconnect the problematic valve or pump from the MM100 for a bit until the MM100 turns it on or off. This way we can check for hardware interference.

It didn't help either. I disconnected cables to valve and after 3 hours got the error again.
It can't be related to interference with this cable.

andrzejzukowski, 2023-10-23:
@enkido what are your boiler and thermostat?

enkido, 2023-10-23:
> @enkido what are your boiler and thermostat?
GB172-14 
RC300
MM100
SM200

andrzejzukowski, 2023-10-27:
Unfortunatelly the gateway replecement didn't help.

MichaelDvP, 2023-10-27:
> it throws service unavailable.

You try to upload wrong binary, esp32 version on a esp32s3. Pick the right file.

andrzejzukowski, 2023-10-27:
> > it throws service unavailable.
> 
> You try to upload wrong binary, esp32 version on a esp32s3. Pick the right file.

True. But for 3.5.1 which I wanted to try there is only esp32 version.
I wanted something <3.6.0 as enkido has 3.6.0 and faces the same issue.

bbqkees, 2023-10-27:
So exchanging the S3 hardware here did not make a difference. Enkido has the E32, which has a different EMS circuit.

There must be many others with a MM100 or similar and no issues. So given we have only 2 cases with this problem that would still somehow point to a disturbance in the installed system of the user.

I am not sure what we can try next. I am tempted to get a (second hand) MM100 so I can measure and test it here.

Although there are these errors, the system still works right?

andrzejzukowski, 2023-10-27:
> Although there are these errors, the system still works right?

There are errors, but after few minutes is dissapear from the display. You can find them only in the errors history in service menu of RC300.
From the whole system perspective there is no bigger disturbances. But when I looked into details:
- not always, but for 50% of cases MM100 sets the temperature to max (30 degrees) and after few minutes it is reverted to normal, based on the heating curve. It looks like it turns some "manual mode" or something. See below diagram, the light blue line is the floor heating target temperature.
[image]
- sometimes this erros interrupts heating, but it starts after few minutes

So there may be other users who have this, but they haven't noticed it yet.

> There must be many others with a MM100 or similar and no issues. So given we have only 2 cases with this problem that would still somehow point to a disturbance in the installed system of the user.

Certainly there are, but we have 2 common factors:
- 3.6.0 or above
- MM100

3.6.0 was released in August, and now we have beginning of the heating season, maybe there are not so many users of >=3.6.0 and MM100 with switched on heating.
I don't expect people with older firmware to upgrate it very often.

Moreover as I wrote there may be users who didn't notice those errors, as itn's not so easy to see it.

> I am not sure what we can try next. I am tempted to get a (second hand) MM100 so I can measure and test it here.

I really would like to try out 3.5.1, but there is no binaries for S3. I don't know how to compile them.
Could you compile them and share with me?

andrzejzukowski, 2023-10-31:
I moved a thermostat close to a boiler, so all boiler, thermostat and MM100 are connected by cables around 70-100cm.
The issue is still in place. Only disconnecting the gateway helps.

andrzejzukowski, 2023-11-27:
@enkido any progress?
I've tried several thing, but nothing helped. I've tried:
- different MM100
- 2 different gateways from Bqqkees.

bbqkees, 2023-11-28:
Does it also happen when the Gateway is turned of but still connected to the bus?

andrzejzukowski, 2023-11-28:
> Does it also happen when the Gateway is turned of but still connected to the bus?

Is it possible to "switch off" the gateway from ems-esp software with jack connected?
Or do you mean to use standard cables to clamps and switch off power supply?

BTW. I oredered a used RC310 to eliminate next element from the riddle.

enkido, 2023-11-28:
> @enkido any progress? I've tried several thing, but nothing helped. I've tried:
> 
> * different MM100
> * 2 different gateways from Bqqkees.

I can try on the weekend.

hawell1, 2023-11-28:
> Does it also happen when the Gateway is turned of but still connected to the bus?

Yes, everything is fine if the power supply is not connected and the gateway is still connected to the bus.

- MM50
- RC300
- GB172T2 - V2

hawell1, 2023-12-01:
That's right, I've also had this problem since installing it.

I compared the RC300 error times to the values ​​in HA and they agree with the temperatures.
I just don't know if it's the cause or the effect

andrzejzukowski, 2023-12-05:
I've replaced RC310 with RC310. It didn't help.

So to sum up:
- replacing MM100
- replacing RC300 with RC310
- a lot of changes in cables configuration
- switching from EMS plus to either EMS or Hardware setting 
didn't helpes.

I have no idea how to proceed with it, but looking on @enkido and @hawell1 issues it sems that it is a software issue.

enkido, 2023-12-05:
> I've replaced RC310 with RC310. It didn't help.
>
> So to sum up:
>
> * replacing MM100
> * replacing RC300 with RC310
> * a lot of changes in cables configuration
> * switching from EMS plus to either EMS or [...]

Can you instrument the software, checkpoints, function parameters and log these. You can query for error to get a time frame, I can also wait for the error to appear and press a button.

bbqkees, 2023-12-08:
> > Does it also happen when the Gateway is turned of but still connected to the bus?
> 
> Yes, everything is fine if the power supply is not connected and the gateway is still connected to the bus.
> 
> * MM50
> * RC300
> * GB172T2 - V2
> 
> the communication error on the RC300 occurs when: the mixer setpoint flow temperature (**mixer_hc1_flowsettemp**) changes without changing the boiler setpoint flow temperature (**boiler_selflowtemp**) or calculated flow temperature (**thermostat_hc1_targetflowtemp**).

@MichaelDvP they say the problem _disappears_ when the Gateway is powered off but still connected to the bus.
So that basically rules out any noise, loops or disturbances reflecting (back) on the bus via the Gateway hardware.
I also sent a few users an additional filter or a modified Gateway with additional common mode filter. This made no difference.

Could this maybe be another issue like data collision on the bus or maybe an EMS-ESP query on the bus that sets a certain parameter in the wrong way?

MichaelDvP, 2023-12-08:
The bus is managed by the boiler who polls every device. Ems-esponly sends if the boiler polls the id. If the id is not unique, there is a collision. Thats why there is the possiblitiy to change the id. Default is service-key 0x0B. 
Selecting read-only mode filters writes to the bus, but poll-ack and reads are send to bus. To avoid any TX to the bus you can use telnet `set tx_mode 0` and restart! I think changing the tx-port is not usefull, because the wired port goes hi-z and could trigger the comparator.

eSportler89, 2024-02-22:
anything new?
Still got the communication error :/

proddy, 2024-08-18:
moving to a discussion