emsesp/EMS-ESP32 #1329: Add timestamp in MQTT shower message

proddy, 2023-10-14:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/1328

<div type='discussions-op-text'>

<sup>Originally posted by **sciurius** October 14, 2023</sup>
The shower MQTT message is `{"duration":XXX}`. Would it be possible to add the time stamp?
In HA, the `last-updated` property is not very stable.</div>

proddy, 2023-10-14:
@MichaelDvP said:

"If the esp32-RTC isproper set by NTP, thermostat or manual we can add it.
Do you need timestamp or a time-string (or both)? Of shower end or start/end?
e.g.

{"duration":600, 
"timestamp": 1697294490,
"time":"2023-10-14T16:42:07"
}"

proddy, 2023-10-14:
I think time in HH:MM:SS is universal and ok. It's just for displaying in the Home Assistant dashboard.

sciurius, 2023-10-14:
A timestamp (seconds since epoch) would avoid timezone issues. Otherwise `"2023-10-14T18:46:44+0200"` is ok.

proddy, 2023-10-15:
@MichaelDvP shall you add or I?

MichaelDvP, 2023-10-15:
For the publish it's only 
```
                        time_t now = time(nullptr);
                        if (now > 1576800000) { // year 2020
                            doc["timestamp"] = now; // if needed
                            tm *   tm_ = localtime(&now);
                            char dt[25];
                            strftime(dt, sizeof(dt), "%FT%T%z", tm_);
                            doc["time"] = dt;
                        }
```
but for the HA config it's better you add it to test and choose the right types.

proddy, 2023-10-15:
thanks, I'll add and do a PR now. Along with https://github.com/emsesp/EMS-ESP32/issues/1323

closed 2023-10-15 by proddy