emsesp/EMS-ESP32 #1331: Bosch heatpump CS6800i not recognized (unknown device)

KarstenBr, 2023-10-15:
### PROBLEM DESCRIPTION

My system setup is the following:
- Bosch CS6800iAW heatpump
- Bosch controller UI800 (integrated in heatpump unit)
- Bosch WLAN module K30RF
- Bosch remote thermostat RT800
- Bosch mixer MM100
While using the latest dev SW 3.6.3-dev.1, all of the above listed devices are recognized correctly except the heatpump itself. It turns up as unknown device 0x08 and provides the expected EMS message IDs for such kind of device. I randomly checked some of its messages' content with help of custom entities and their values looked meaningful.
Could you therefore please add this device to the next EMS-ESP dev SW?

### REQUESTED INFORMATION
System information output:
{
  "System Info": {
    "version": "3.6.3-dev.1",
    "platform": "ESP32-S3",
    "uptime": "000+00:48:55.662",
    "uptime (seconds)": 2935,
    "free mem": 184,
    "max alloc": 155,
    "free app": 6094,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -73,
    "IPv4 address": "192.168.190.62/255.255.255.0",
    "IPv4 gateway": "192.168.190.1",
    "IPv4 nameserver": "192.168.190.1",
    "IPv6 address": "fe80:0000:0000:0000:3685:18ff:fe74:0628",
    "static ip config": false,
    "enable IPv6": true,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.190.1",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 936,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 3,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 7954,
    "bus reads (tx)": 1201,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 31
  },
  "Devices": [
    {
      "type": "thermostat",
      "name": "Rego 3000/UI800",
      "device id": "0x10",
      "product id": 253,
      "version": "47.07",
      "entities": 57,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E",
      "handlers ignored": "0x04C9 0x02E1 0x02EB 0xE7 0x0507 0x0508 0x0509 0x050A 0x061E 0x059D 0xF9 0xBF"
    },
    {
      "type": "thermostat",
      "name": "RT800",
      "device id": "0x38",
      "product id": 3,
      "version": "21.03",
      "entities": 3,
      "handlers received": "0x042B 0x047B",
      "handlers pending": "0x0273 0x0A6A",
      "handlers ignored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D7",
      "handlers ignored": "0xD2 0xE2 0x0255 0x02CD 0xBF"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "device id": "0x48",
      "product id": 252,
      "version": "06.02",
      "entities": 0
    },
    {
      "type": "unknown",
      "name": "unknown",
      "device id": "0x08",
      "product id": 8,
      "version": "00.00",
      "entities": 0,
      "handlers ignored": "0xE3 0xE4 0xE9 0xE6 0xD1 0xE5 0x048F 0x0494 0x0495 0x04A2 0x04A8 0x04AC 0x04AE 0x04AF 0x04CA 0x04CB 0xBF 0x14 0xF9 0x048D 0xF7 0xF6"
    }
  ]
}

### EXPECTED BEHAVIOUR
Bosch CS 6800iAW heatpump (device ID 0x08) recognized as known device incl. the related entities of the message IDs 0xE3 0xE4 0xE9 0xE6 0xD1 0xE5 0x048F 0x0494 0x0495 0x04A2 0x04A8 0x04AC 0x04AE 0x04AF 0x04CA 0x04CB 0xBF 0x14 0xF9 0x048D 0xF7 0xF6

### SCREENSHOTS
[image]
[image]

proddy, 2023-10-15:
We'll add it.

MichaelDvP, 2023-10-15:
included in mytestbuild https://github.com/MichaelDvP/EMS-ESP32/releases/tag/test
version 00.00 looks odd, could you do a `read 8 2` in terminal and give us the reply?

KarstenBr, 2023-10-15:
ems-esp:$ read 8 2
000+09:29:05.701 N 1: [emsesp] Me(0x0B) -R-> unknown(0x08), Version(0x02), length: 0x1B
000+09:29:05.722 N 2: [emsesp] unknown(0x08) -W-> Me(0x0B), Version(0x02), data: 08 00 00 00 00 00 00 00 00 01 03 18

The followining SW versions are shown in the diagnostic mode of the heatpump's integrated display:
- interior unit: 23.01.31-5.26-132ef3cb
- external unit: 23.03.03-5.27-a79100d5
- control panel: NF47.07

proddy, 2023-10-15:
@MichaelDvP looks like we need to handle a 3rd subscriber in `EMSESP::process_version()` by looking at the telegram length. Or we just count from the left ignoring all zeros.

MichaelDvP, 2023-10-15:
Hm, first subscriber is valid with device 08 and version 00.00, second and third have no device and also version 00.00, Brand is 01, followed by unknown 03 18. I think we leave it with version 00.00.

proddy, 2023-10-15:
ok, I saw the number 3 and 28 in the telegram and thought we could decipher the version number

KarstenBr, 2023-10-18:
@MichaelDvP I have meanwhile checked your testbuild dev2i. The heat pump is correctly recognized and received data of the related entities look meaningful. Nevertheless I recognized that each time after booting dev2i the EMS status shows some errors in the “EMS reads (tx)” section so that its quality drops to 92% while it is always being 100% with dev1, dev2 and dev3 versions of 3.6.3.
[image]

timostark, 2023-11-02:
@KarstenBr Sorry for a little off-topc: I would be very interested in your setup. Can you show (physically) how you connected the Gateway S3 to your CS6800 and which adapter cables you have chosen? Just this one: https://bbqkees-electronics.nl/product/gateway-s3-standard-wifi-ausgabe/?lang=de and use standard 0.5mm^2 cables to connect it to the EMS-Bus as described in https://junkers-de-de-b.boschtt-documents.com/download/pdf/file/6721872430.pdf?token=899ajude8c3scbu4obfvn4vpm3
?

[image]

bbqkees, 2023-11-03:
@timostark There should be a service jack on a cable under the hood of the inside unit. Then you do not need to use the EMS bus wires.
https://bbqkees-electronics.nl/wiki/gateway/connecting-to-the-boiler.html#service-jack-on-heat-pumps

closed 2023-11-18 by proddy