emsesp/EMS-ESP32 #1334: Buderus Logamax kompakt WS170 not recognized correctly

airhead1234, 2023-10-15:
My Buderus Logamax kompakt WS170 boiler is erroneously reported as Greenstar HIU. In EMS-ESP it is ecognized new boiler with deviceID 0x08.

```
{
  "System Info": {
    "version": "3.6.2",
    "platform": "ESP32-S3",
    "uptime": "000+04:06:13.260",
    "uptime (seconds)": 14773,
    "free mem": 183,
    "max alloc": 171,
    "free app": 6096,
    "reset reason": "Power on reset / Power on reset"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -66,
    "IPv4 address": "192.168.179.100/255.255.255.0",
    "IPv4 gateway": "192.168.179.1",
    "IPv4 nameserver": "192.168.179.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.179.1",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 4801,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "entity format": 1,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": true,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 19876,
    "bus reads (tx)": 4430,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": true,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 28
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Greenstar HIU",
      "device id": "0x08",
      "product id": 219,
      "version": "51.04",
      "entities": 51,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0772 0x0779",
      "handlers fetched": "0xE6 0xEA 0x04",
      "handlers pending": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xEF 0xF7 0x077B 0xF6 0xF9 0x02CC 0x03EE 0x0774 0x0775 0x0776 0x0777 0x0778 0x077A"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 50,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xF6 0xF7 0xE7 0x02E0 0x02EA 0xBF"
    }
  ]
}
```

proddy, 2023-10-15:
apart from the incorrect naming, everything works?

airhead1234, 2023-10-16:
Thank you for the quick reply. In general terms things work ok. However, my feeling is that some data points do not make sense. How could I check this properly? I am still a beginner to EMS-ESP and heating overall.

First, I get 52 entities for a system which is basically a heat exchanger and quite a few of these entities do not make sense. E.g. I have things like system pressure, gas, flame current, fan, ignition etc. These entities do not fit the technology and they also never change. They are either "off" or "0" continuously.

Second, for some entities the values do not seem to make sense. For example the heat network flow temp is stated as 386 °C which it clearly is not.

How can I help integrate the WS170 properly into EMS-ESP?

proddy, 2023-10-16:
The Greenstar HIU is probably the exact same hardware as the Buderus WS170. Both are heat exchangers. We can change the title.

It's interesting you see values for system pressure, ignition etc. Could you go into the terminal using `telnet` and type `read 8 e4` and post here the output of that telegram. That's the one that has this information. I want to check that those data fields are marked as 'disabled'.

airhead1234, 2023-10-16:
My take is that the Greenstar HIU and the Buderus WS170 are related devices but still different in several aspects. For example, the Greenstart HIU seems to have two internal heat exchangers whereas the Buderus WS170 has only one. From this I would figure that the Greenstar HIU should have more entities than its Buderus sibling.

Here is what I get when I run `read 8 e4` via `telnet`:

`000+23:40:49.823 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 35 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 (offset 6)`

proddy, 2023-10-16:
> `000+23:40:49.823 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 35 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 (offset 6)`

can you do it again with the offset? Just a `read 8 e4`

airhead1234, 2023-10-16:
I have nothing to do with the offset. When I entered `read 8 e4 `this is what I got.

```
ems-esp:$ read 8 e4
001+05:20:43.077 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 85 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 (offset 6)

```

proddy, 2023-10-16:
oh ok.

airhead1234, 2023-10-18:
Was the result of `read 8 e4` helpful to find out which data fields make sense and which don't? What is the offset all about, by the way.

Beyond that, there seem to be more things a little odd with WS170. As an example "hc1 cooling" is switch to ON whenever the heating is turned on. And this for a system that have no cooling capability at all.

How can I help implement WS170 in EMS-ESP properly? Thanks!

proddy, 2023-10-18:
I still need to come back with this. I added some test code to simulate your heat exchange so I could trace what was happening. And then that 'offset' threw me. @MichaelDvP any idea why after a `read` command the UBA would provide an offset?

MichaelDvP, 2023-10-19:
> And then that 'offset' threw me. @MichaelDvP any idea why after a read command the UBA would provide an offset?

The read command tries to get all offsets, but the newer heatpumps do not send all offsets. I see it in exhaust-temperature thread, the offsets 27-34 are not in the response. But others have these offsets and the exhausttemp is reported in offset 31. Maybe there is a telegram in the 0xFx range to configure if offsets are shown or not, maybe 0xF6, i see this more often in logs last time checked by the bosch-internet module.

airhead1234, 2023-10-21:
Thank you for looking into this.

Maybe this is something simple in the meantime: The entity `heat network flow temp` ist reported with an incorrect decimal separation. EMS-ESP e.g. shows 592,0 °C whereas the RC310 shows 59,2 °C. The latter seems to be the more reasonable number ;-).

proddy, 2023-10-21:
> Thank you for looking into this.
> 
> Maybe this is something simple in the meantime: The entity `heat network flow temp` ist reported with an incorrect decimal separation. EMS-ESP e.g. shows 592,0 °C whereas the RC310 shows 59,2 °C. The latter seems to be the more reasonable number ;-).

could you run a `read 8 779` from a telnet console. Just want to check against https://github.com/emsesp/EMS-ESP32/issues/1158#issuecomment-1507119714

MichaelDvP, 2023-10-21:
It's just missing DIV10 in these two lines https://github.com/emsesp/EMS-ESP32/blob/df9f75a5c99a6360d3cf68b7d207f37e3558b17a/src/devices/boiler.cpp#L105-L106

airhead1234, 2023-10-21:
> > Thank you for looking into this.
> > Maybe this is something simple in the meantime: The entity `heat network flow temp` ist reported with an incorrect decimal separation. EMS-ESP e.g. shows 592,0 °C whereas the RC310 shows 59,2 °C. The latter seems to be the more reasonable number ;-).
> 
> could you run a `read 8 779` from a telnet console. Just want to check against [#1158 (comment)](https://github.com/emsesp/EMS-ESP32/issues/1158#issuecomment-1507119714)

Here you go:

```
ems-esp:$ read 8 779
001+20:51:49.736 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), HIUMonitor(0x0779), data: 05 07 00 01 2A 02 63 00 15 00 16
```

proddy, 2023-10-21:
> It's just missing DIV10 in these two lines
> 
> https://github.com/emsesp/EMS-ESP32/blob/df9f75a5c99a6360d3cf68b7d207f37e3558b17a/src/devices/boiler.cpp#L105-L106

that's it. I'll add to dev.

airhead1234, 2023-10-27:
I have played a little more and found a few entities that are either not reported or not reported correctly. I have compiled them in the table below. Column 1 shows what I can see on the RC310 connected to the WS170. Column 2 shows what I see or do not see in EMS-ESP. Column 3 are my thoughts on what might make sense to do.

Hope this is helpful.

| RC310 entity | EMS-ESP entity | Comment |
|--------|--------|--------|
| DHW volumetric flow rate [in liters per minute] | Cold water flow rate | Needs DIV10; can this be relabeled from cold water to warm water? |
| Supply temperature (primary side) [in °C] | Heat network flow temp | DIV10 already addressed via https://github.com/emsesp/EMS-ESP32/issues/1334#issuecomment-1773912161 |
| Return temperature (on the heat source side) [in °C] | Not available in EMS-ESP | Can this be added? |
| DHW control valve opening [0 % (valve closed) to 100 % (fully open)] | Not available in EMS-ESP | Can this be added? |
| Heating control valve opening [0 % (valve closed) to 100 % (fully open)] | Burner current power | Values do not match, e.g. RC310 reports 5% and EMS-ESP reports 8%; can this be fixed? |

MichaelDvP, 2023-10-28:
Please go to settings->customization:
- set all entities with zero value thatare not eimplemented to hidden/exclude
- rename entities if the name does not match
- go to system->up/download, download customization and attach it here
So we can see what to remove and rename
Helpfull is also after the cleanup/renaming to go to dashboard->boiler, download the csv and attach it.
 
If the value from burner power does not match to control valve, than it is something other and heating valve is not read yet.

For the missing values return temp and dhw/heating control valves: 
- start a full log
- note the actual values of return temp, etc. at the beginning of the log
- wait a few minutes, min. to a change off the values
- note the new values and stop the log
- attach log here and tell the values on start and end of log
We'll try to identify the telegram/offset of these values and add it

airhead1234, 2023-10-29:
Thank you for the detailed instructions. Here comes my first go at finding missing values. I hope I did this correctly. Attached are 3 log files capturing 3 takes of entities changing. The text below states the changing values with rough timing.

**Take 1** [take_1.txt](https://github.com/emsesp/EMS-ESP32/files/13197704/take_1.txt)

_Entity: DHW volumetric flow rate [in liters per minute] --- seems to be captured already but with 10x value._
0:00 0.0 l/min
1:05 5.7 l/min
1:35 5.3 l/min
1:50 1.0 l/min
1:55 0.0 l/min

_Entity: Return temperature (on the heat source side) [in °C]_
0:00 24.6 °C
0:35 24.5 °C

**Take 2** [take_2.txt](https://github.com/emsesp/EMS-ESP32/files/13197708/take_2.txt)

_Entity: DHW control valve opening [0 % (valve closed) to 100 % (fully open)]_
0:00 0 %
0:37 55 %
0:41 51 %
0:46 48 %
0:57 46 %
1:02 0 %

**Take 3** [take_3.txt](https://github.com/emsesp/EMS-ESP32/files/13197707/take_3.txt)

_Entity: Heating control valve opening [0 % (valve closed) to 100 % (fully open)]_
0:00 0 %
2:08 5 %
2:22 6 %
2:33 8 %
3:03 11 %
3:08 12 %
3:13 13 %
3:33 15 %

_Other: The logging of take 3 contains error messages while the RC310 does not show any errors. Ist this of any relevance?_

```
000+15:16:22.802 N 163: [emsesp] boiler(0x08) -B-> All(0x00), ErrorMessage(0xBF), data: 08 DB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+15:16:23.018 N 164: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B
000+15:16:23.074 N 165: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB 00 00 00 00 00
000+15:16:23.199 N 166: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 27)
000+15:16:23.253 N 167: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+15:16:23.407 N 168: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 54)
000+15:16:23.463 N 169: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 54)
000+15:16:23.559 N 170: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 81)
000+15:16:23.613 N 171: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: DB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB 00 00 00 00 00 00 (offset 81)
000+15:16:23.769 N 172: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 108)
000+15:16:23.821 N 173: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 108)
000+15:16:23.948 N 174: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 135)
000+15:16:24.001 N 175: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 DB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB (offset 135)
000+15:16:24.156 N 176: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 162)
000+15:16:24.212 N 177: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 DB 00 00 00 00 00 00 00 (offset 162)
000+15:16:24.307 N 178: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAErrorMessage3(0xC2), length: 0x1B (offset 189)
000+15:16:24.342 N 179: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 189)
```

Thank you for looking into this and I will follow up with the other todos you had for me separately.

airhead1234, 2023-11-04:
@MichaelDvP 

I have now hidden and excluded entities that continuously have zero value for my boiler as per your instructions.

Please find attached both the customization export as well as the boiler CSV.

[emsesp_customizations.json](https://github.com/emsesp/EMS-ESP32/files/13256021/emsesp_customizations.json)
[Kessel_Greenstar HIU_Logamax kompakt WS170.csv](https://github.com/emsesp/EMS-ESP32/files/13256022/Kessel_Greenstar.HIU_Logamax.kompakt.WS170.csv)

Thanks!

airhead1234, 2023-11-11:
@MichaelDvP 

Running 3.6.3-dev.5b I do see the new entities I had asked for in https://github.com/emsesp/EMS-ESP32/issues/1334#issuecomment-1784050477. Thanks!

One of these entities, however, behaves strangely: "**Return temperature (on the heat source side) [in °C]**" is only visible once in a while and disappears again. It seems like the entity pops up if its values change and after a few seconds then disappears again. Other entities also stay (in both the EMS-EPS UI as well as MQTT) if their values do not change for a longer period of time.

Maybe you can look into this. Thanks!

proddy, 2023-11-11:
@airhead1234 is this `sysrettemp` and how often is up appearing/disappearing? 

@MichaelDvP maybe we should add some logic to make certain device values 'sticky' so if a x8000 comes in it will still retain its old value?

airhead1234, 2023-11-11:
@proddy: Yes, this is `sysrettemp`. I appears every minute or every other minute and stays for less than 5 seconds before it disappears again.

Of the ~50 entities I get from my boiler, `sysrettemp` seems to be the only one with this beavior.

proddy, 2023-11-11:
thanks, could you open up a Telnet console and do a `watch 53` and leave it for 2-3 minutes, then paste the content into a .txt file, zip it and attach it to this issue?

MichaelDvP, 2023-11-11:
I think it's `retTemp` and for the HIU read from telegram 0x779 and overwritten on receive 0xE4. In 0xE4 we have to ignore the value if zero and not set to `NOTSET`. (Another bug is double register the value for HIU and common boiler).

airhead1234, 2023-11-11:
I have had `watch 53 `for 15 minutes but not a single telegram has popped up.

MichaelDvP, 2023-11-11:
Please do a `read 8 E4` and `read 8 779` I like to see if the retTemp in E4 is 0x0000 or 0x8000.

airhead1234, 2023-11-11:
Here you go:

```
ems-esp:$ read 8 E4
000+02:25:48.536 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAMonitorFastPlus(0xE4), data: 25 01 83 64 08 01 00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 (offset 6)
```

```
ems-esp:$ read 8 779
000+02:26:20.770 N 2: [emsesp] boiler(0x08) -W-> Me(0x0B), HIUMonitor(0x0779), data: 05 07 00 01 46 02 54 05 00 00 00
```

airhead1234, 2023-11-11:
Thank you for the quick fix. Using `EMS-ESP-3_6_3-test_7-ESP32_S3.bin` the return temperature is displayed permantently.

However, when moving from `3.6.3-dev.5b` to `3.6.3-test.7` four entities of the boiler have disappeared of which at least two do make sense for the WS170.

`gas` and `heating pump` are missing now and also do not pop up after waiting for a while. These two do make sense on the WS170. If you could bring them back this would be much appreciated.

`exhaust temperature` and `total energy` are missing as well but do not make much sonse on the WS170 anyways.

MichaelDvP, 2023-11-12:
Yes, i have removed some burner related entities from heatpumps and HIU to save memory. The pump was a mistake, i'll add it back, but what is `gas` used for in your system? 
Check what you need from the excluded: exhausttemp, heatblock, gas, gas stage 2, flamecurrent, fan working, ignition, oil preheater, minburnpower, maxburnpower, boiler hyst on, boiler hyst off, absburnpower.

airhead1234, 2023-11-12:
Thanks. I have split the entities you listed in 3 categories:

**Please add back for the WS170:**
- `gas` is what turns on whenever the actual heating is on, i.e. when `heatvalve` opens `gas` turns on. However, `gas` is probabley not the best name for this entity in case of the WS170. If there is anything that fits better, please rename. 
- `heating pump` is what it says and works for the WS170.

**These entities have a value but neither the value not the entity name make sense for the WS170. Maybe you can make sense out of this:**
- `exhausttemp` has a value for the WS170 - 2,571.1° C at the moment - but does not make sense

**These entities to my understanding do not make sense for the WS170:**
- `heatblock` has no value and never changes on my WS170
- `gas stage 2` has no value and never changes on my WS170
- `flamecurrent` has no value and never changes on my WS170
- `fan working` is always "off" on my WS170
- `ignition` is always "off" on my WS170
- `oil preheater` has no value and never changes on my WS170
- `minburnpower` has no value and never changes on my WS170
- `maxburnpower` has no value and never changes on my WS170
- `boiler hyst on` has no value and never changes on my WS170 
- `boiler hyst off` has no value and never changes on my WS170 
- `absburnpower` has no value and never changes on my WS170

airhead1234, 2023-11-17:
@MichaelDvP 

Could you please modify entity `heatingactive` in such a way that it displays what `burngas` used to display for the WS170 before you removed `burngas`. 

`heatingactive` is permanently off for my system, but I figure it should be on whenenver the heatvalve opens. `burngas` did capture exactly that but as a name does not make any sense for a system without gas.

Thank you!

airhead1234, 2023-11-18:
One more thing: Similar to `heatingactive` which is permanently off for my system, also `tapwateractive` is permanently off. In the log file attached I have captured when `tapwateractive` should change its state.

**Take 4** [take_4.txt](https://github.com/emsesp/EMS-ESP32/files/13401614/take_4.txt)

_Entity: `tapwateractive`_
0:00 Off
0:21 On
0:55 Off
1:10 On
1:30 Off

Could you please implement this entity for the WS170 in the EMS-ESP accordingly. Thanks!

MichaelDvP, 2023-11-19:
> before you removed burngas.

It was added back as `heating`
> Entity: tapwateractive
0:00 Off
0:21 On

Hmm, i've found 0x00774 offset 13 changing it's state on these timings, but it goes to 100 when you write Off and to 0 if you write On, or does `0:21 On` mean `On until 0:21, then off`?
- 0:00 Off - (1s later) +01:39:11.621 N 9: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0774), data: 08 00 00 00 00 00 00 00 00 00 00 00 00 64 00 00 00 50 00 50 00 00
- 0:21 On - 01:39:30.656 N 28: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0774), data: 00 (offset 13)
- 0:55 Off -01:40:06.680 N 141: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0774), data: 64 (offset 13)
- 1:10On - 01:40:19.747 N 166: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0774), data: 00 (offset 13)
- 1:30 Off -01:40:41.597 N 221: [emsesp] boiler(0x08) -B-> All(0x00), ?(0x0774), data: 64 (offset 13)

airhead1234, 2023-11-19:
Thanks for the quick reply. No, `0:21 On` means that it was `Off` until 0:21 and turn to `On` at 0:21.

I have done a new recording trying to be more precise in my timing. Maybe you can have a look again.

**Take 5** [take_5.txt](https://github.com/emsesp/EMS-ESP32/files/13402838/take_5.txt)

Entity: `tapwateractive`
0:00 Off
0:25 On, i.e. turning from Off to On
0:45 Off, i.e. turning from On to Off
1:06 On, i.e. turning from Off to On
1:26 Off, i.e. turning from On to Off

Please note that the reverse version of On and Off could be the `heating` entity. The WS170 has only one heat exchanger. If `heating` is On while `tapwateractive` is turned On, `heating` will turn Off. Once `tapwateractive` is turned Off again `heating` could be turned On again - depending on whether it is cold enough etc. I would assume that there are some time delays between those effects, hence I tried to be more precise with my time information above.

Thank you!

MichaelDvP, 2023-11-19:
Can you add some custom entities and monitor with:
0x774, offset13, bool or uint as mentioned before (value changes 0/100)
0xE3, offset 5, bool or UINT for tapwater (value changes 0/1)
0xE3, offset 2, bool or UINT for heating (value changes 0/1)

You can also monitot 0xE4, offset 4 and 5, they seems to change from 00 00 to 08 01 in the time range you changes tapwater.

airhead1234, 2023-11-19:
Thank you, will do. Which `Device ID` to use for those custom entities? I assume this has to be the boiler, but do not know where to look up its `Device ID`.

airhead1234, 2023-11-19:
0xE3, offset 5 turns to On once the tapwater is On.
0xE3, offset 2 turns to On once the heating is On.

0x774, offset13 is a little strange:
- Generally, it is 0 if tapwater is On and 100 if heating is On.
- However, I have sometimes noticed a delay in its switching. Tapwater and heating sometimes change first and 0x774, offset13 changes 2-3 seconds later.
- I will try to investigate this some more.

MichaelDvP, 2023-11-19:
> 0x774, offset13 is a little strange:

Could it be the heatingPumpModulation?

airhead1234, 2023-11-19:
Good idea with pump modulation. I will investigate this. 

Another interesting thing: 0xE3, offset 2 is similar to what `burngas` used to be, but is not exactly the same. Both are On and Off mostly at the same time, but sometimes when `burngas` is Off 0xE3, offset 2 stays On. In these situation the heating valve is closed and also my RC310 does not show heating anymore. So, `burngas` seems to be the better representation of the actual heating than 0xE3, offset 2. Maybe also 0xE3, offset 2 is related to the pump?

airhead1234, 2023-12-02:
@MichaelDvP 

I have looked into heating and tapwater of the WS170 a little more. It seems to me that 0xE3, offset 2 indicates that the WS170 is in general heating mode whereas 0xE3, offset 5 indicates that the WS170 is general tapwater mode. Due to its architecture with only one heat exchanger the WS170 is either in one general mode or the other. 

However, the general mode does not actually indicate whether it really is heating or not or running warm tapwater or not.

Whether the WS170 is actually heating is best indicated by what used to be `burngas`, e.g. in version 3.6.3-dev.5b. As you can see in the screenshot attached `burngas` is sometimes OFF while 0xE3, offset 2 is still ON. During these times the RC310 physically still shows a heating sign on its display and the pipes are warm.

[image]

If I could make a wish you would set `heatingactive` to what `burngas` used to be for the WS170. As of now `heatingactive` is permanently OFF for the WS170.

There is also `tapwateractive` which is permanently OFF, but we still need to find out where the information of tapwater actually running is stored.

airhead1234, 2023-12-09:
`wwcurflow` shows the tapwater being dispensed by the WS170.

Oddly, the value reported by EMS-ESP "jumps" between the actual value and zero. Compared with what is reported on the RC310 the value matches, but on the RC310 it is reported continuously. 

Hopefully the screenshot attached explains what I am trying to describe. Whenever the tapwater valve is open (graph on top) there should be a flow of tapwater being reported by `wwcurflow` (graph on bottom).

[image]

airhead1234, 2023-12-10:
Thank you, @MichaelDvP. With dev.4 it works as expected.

airhead1234, 2023-12-16:
Thank you @MichaelDvP and @proddy for modifying EMS-ESP to make it work with my WS170. Looks like most values are mapped now. I am closing this issues.

Thanks again!

closed 2023-12-16 by airhead1234