emsesp/EMS-ESP32 #1335: Energy statistics entity nrgheat doublecounts ww preperation

tp1de, 2023-10-16:
@MichaelDvP 
I switched on heating 2 days ago and recognized that nrgheat includes now ww preparation as well. 

I see that my heating pumps are off when ww preparation is done. Nevertheless the nrgheat value increases during ww preparation.
thermostat_hc .._wwprio is true so nrgheat should not increase. 

How do you determine and differentiate between heating/ww preparation?

MichaelDvP, 2023-10-16:
Please check if `heating_active` and `tapwater_active` is right. Do have a 3-way-valve or charge pump? With chargepump check if the heatingpump turns off for dhw preparation.

tp1de, 2023-10-16:
tapwater_active is ok. I am using wwcharging since I do have a ww storage tank- but they share the same data.

heatingactive was on during ww preparation and off afterwards since boiler temp was too high for heating circuits.
So heating active is not the right indicator. Within RC310 there are parameters to be configure if ww prep has priority to heating circuits or not. This can be set per hc. I believe that most installations will use ww priority.

To check this within ems-esp either test thermostat hcxx wwprio if aktivated or check the heating pumps if switched off.

tp1de, 2023-10-16:
Since it it not possible to split energy consumption between ww and heating when ww priority is not set, 
I believe ist fair enough to just check tapwater_active or wwcharging. When active then heating is off. 
This avoids double counts.

tp1de, 2023-10-30:
@MichaelDvP There is no change in latest dev.5 version.  It's still the double count. 
Could you please give a short status update?

MichaelDvP, 2023-10-30:
It's fixed in my dev2, I'll make a PR when the changes are confirmed to work (EM100, roomcontrol, heatpump energy, etc.)
My dev build is much behind the official, i concentrate on the dev2 now.

proddy, 2023-11-19:
@MichaelDvP did this make it to the 3.6.3?

tp1de, 2023-11-19:
Energy statistics are very exact .... better then from km200.
Please release.

tp1de, 2023-12-02:
@MichaelDvP I lost any transparency if this topic is still open or in the meantime within original dev?

MichaelDvP, 2023-12-02:
I think it's in the official dev, but i have no possibility to check, If you think it's ok now, close the issue.

closed 2023-12-03 by tp1de