emsesp/EMS-ESP32 #1338: unnecessary dhw prefixes

pswid, 2023-10-16:
I just noticed that the Energy Dasboard in HomeAssistant no longer shows the energy used by DHW. This is because in version [3.6.2](https://github.com/emsesp/EMS-ESP32/releases/tag/v3.6.2) the names of several entities related to DHW have changed (more DHW tags added). What is this change for? In my opinion, the _statistical_ entities of the entire system (such as the number of starts, operating time, energy consumed and supplied for heating, cooling, pool and dhw) should not contain any prefixes (neither DHW nor HC). They introduce _confusion_ in the nomenclature and spoil the view (see below). 
In addition, HomeAssistant does not have the option of maintaining continuity of energy monitoring of a specific device if the entity name changes (there is no option to copy the history of the old entity to the new one nor merge two entities). So, device name/id modification of long-term recorded entity is a breaking change!

I strongly suggest to revert https://github.com/emsesp/EMS-ESP32/commit/ad81f84c02dcca0a95bc4740b510e2a736fef930 commit.

[image]

proddy, 2023-10-16:
You have a valid point about the naming. We'll look into it.

Can you share the Energy Dashboard setup screen? I don't think the sensor names have changed use in HA discovery.

MichaelDvP, 2023-10-18:
Ok, this is the complain we have expected in #1301. I'll remove from the texts what i can identify.

bluegaspode, 2023-10-22:
this 'dwh prefix change' not only changed wordings in the UI, also the API keys

/api/boiler/info
have changed (breaking my data ingestion to influx for some values).

fields where I see it a change and 'dwh' is used as prefix and suffix.

dhw operating time compressor dhw
dhw energy consumption compressor dhw
dhw aux elec. heater energy consumption dhw
dhw total energy warm supplied dhw
dhw hp dhw
dhw heat limit dhw

all these value have 'siblings' for heating/cooling/pool  where the differenciation happens only via the suffix.

proddy, 2023-10-22:
@MichaelDvP shall we just revert the change and look again at how to fix the translations?

MichaelDvP, 2023-10-23:
No, the split to dhw(ww) tags was early in v2 versions for all boiler entities, including siblings like burnerstarts (heating/dhw). The later added heatpump entities are not always tagged right. People using the longname entity creation will have a change in any way, so if we change it, make it right. 
Also with ~150 entities for heatpumps we will soon run into mqtt overflow and have to think about tags for cooling/pool to split to more boiler_data_xy messages.

proddy, 2024-01-06:
@pswid do you still see this behaviour on 3.6.5-dev-8 ?

pswid, 2024-01-08:
Unfortunately there is no improvement in 3.6.5dev8. Those five entities I mentioned at the beginning of the thread are still prefixed with DHW.
I don't really understand the reason why these entities were preceded by these prefixes. The fact that they refer to the "sub system" of domestic hot water is, IMHO, not a sufficient reason. These are entities containing "global" statistical information of the entire system and as such should not have any prefixes. Entities related to heating (e.g. `nrgheat `or `heatingstarts`) do not have the HC prefix, right?
I am still in favor of removing the DHW prefixes from these five entities. If someone would like to keep them, he can add prefixes using customizations. Unfortunately, it doesn't work the other way (you can't remove the prefix with customizations).

pswid, 2024-01-08:
One more thing, I didn't mention earlier that I'm still using Entity ID format: "single instance, long name (v3.4)"

proddy, 2024-01-08:
the reason goes back to how version 2 (and then 3) was architected. The earlier versions of EMS-ESP ran on ESP8266's with limited memory so we had to pull every trick in the book to make it fit into the available memory, specially with MQTT queue's. So the message types are organized as what we call 'tags' which is why you see boiler_data and boiler_ww etc. 

We should change this. I need to look into how with Michael.

proddy, 2024-01-10:
I'm looking into this.

pswid, 2024-01-11:
It is not a problem that the boiler data is divided into boiler_data and boiler_data_ww topics. The only problem is that (since version 3.6.2?) the names of some "statistical" entities regarding the whole heating/cooling system have changed by adding DHW prefix to them (and remove DHW sufix). In my opinion, this change was mistaken because it broke the "consistency of naming" of the boiler's statistical entities. Also, and even more importantly, this caused a change of the uniq_id and obj_id in the discovery topics for those who, like me, still use "long name (v.3.4)" option. This resulted in the loss of continuity of long-term recording of these entities in HomeAssistant. BTW, this is the main reasons why my main EMS-ESP still runs on v.3.6.1.

proddy, 2024-01-11:
yes, I get that. I added a new test function to simulate your setup, and without everything I try and don't see the 'dhw' added to the end of the name. I'll try and spend some more time this weekend.

proddy, 2024-01-13:
sorry, just can't reproduce

[image]

pswid, 2024-01-14:
Thanks for your time but it seems you didn't understand me. Currently, there is no longer `dhw` added to the end of "statistical" entity names (at least in the EN version). Now `dhw` is only at the beginning and (in my opinion) it **should be at the end**, as it was before. Therefore, in order not to break "naming consistency", the `dhw` prefix/tag should **not** be used in entities:

`operating time compressor xxx`
`energy consumption compressor xxx`
`aux elec. heater energy consumption xxx`
`total energy supplied xxx`
`heat limit xxx`
`energy xxx`

All these entities should be without tag/prefix, regardless of whether xxx is heating, cooling, **dhw**(!), pool, total, etc. Currently, of the above-mentioned entities, only hot water entities have dhw as a prefix and it looks bad, especially in Polish (English tag "dhw" is Polish "CWU").

for example `dhw operating time compresor` should be `operting time compresor dhw` and in Polish `CWU energia pobrana przez sprężarkę` should be `energia pobrana przez sprężarkę na c.w.u.`  

[image]

proddy, 2024-01-14:
ah I see now. I'll do a quick change and create a binary. Which ESP32 platform are using? plain all 4MB, or a S2, S3, C3 etc?

commit 18dd207d (2024-01-17): fixes #1338
commit 4ad5c729 (2024-01-17): fixes #1338
closed 2024-01-17 by proddy